aircrack-ng/rtl8188eus #148: make command errors in Kali Linux

MohammadRezaArani, 2021-11-14:
```
sudo make                                                                                                                                     
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.14.0-kali2-amd64/build M=/home/sam78/Desktop/rtl8188eus  modules
make[1]: Entering directory '/usr/src/linux-headers-5.14.0-kali2-amd64'
  CC [M]  /home/sam78/Desktop/rtl8188eus/os_dep/osdep_service.o
In file included from /home/sam78/Desktop/rtl8188eus/include/drv_types.h:30,
                 from /home/sam78/Desktop/rtl8188eus/os_dep/osdep_service.c:19:
/home/sam78/Desktop/rtl8188eus/include/wifi.h:970: warning: "IEEE80211_MAX_AMPDU_BUF" redefined
  970 | #define IEEE80211_MAX_AMPDU_BUF 0x40
      | 
In file included from /home/sam78/Desktop/rtl8188eus/include/osdep_service_linux.h:83,
                 from /home/sam78/Desktop/rtl8188eus/include/osdep_service.h:50,
                 from /home/sam78/Desktop/rtl8188eus/include/drv_types.h:27,
                 from /home/sam78/Desktop/rtl8188eus/os_dep/osdep_service.c:19:
/usr/src/linux-headers-5.14.0-kali2-common/include/linux/ieee80211.h:1657: note: this is the location of the previous definition
 1657 | #define IEEE80211_MAX_AMPDU_BUF  0x100
      | 
/home/sam78/Desktop/rtl8188eus/os_dep/osdep_service.c: In function ‘isFileReadable’:
/home/sam78/Desktop/rtl8188eus/os_dep/osdep_service.c:2203:11: error: implicit declaration of function ‘get_fs’; did you mean ‘get_sa’? [-Werror=implicit-function-declaration]
 2203 |   oldfs = get_fs();
      |           ^~~~~~
      |           get_sa
/home/sam78/Desktop/rtl8188eus/os_dep/osdep_service.c:2203:11: error: incompatible types when assigning to type ‘mm_segment_t’ from type ‘int’
/home/sam78/Desktop/rtl8188eus/os_dep/osdep_service.c:2204:3: error: implicit declaration of function ‘set_fs’; did you mean ‘sget_fc’? [-Werror=implicit-function-declaration]
 2204 |   set_fs(get_ds());
      |   ^~~~~~
      |   sget_fc
/home/sam78/Desktop/rtl8188eus/os_dep/osdep_service.c:2204:10: error: implicit declaration of function ‘get_ds’; did you mean ‘get_da’? [-Werror=implicit-function-declaration]
 2204 |   set_fs(get_ds());
      |          ^~~~~~
      |          get_da
/home/sam78/Desktop/rtl8188eus/os_dep/osdep_service.c: In function ‘retriveFromFile’:
/home/sam78/Desktop/rtl8188eus/os_dep/osdep_service.c:2241:12: error: incompatible types when assigning to type ‘mm_segment_t’ from type ‘int’
 2241 |    oldfs = get_fs();
      |            ^~~~~~
/home/sam78/Desktop/rtl8188eus/os_dep/osdep_service.c: In function ‘storeToFile’:
/home/sam78/Desktop/rtl8188eus/os_dep/osdep_service.c:2276:12: error: incompatible types when assigning to type ‘mm_segment_t’ from type ‘int’
 2276 |    oldfs = get_fs();
      |            ^~~~~~
cc1: some warnings being treated as errors
make[3]: *** [/usr/src/linux-headers-5.14.0-kali2-common/scripts/Makefile.build:276: /home/sam78/Desktop/rtl8188eus/os_dep/osdep_service.o] Error 1
make[2]: *** [/usr/src/linux-headers-5.14.0-kali2-common/Makefile:1882: /home/sam78/Desktop/rtl8188eus] Error 2
make[1]: *** [/usr/src/linux-headers-5.14.0-kali2-common/Makefile:232: __sub-make] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.14.0-kali2-amd64'
make: *** [Makefile:2286: modules] Error 2

```
[image]
[image]
[image]
[image]

**Ok then Tell me what to do next??????
14 Nov 2021 ==> 1 full day trying to do this but still no progress!**

MrPombo, 2021-12-03:
To com o mesmo problema

xxxPxxx, 2021-12-30:
Tbm estava com este problema.
Descobri que na pasta lib/modules/  havia 2 pastas. 5.14 e 5.15.
A pasta 5.15 havia a pasta BUILD, necessária para o MAKE.
O make usa a pasta 5.14, então copiei o a pasta BUILD de 5.15 para a pasta 5.14 e funcionou.

closed 2022-03-19 by kimocoder