aircrack-ng/rtl8188eus #150: net/ipx.h: No such file or directory

inegan, 2021-11-25:
[image]

when trying Sudo make i get this

this happen after kernel update to 5.15.1-amd64-desktop

Please help

Gtkall, 2021-11-28:
I get this error too. Seems like they finally [dropped support](https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/commit/include/net?id=6c9b40844751ea30c72f7a2f92f4d704bc6b2927) from the 5.15 kernel.

drygdryg, 2021-11-30:
Linux 5.15 support implemented here: https://github.com/aircrack-ng/rtl8188eus/pull/146
You can build driver from my fork before the maintainer has committed the changes to this repository.

tarzq306, 2021-12-01:
> Linux 5.15 support implemented here: #146 You can build driver from my fork before the maintainer has merged the changes to this repository.

Thanks for forking this repo. Now I'm was able to build on my fedora 35 running on kernel 5.15.

mohmus82, 2022-01-01:
i don't get it so can you you explain more

n3gwg, 2022-01-03:
Perhaps if you could leave the link that we should clone from to get the new repo with bug removed, that'd be wonderful. What say you?

TierGamerpy, 2022-01-03:
> Linux 5.15 support implemented here: #146 You can build driver from my fork before the maintainer has merged the changes to this.

Thanks i was able to install this drive now.

Amnaik17, 2022-01-04:
> @drygdryg
> 
> > Linux 5.15 support implemented here: #146 You can build driver from my fork before the maintainer has merged the changes to this.
> 
> Thanks i was able to install this drive now.

Can you please tell me how to do that... I dont know

n3gwg, 2022-01-04:
Several people have asked for instructions, I asked for a link to the repo, and no one has responded. I suspect either no developers wants to "waste his time providing instructions" or perhaps they are just understanding that people need some guidance. If I had a link that could be used to clone the repo I could likely compile it and explain how, but, my question got ignored as well.

Amnaik17, 2022-01-04:
> Several people have asked for instructions, I asked for a link to the repo, and no one has responded. I suspect either no developers wants to "waste his time providing instructions" or perhaps they are just not understanding that people need some guidance. If I had a link that could be used to clone the repo I could likely compile it and explain how, but, my question got ignored as well. At this point in the absence of a link to the repo or instructions, all we can do is wait until the code gets merged into the code base. Then the master repo will have it and one can pull it and compile it that way. Oh well...

https://github.com/Amnaik17/rtl8188eus.git

this is my repo bt idk weather it is correct or not

Gtkall, 2022-01-04:
> Several people have asked for instructions, I asked for a link to the repo, and no one has responded. I suspect either no developers wants to "waste his time providing instructions" or perhaps they [...]

[Here](https://github.com/drygdryg/rtl8188eus) is @drygdryg 's fork, 1 commit ahead, proven working via compilation on my system. You were only two clicks away from finding it btw...

n3gwg, 2022-01-04:
So now there are two different source code repos, I presume the second one is the one that actually works? At least now I can download them and try to compile them and test them. I will see if I can get one of them to work on Kali.

Thank you.

n3gwg, 2022-01-05:
I am trying to build this on Kali Linux and I get this output, any recommendations?

2495 /dev/pts/1 kali@kali:~/amnaik17/rtl8188eus $ make
cut: /etc/redhat-release: No such file or directory
Makefile:2133: *** missing separator.  Stop.

kali@kali:~/amnaik17/rtl8188eus $ sudo touch /etc/redhat-release

kali@kali:~/amnaik17/rtl8188eus $ make
Makefile:2133: *** missing separator.  Stop.
 
kali@kali:~/amnaik17/rtl8188eus $ 

A look at the makefile dictates it is looking to determine the Red Hat version (I am attempting compilation under Kali Linux). Even if the test fails, just having touched the file ought be enough for the makefile to continue upon its way. Is there something I am missing or is this another bug?

Thanks!

bolokanar, 2022-01-05:
> I am trying to build this on Kali Linux and I get this output, any recommendations?
>
> 2495 /dev/pts/1 kali@kali:~/amnaik17/rtl8188eus $ make cut: /etc/redhat-release: No such file or directory [...]

/etc/redhat-release can be safely ignored.

To install drygdryg's fix simply git clone:

git clone https://github.com/drygdryg/rtl8188eus.git

Do 'make' and then 'make install'.

P.S.: Or 'sudo make install' if you do it as non-root. (I don't use really use sudo, kinda forgot about it)

n3gwg, 2022-01-05:
So the story is this: I got some "el cheapo" WiFi 5 AC1200Mbps Alibaba and the bazillion thieves special for like $10 shipped to use on my GD8200 laptop (it has an Intel N board in it, though maybe I'll replace the mini PCI board at some point). I did not realize (nor did I ask) what chip set this wench had before ordering it (I ought to have been more careful to a Ralink or MediaTek), but that was my mistake.

I know that in a github forum everyone is presumed to be an expert developer, but I am not, nor have I ever played a devleoper on TV! :)

I thought I had the right repo, clearly not. Upon following your instructions I have achieved bliss!

However, it is worthy of notation that instead I ran "sudo dkms-install.sh" instead.
My device is:
Bus 001 Device 008: ID 0bda:b812 Realtek Semiconductor Corp. RTL88x2bu [AC1200 Techkey]

I hope this driver will work for it, I will be rebooting tto test it now.

bolokanar, 2022-01-05:
@n3gwg If you followed the instructions as-is it should have installed properly, granted everything is fine with your setup.

The problem is that right now the drivers for this wifi act like crap. It used to work fine but the newer kernel has made some changes that apparently make the dongle rather unusable. Wifi link is piss poor and it has speeds that put even 1970s to shame.

P.S. This fix is for making the driver compile. But it appears the wifi doesn't work well with neither the stock driver (the one that comes builtin with the Linux kernel) nor any other driver that I know about.

aikonmani, 2022-01-06:
TP-LINK TL-WN725N RTL8188EU working driver for kali linux kernel 5.15....

https://github.com/drygdryg/rtl8188eus.git

n3gwg, 2022-01-06:
Aikonmani,

You posted a link that was posted earlier, are you suggesting there have been some updates to the code today or since the last link was posted for drygdryg's code? I will have to try to download and build it again I suppose.

n3gwg, 2022-01-06:
kimocoder,

If the driver is built into 5.15 (I have Kernel 5.15 running now), why would I need to build it from source then?

What driver do I modprobe in?

I tried modprobe rtl8xxxu (which loaded it), but other than the wlan0 (for mini PCI card in the laptop), no new adapter showed up in ifconfig for Realtek.

Did I miss something?

bolokanar, 2022-01-06:
> kimocoder,
> 
> If the driver is built into 5.15 (I have Kernel 5.15 running now), why would I need to build it from source then?
> 
> What driver do I modprobe in?

The kernel has a builtin driver called: r8188eu and it is loaded by default out-of-the-box.
This driver here is named: 8188eu. So r8188eu has to be blacklisted.
And no no new commits. I have tried all of the rest pulls and neither works for me.

To be clear: TL-WN722N v3.2 over here. And no drivers work well as of time of writing.
Wifi's link strength drops to like 10 or sometimes even zero when there is traffic. Peaks up a little and then crashes to 0-10 again in a loop. The speeds that I get are like 1024-3000 bytes.

I use usb tethering via my Galaxy phone to have WIFI connection for now.

bolokanar, 2022-01-06:
> kimocoder,
> I tried modprobe rtl8xxxu (which loaded it), but other than the wlan0 (for mini PCI card in the laptop), no new adapter showed up in ifconfig for Realtek.

For RTL88x2bu you need to try different driver...
Maybe this? https://github.com/RinCat/RTL88x2BU-Linux-Driver
Or this: https://github.com/cilynx/rtl88x2bu

n3gwg, 2022-01-06:
bolokanar, et alia:

This driver indeed did work! I modprobed the drive and BAMM, POW (in full batman style) the LED started flashing, the device showed up in ifcfg, and one more person in the world began smiling! :)

Thank you all for your time and input.

Hopefully this new driver will swiftly make its way into the next version of the kernel.

Stuart

Cat22, 2022-01-09:
I had this same error when trying to build the rtl8821cu driver for the usb wireless dongle. 
I could build just fine in 5.8.11 but in 5.15.5 i got that net/px.h error
After reviewing the Makefile, I realized that since no one was ever going to use ipx that all 
i really needed was the ipx.h files  so it could resolve #defines and things like that
So, what i did was copy the  ipx.h files from:
```
/usr/src/linux-5.8.11/include/uapi/linux/ipx.h
/usr/src/linux-5.8.11/include/net/ipx.h

```to
```
/usr/src/linux-5.15.5/include/uapi/linux/ipx.h
/usr/src/linux-5.15.5/include/net/ipx.h
```
and since it was never going to try to link against ipx it worked just fine

n3gwg, 2022-01-10:
I just realized that the distinct possibility exists that the Linux header files for the version of the kernel you are attempting compilation pursuant to are ostensibly absent.

n3gwg, 2022-01-10:
The RinCat Version of the driver mentioned herein above worked for me with absolute perfection. In fact, I was able to do a DKMS installation.

ing-santillan, 2022-01-10:
The file is in a new folder --> /usr/include/netipx/ipx.h

Its-Choudhary, 2022-01-10:
could you please upload the ipx files here. cause some users installed new kali linux so the old kernels are not present. please its a humble request

levinetworks, 2022-01-10:
> The file is in a new folder --> /usr/include/netipx/ipx.h

I'm quite new to Kali Linux and I am installing the RTL8188EUS driver for my wifi adapter. I get that the file is in a new folder, but don't know what to do with that information. Could you provide me with some instructions?

n3gwg, 2022-01-10:
Alberto,

Whilst I see no reason for you to have to apologize for learning English (that is certainly a useful endeavor in today's world), this is a technical forum and you would be most greatly served by assuring that your thoughts are expressed as you intend them to be (so that we understand you and give you the proper answers you're looking for). Thus, I would make the following suggestion: you might wish to use Google Translate to first translate your message from your native language into English. Review the English, then translate it back via Google Translate to your native language. Does it still comport with your intended meaning?

Please note I do not mean this suggestion in any sort of derisive, pontificating, or condescending manner, but purely in a constructive manner.

I presume that most forum members would agree with me, in that the intention of posting in this forum is either to receive or provision assistance. As such, I would presume you surely wish your thoughts to be understood in the manner you intend them so that you will receive the correct guidance you were searching for.

Good luck!

ing-santillan, 2022-01-10:
Thanks, I know your comment is constructive but I think my comment helped the forum more than yours.  I know Google translate thanks. 

Long life and property! 🖖

mmbaguette, 2022-01-12:
Using TP-Link 802.11ac NIC. I don't know much about WiFi adapters, but I'm trying to get this to work on Kali Linux in VMware. I'm following this https://www.youtube.com/watch?v=VAMP8DqSDjg and then this: https://www.youtube.com/watch?v=tYnjMiTTdms. I'm here because I ran into the same issue with `net/ipx.h: No such file or directory`. 

I followed these instructions after I got that error.

> To install drygdryg's fix simply git clone:
> 
> git clone https://github.com/drygdryg/rtl8188eus.git
> 
> Do 'make' and then 'make install'.
> 
> P.S.: Or 'sudo make install' if you do it as non-root. (I don't really use sudo, kinda forgot about it)

My WiFi adapter isn't blinking when I connect it to Kali Linux in VMware. What am I doing wrong?

cyber-nom, 2022-01-25:
> I get this error too. Seems like they finally [dropped support](https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/commit/include/net?id=6c9b40844751ea30c72f7a2f92f4d704bc6b2927) from the 5.15 kernel.

buddy can you help with this i am getting same error

mmbaguette, 2022-01-25:
@cyber-nom Check everywhere on your WiFi adapter and see what the model is or the brand.

VGLALALA, 2022-02-15:
> > I am trying to build this on Kali Linux and I get this output, any recommendations?
> > 2495 /dev/pts/1 kali@kali:~/amnaik17/rtl8188eus $ make cut: /etc/redhat-release: No such file or directory Makefile:2133: *** missing separator. Stop.
> > kali@kali:~/amnaik17/rtl8188eus $ sudo touch /etc/redhat-release
> > kali@kali:~/amnaik17/rtl8188eus $ make Makefile:2133: *** missing separator. Stop.
> > kali@kali:~/amnaik17/rtl8188eus $
> > A look at the makefile dictates it is looking to determine the Red Hat version (I am attempting compilation under Kali Linux). Even if the test fails, just having touched the file ought be enough for the makefile to continue upon its way. Is there something I am missing or is this another bug?
> > Thanks!
> 
> /etc/redhat-release can be safely ignored.
> 
> To install drygdryg's fix simply git clone:
> 
> git clone https://github.com/drygdryg/rtl8188eus.git
> 
> Do 'make' and then 'make install'.
> 
> P.S.: Or 'sudo make install' if you do it as non-root. (I don't really use sudo, kinda forgot about it)

I tried this link, it doesn't work, do you have any other solution? Thanks

VGLALALA, 2022-02-15:
> Using TP-Link 802.11ac NIC. I don't know much about WiFi adapters, but I'm trying to get this to work on Kali Linux in VMware. I'm following this https://www.youtube.com/watch?v=VAMP8DqSDjg and then this: https://www.youtube.com/watch?v=tYnjMiTTdms. I'm here because I ran into the same issue with `net/ipx.h: No such file or directory`.
> 
> I followed these instructions after I got that error.
> 
> > To install drygdryg's fix simply git clone:
> > git clone https://github.com/drygdryg/rtl8188eus.git
> > Do 'make' and then 'make install'.
> > P.S.: Or 'sudo make install' if you do it as non-root. (I don't really use sudo, kinda forgot about it)
> 
> My WiFi adapter isn't blinking when I connect it to Kali Linux in VMware. What am I doing wrong?
> 
> **Edit:** I used this amazing github repo which has drivers compatible for my TP-Link Archer T3U WiFi adapter. Problem solved! https://github.com/RinCat/RTL88x2BU-Linux-Driver

I got a TPLink 722N wifi adapter, is taht driver going to work on my wifi adapter?

mmbaguette, 2022-02-15:
@VGLALALA I was just going to ask you for the model. Thanks. Check my post above where I made an edit. I found the drivers that TP-link users need.

ing-santillan, 2022-02-15:
This Driver Is working https://github.com/drygdryg/rtl8188eus.git

ing-santillan, 2022-02-15:
I share this guide https://docs.google.com/document/d/1OBzHjH9kuYtBVASk21EtLstYlhVv92B7cTlClro9ScI/edit?usp=sharing

ing-santillan, 2022-02-15:
I share this guide:
https://docs.google.com/document/d/1OBzHjH9kuYtBVASk21EtLstYlhVv92B7cTlClro9ScI/edit?usp=sharing

winhtikekyaw09, 2022-02-15:
> I share this guide: https://docs.google.com/document/d/1OBzHjH9kuYtBVASk21EtLstYlhVv92B7cTlClro9ScI/edit?usp=sharing
Thanks man. Now the monitor mode is here on my kali linux

ing-santillan, 2022-02-18:
Aré you working in virtual machine?

VGLALALA, 2022-02-22:
> Are you working in virtual machine?

Ya, I'm on a VMware Fusion virtual machine on Mac

Elon-Fask, 2022-02-23:
> I get this error too. Seems like they finally [dropped support](https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/commit/include/net?id=6c9b40844751ea30c72f7a2f92f4d704bc6b2927) from the 5.15 kernel.

I have the latest kernel version and even have the same issue.

I've tried to change the scripte file ( that file containe the error indicated by make command ), but that's show a lot of error. 

the simple solution here as i think, is to change the tp-link 722n V2/3 to V1 or a the batter sulotion is to buy alfa wirless adapter, but unfortunatlly, it's not available in all countries, you'll probably find a fake versions, because the magority of people have no idea about this filed.

that's all about this issue, 
i still believe that there's a sulotion.

mdshouyb908-sudo, 2022-03-07:
┌──(root㉿kali)-[~/rtl8188eus]
└─# make
cut: /etc/redhat-release: No such file or directory
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.16.0-kali1-amd64/build M=/root/rtl8188eus  modules
make[1]: Entering directory '/usr/src/linux-headers-5.16.0-kali1-amd64'
/bin/sh: 1: /usr/src/linux-headers-5.16.0-kali1-common/scripts/pahole-flags.sh: not found
/bin/sh: 1: /usr/src/linux-headers-5.16.0-kali1-common/scripts/pahole-flags.sh: not found
/bin/sh: 1: /usr/src/linux-headers-5.16.0-kali1-common/scripts/pahole-flags.sh: not found
cut: /etc/redhat-release: No such file or directory
  CC [M]  /root/rtl8188eus/core/rtw_cmd.o
  CC [M]  /root/rtl8188eus/core/rtw_security.o
  CC [M]  /root/rtl8188eus/core/rtw_debug.o
  CC [M]  /root/rtl8188eus/core/rtw_io.o
  CC [M]  /root/rtl8188eus/core/rtw_ioctl_query.o
  CC [M]  /root/rtl8188eus/core/rtw_ioctl_set.o
  CC [M]  /root/rtl8188eus/core/rtw_ieee80211.o
  CC [M]  /root/rtl8188eus/core/rtw_mlme.o
  CC [M]  /root/rtl8188eus/core/rtw_mlme_ext.o
  CC [M]  /root/rtl8188eus/core/rtw_mi.o
  CC [M]  /root/rtl8188eus/core/rtw_wlan_util.o
  CC [M]  /root/rtl8188eus/core/rtw_vht.o
  CC [M]  /root/rtl8188eus/core/rtw_pwrctrl.o
  CC [M]  /root/rtl8188eus/core/rtw_rf.o
  CC [M]  /root/rtl8188eus/core/rtw_recv.o
  CC [M]  /root/rtl8188eus/core/rtw_sta_mgt.o
  CC [M]  /root/rtl8188eus/core/rtw_ap.o
  CC [M]  /root/rtl8188eus/core/mesh/rtw_mesh.o
  CC [M]  /root/rtl8188eus/core/mesh/rtw_mesh_pathtbl.o
  CC [M]  /root/rtl8188eus/core/mesh/rtw_mesh_hwmp.o
  CC [M]  /root/rtl8188eus/core/rtw_xmit.o
  CC [M]  /root/rtl8188eus/core/rtw_p2p.o
  CC [M]  /root/rtl8188eus/core/rtw_rson.o
  CC [M]  /root/rtl8188eus/core/rtw_tdls.o
  CC [M]  /root/rtl8188eus/core/rtw_br_ext.o
/root/rtl8188eus/core/rtw_br_ext.c:20:18: fatal error: net/ipx.h: No such file or directory
   20 |         #include <net/ipx.h>
      |                  ^~~~~~~~~~~
compilation terminated.
make[2]: *** [/usr/src/linux-headers-5.16.0-kali1-common/scripts/Makefile.build:292: /root/rtl8188eus/core/rtw_br_ext.o] Error 1
make[1]: *** [/usr/src/linux-headers-5.16.0-kali1-common/Makefile:1870: /root/rtl8188eus] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.16.0-kali1-amd64'
make: *** [Makefile:2062: modules] Error 2

savetheword, 2022-03-13:
> TP-LINK TL-WN725N RTL8188EU working driver for kali linux kernel 5.15....
> 
> https://github.com/drygdryg/rtl8188eus.git

Thank you so mach bro

savetheword, 2022-03-13:
> I get this error too. Seems like they finally [dropped support](https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/commit/include/net?id=6c9b40844751ea30c72f7a2f92f4d704bc6b2927) from the 5.15 kernel.

hi this is support 5.15 kernall you can use this repository
https://github.com/drygdryg/rtl8188eus

savetheword, 2022-03-14:
Guys watch this video its may be solve your problem https://youtu.be/IvBgRmhD5YA

savetheword, 2022-03-14:
> 

https://youtu.be/IvBgRmhD5YA watch this

Elon-Fask, 2022-03-16:
even if you passed the error and the wireless works, 
after turning-off your computer, you'll find out that nothing happend, 
so that's means that every time you need to re-type the " sudo modprob 8188eus " command

Elon-Fask, 2022-03-16:
here is the solution
https://github.com/Elon-Fask/Kali-Linux

closed 2022-03-19 by kimocoder

realFranco, 2022-07-15:
## Disclaimer: 

> Considerig the issue related with Kali-Linux distro I will post the solution that works on Ubuntu 20.04 LTS distro.

The problem that I encounter was the same as the exposed at the issue:

`// Header file "net.ipx.h" not exist, resulting in compilation problems.`

This set of bash scripts help me into the solution:

```
// REF: https://askubuntu.com/questions/1047832/tp-link-tl-wn725n-driver-for-18-04-version

sudo apt-get update
sudo apt-get install -y linux-headers-$(uname -r) build-essential git
git clone https://github.com/lwfinger/rtl8188eu
cd rtl8188eu
make all
sudo make install
sudo insmod 8188eu.ko
```
As you can see at the image, I have the wi-fi adapter up and running.

[image]

DRawson5570, 2024-01-27:
> Several people have asked for instructions, I asked for a link to the repo, and no one has responded. I suspect either no developers wants to "waste his time providing instructions" or perhaps they [...]

https://github.com/aircrack-ng/rtl8188eus.git