aircrack-ng/rtl8188eus #153: won't compile on raspberry pi 3 b running on kali linux

voldmort21, 2022-01-06:
when i try to compile it gives me this 

make ARCH=arm CROSS_COMPILE= -C /lib/modules/5.4.83-Re4son-v7+/build M=/home/kali/rtl8188eus  modules
make[1]: Entering directory '/usr/src/linux-headers-5.4.83-Re4son-v7+'
  CC [M]  /home/kali/rtl8188eus/core/rtw_cmd.o
/tmp/ccMxTiSd.s: Assembler messages:
/tmp/ccMxTiSd.s:1625: Error: selected processor does not support `dmb ish' in ARM mode
/tmp/ccMxTiSd.s:1636: Error: selected processor does not support `dmb ish' in ARM mode
make[2]: *** [scripts/Makefile.build:266: /home/kali/rtl8188eus/core/rtw_cmd.o] Error 1
make[1]: *** [Makefile:1732: /home/kali/rtl8188eus] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.4.83-Re4son-v7+'
make: *** [Makefile:2313: modules] Error 2
                                              
no matter what i do it don't work

bolokanar, 2022-01-06:
Dublicate? https://github.com/aircrack-ng/rtl8188eus/issues/147

Have you tried this suggestion: https://github.com/aircrack-ng/rtl8188eus/issues/147#issuecomment-986655124

> Not a proper solution, but what worked for me was to use gcc-8 to compile instead of the default gcc-11
> On Debian this works by running "make CC=gcc-8" (replace "gcc-8" with the appropriate binary-name for your older gcc).

voldmort21, 2022-01-07:
it didn't work for me because none of gcc's old version available on kali .. only gcc 11 and clang 13 

udo make CC=gcc                               
[sudo] password for kali: 
cut: /etc/redhat-release: No such file or directory
make ARCH=arm CROSS_COMPILE= -C /lib/modules/5.4.83-Re4son-v7+/build M=/home/kali/rtl8188eus  modules
make[1]: Entering directory '/usr/src/linux-headers-5.4.83-Re4son-v7+'
cut: /etc/redhat-release: No such file or directory
  CC [M]  /home/kali/rtl8188eus/core/rtw_cmd.o
/tmp/ccqLpl7s.s: Assembler messages:
/tmp/ccqLpl7s.s:1409: Error: selected processor does not support `dmb ish' in ARM mode
/tmp/ccqLpl7s.s:1424: Error: selected processor does not support `dmb ish' in ARM mode
make[2]: *** [scripts/Makefile.build:266: /home/kali/rtl8188eus/core/rtw_cmd.o] Error 1
make[1]: *** [Makefile:1732: /home/kali/rtl8188eus] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.4.83-Re4son-v7+'
make: *** [Makefile:2062: modules] Error 2

voldmort21, 2022-01-07:
SOLVED USING GCC-10

closed 2022-01-07 by voldmort21