aircrack-ng/rtl8188eus #155: driver dosent get compiled

Usuck-crypto2005, 2022-01-18:
i use latest version of kali bur it still dosent work i am not any developer or expert on kali but if i would get some help i would be apreciated 

cut: /etc/redhat-release: No such file or directory
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.15.0-kali2-amd64/build M=/home/kali/rtl8188eus  modules
make[1]: Entering directory '/usr/src/linux-headers-5.15.0-kali2-amd64'
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: gcc-11 (Debian 11.2.0-12) 11.2.0
  You are using:           gcc-11 (Debian 11.2.0-13) 11.2.0
cut: /etc/redhat-release: No such file or directory
  CC [M]  /home/kali/rtl8188eus/core/rtw_br_ext.o
/home/kali/rtl8188eus/core/rtw_br_ext.c:20:18: fatal error: net/ipx.h: No such file or directory
   20 |         #include <net/ipx.h>
      |                  ^~~~~~~~~~~
compilation terminated.
make[2]: *** [/usr/src/linux-headers-5.15.0-kali2-common/scripts/Makefile.build:282: /home/kali/rtl8188eus/core/rtw_br_ext.o] Error 1
make[1]: *** [/usr/src/linux-headers-5.15.0-kali2-common/Makefile:1892: /home/kali/rtl8188eus] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.15.0-kali2-amd64'
make: *** [Makefile:2062: modules] Error 2

dmijsMe71274, 2022-01-20:
Hello, we have the same problem, i sympathize with you. i destroyed, rebuilt my VM many times to know if it's the same error in kernel 5.15.0-kali2-amd64... even downgraded to 5.10.0-kali9-amd64... but no success. Also tried version 5.3. & 5.7...
this mostly caught my attention:

"The kernel was built by: gcc-11 (Debian 11.2.0-12) 11.2.0
You are using: gcc-11 (Debian 11.2.0-13) 11.2.0
cut: /etc/redhat-release: No such file or directory.."

Hope some good soul knows a work around this. :)

gudarzi, 2022-01-23:
use this: https://github.com/aircrack-ng/rtl8188eus/pull/152

Ashish21276, 2022-01-28:
same problem

chShiraz, 2022-02-05:
I am facing the same issue, hope someone can help us out with this.

megar9876, 2022-02-09:
sudo apt update
sudo apt install bc
sudo rmmod r8188eu.ko
git clone https://github.com/isqad/rtl8188eus.git
cd rtl8188eus
sudo -i
echo "blacklist r8188eu" > "/etc/modprobe.d/realtek.conf"
exit
make
sudo make install
sudo modprobe 8188eu

Works for me finally

shashankx86, 2022-03-04:
use this https://github.com/drygdryg/rtl8188eus

closed 2022-03-19 by kimocoder