aircrack-ng/rtl8188eus #158: build for kernel 5.10.0-xilinx-v2021.1

SajedeNick1999, 2022-02-02:
Hello.
I have linux with kernel 5.10.0-xilinx-v2021.1 builded with petalinux
I have problem in make command. I receive this error:
`cut: /etc/redhat-release: No such file or directory
make ARCH=arm CROSS_COMPILE= -C /lib/modules/5.10.0-xilinx-v2021.1/build M=/home/ubuntu/rtl8188eus  modules
make[1]: Entering directory '/lib/modules/5.10.0-xilinx-v2021.1/build'
cut: /etc/redhat-release: No such file or directory
  CC [M]  /home/ubuntu/rtl8188eus/core/rtw_cmd.o
cc1: error: cannot load plugin ./scripts/gcc-plugins/arm_ssp_per_task_plugin.so
   ./scripts/gcc-plugins/arm_ssp_per_task_plugin.so: cannot open shared object file: No such file or directory
scripts/Makefile.build:279: recipe for target '/home/ubuntu/rtl8188eus/core/rtw_cmd.o' failed
make[2]: *** [/home/ubuntu/rtl8188eus/core/rtw_cmd.o] Error 1
Makefile:1805: recipe for target '/home/ubuntu/rtl8188eus' failed
make[1]: *** [/home/ubuntu/rtl8188eus] Error 2
make[1]: Leaving directory '/lib/modules/5.10.0-xilinx-v2021.1/build'
Makefile:2062: recipe for target 'modules' failed
make: *** [modules] Error 2`
I also tried to add this module to kernel in petalinux project but another error appear.
can any one help me what I have to do?

darkos123, 2022-02-04:
throw your tp-link and buy alpha

SajedeNick1999, 2022-02-06:
> throw your tp-link and buy alpha
@darkos123 
Thank you for your response
But is there any way to fix it without changing device?

KanuX-14, 2022-03-21:
Did you tried using another kernel version? Looks like `arm_ssp_per_task_plugin.so` is not built in your current kernel.