aircrack-ng/rtl8188eus #159: Kernel Version 5.15.0

chShiraz, 2022-02-05:
Not working on Kernel Version 5.15.0 Got following error
/home/kali/rtl8188eus/core/rtw_br_ext.c:20:18: fatal error: net/ipx.h: No such file or directory
20 | #include <net/ipx.h>
| ^~~~~~~~~~~
compilation terminated.
make[2]: *** [/usr/src/linux-headers-5.15.0-kali2-common/scripts/Makefile.build:282: /home/kali/rtl8188eus/core/rtw_br_ext.o] Error 1
make[1]: *** [/usr/src/linux-headers-5.15.0-kali2-common/Makefile:1892: /home/kali/rtl8188eus] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.15.0-kali2-amd64'
make: *** [Makefile:2062: modules] Error 2

slipfan3301, 2022-02-12:
i found the solution -  https://github.com/Munazirul/rtl8188eus

Azazel009, 2022-02-12:
> i found the solution - https://github.com/Munazirul/rtl8188eus

Bruv if I were a girl, I'd literally suck your dick off like a little kid sucking onto his favourite lollipop. Thanks a lot bruv you literally saved my ass and all the other guys' ass who will be visiting this page in the mere future.

gettmure, 2022-03-02:
yo guys, I am gettings this error

```bash
cut: /etc/redhat-release: No such file or directory
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.15.0-kali3-amd64/build M=/home/gettmure/code/drivers/rtl8188eus  modules
make[1]: *** /lib/modules/5.15.0-kali3-amd64/build: No such file or directory.  Stop.
make: *** [Makefile:2062: modules] Error 2
```

solution above does not help. `uname -a` output:

```bash
Linux kali 5.15.0-kali3-amd64 #1 SMP Debian 5.15.15-2kali1 (2022-01-31) x86_64 GNU/Linux
```

any suggestions?

closed 2022-03-19 by kimocoder