aircrack-ng/rtl8188eus #160: Fatal error: net/ipx.h: No such file or directory

mbino94, 2022-02-18:
I was trying to Enable the monitor mode on my TL WN722N Wifi Adapter 
depending on David bombal's tutorial he listed down this Steps
[https://github.com/davidbombal/Kali-Linux/blob/e1d4587029b723db89c26be5068d3b8d2261a8ff/TP-Link%20TL-WN722N%20adapter](url)

and when i run the `sudo make` command, i get this error

`sudo make         
[sudo] password for kali: 
cut: /etc/redhat-release: No such file or directory
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.15.0-kali3-amd64/build M=/home/kali/rtl8188eus  modules
make[1]: Entering directory '/usr/src/linux-headers-5.15.0-kali3-amd64'
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: gcc-11 (Debian 11.2.0-14) 11.2.0
  You are using:           gcc-11 (Debian 11.2.0-16) 11.2.0
cut: /etc/redhat-release: No such file or directory
  CC [M]  /home/kali/rtl8188eus/core/rtw_br_ext.o
/home/kali/rtl8188eus/core/rtw_br_ext.c:20:18: fatal error: net/ipx.h: No such file or directory
   20 |         #include <net/ipx.h>
      |                  ^~~~~~~~~~~
compilation terminated.
make[2]: *** [/usr/src/linux-headers-5.15.0-kali3-common/scripts/Makefile.build:282: /home/kali/rtl8188eus/core/rtw_br_ext.o] Error 1
make[1]: *** [/usr/src/linux-headers-5.15.0-kali3-common/Makefile:1892: /home/kali/rtl8188eus] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.15.0-kali3-amd64'
make: *** [Makefile:2062: modules] Error 2
                                                                `

helmiau, 2022-03-03:
same as mine

shashankx86, 2022-03-04:
@mbino94 @helmiau use this https://github.com/drygdryg/rtl8188eus

AAUUDDII, 2022-03-07:
> @mbino94 @helmiau use this https://github.com/drygdryg/rtl8188eus

thank you :)

kimocoder, 2022-03-19:
Problem resolved. Update driver and build again.

closed 2022-03-19 by kimocoder