aircrack-ng/rtl8188eus #162: I have this error when i do make

SGaiero, 2022-03-06:
cut: /etc/redhat-release: No existe el fichero o el directorio
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.16.0-kali1-amd64/build M=/home/sgaiero/rtl8188eus  modules
make[1]: se entra en el directorio '/usr/src/linux-headers-5.16.0-kali1-amd64'
/bin/sh: 1: /usr/src/linux-headers-5.16.0-kali1-common/scripts/pahole-flags.sh: not found
/bin/sh: 1: /usr/src/linux-headers-5.16.0-kali1-common/scripts/pahole-flags.sh: not found
/bin/sh: 1: /usr/src/linux-headers-5.16.0-kali1-common/scripts/pahole-flags.sh: not found
cut: /etc/redhat-release: No existe el fichero o el directorio
  CC [M]  /home/sgaiero/rtl8188eus/core/rtw_br_ext.o
In file included from /home/sgaiero/rtl8188eus/core/rtw_br_ext.c:20:
/usr/include/netipx/ipx.h:21:10: fatal error: sys/types.h: No existe el fichero o el directorio
   21 | #include <sys/types.h>
      |          ^~~~~~~~~~~~~
compilation terminated.
make[2]: *** [/usr/src/linux-headers-5.16.0-kali1-common/scripts/Makefile.build:292: /home/sgaiero/rtl8188eus/core/rtw_br_ext.o] Error 1
make[1]: *** [/usr/src/linux-headers-5.16.0-kali1-common/Makefile:1870: /home/sgaiero/rtl8188eus] Error 2
make[1]: se sale del directorio '/usr/src/linux-headers-5.16.0-kali1-amd64'
make: *** [Makefile:2062: modules] Error 2

karthik558, 2022-03-13:
same issue here too , some drivers have issue on 5.16+ kernel

karthik558, 2022-03-13:
Maybe this issue cuz of vmware or what !! Idk , someone help if u guys know

drygdryg, 2022-03-15:
Try installing the `libc6-dev` package (taken from [here](https://stackoverflow.com/a/50072645/9310156))

closed 2022-03-19 by kimocoder

MURSHIG, 2022-03-27:
How u solved it

karthik558, 2022-03-27:
> How u solved it

pull latest changes from repo or git clone https://github.com/aircrack-ng/rtl8188eus and do make and make install

karthik558, 2022-03-27:
https://github.com/karthik558/setup_kali_env/tree/main/TP-LINK%20DRIVER for reference

iPurya, 2022-04-29:
@karthik558 not working