aircrack-ng/rtl8188eus #173: 5.15.0-kali3-amd64/build: No such file or directory

evildevill, 2022-03-24:
sudo make
cut: /etc/redhat-release: No such file or directory
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.15.0-kali3-amd64/build M=/home/wasii/rtl8188eus  modules
make[1]: *** /lib/modules/5.15.0-kali3-amd64/build: No such file or directory.  Stop.
make: *** [Makefile:2062: modules] Error 2

ajaygovind, 2022-04-15:
you have to install Linux headers for 5.15 kernel.
run this : `sudo apt-get install linux-headers-5.15.0-kali3-amd64`
after that run sudo make again and sudo make install.

anthonyvw124, 2022-05-10:
well i have the same problem, but when i used you command, it said that it could not find the package:
`└─$ sudo apt-get install linux-headers-5.15.0-kali3-amd64
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
E: Unable to locate package linux-headers-5.15.0-kali3-amd64
E: Couldn't find any package by glob 'linux-headers-5.15.0-kali3-amd64'
E: Couldn't find any package by regex 'linux-headers-5.15.0-kali3-amd64'`

evildevill, 2022-05-24:
@ajaygovind install all the required headers manually then It'll be resolved