aircrack-ng/rtl8188eus #179: Does not build on kali running 5.17

mcars, 2022-06-11:
When attempting to build the module, make fails:

└─$ make      
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.17.0-kali3-amd64/build M=/opt/rtl8188eus  modules
make[1]: Entering directory '/usr/src/linux-headers-5.17.0-kali3-amd64'
  CC [M]  /opt/rtl8188eus/core/rtw_cmd.o
  CC [M]  /opt/rtl8188eus/core/rtw_security.o
  CC [M]  /opt/rtl8188eus/core/rtw_debug.o
  CC [M]  /opt/rtl8188eus/core/rtw_io.o
  CC [M]  /opt/rtl8188eus/core/rtw_ioctl_query.o
  CC [M]  /opt/rtl8188eus/core/rtw_ioctl_set.o
  CC [M]  /opt/rtl8188eus/core/rtw_ieee80211.o
  CC [M]  /opt/rtl8188eus/core/rtw_mlme.o
  CC [M]  /opt/rtl8188eus/core/rtw_mlme_ext.o
/opt/rtl8188eus/core/rtw_mlme_ext.c: In function ‘mgt_dispatcher’:
/opt/rtl8188eus/core/rtw_mlme_ext.c:1838:25: warning: attribute ‘fallthrough’ not preceding a case label or default label
 1838 |                         __attribute__ ((__fallthrough__));
      |                         ^~~~~~~~~~~~~
  CC [M]  /opt/rtl8188eus/core/rtw_mi.o
  CC [M]  /opt/rtl8188eus/core/rtw_wlan_util.o
  CC [M]  /opt/rtl8188eus/core/rtw_vht.o
  CC [M]  /opt/rtl8188eus/core/rtw_pwrctrl.o
  CC [M]  /opt/rtl8188eus/core/rtw_rf.o
  CC [M]  /opt/rtl8188eus/core/rtw_recv.o
  CC [M]  /opt/rtl8188eus/core/rtw_sta_mgt.o
  CC [M]  /opt/rtl8188eus/core/rtw_ap.o
  CC [M]  /opt/rtl8188eus/core/mesh/rtw_mesh.o
  CC [M]  /opt/rtl8188eus/core/mesh/rtw_mesh_pathtbl.o
  CC [M]  /opt/rtl8188eus/core/mesh/rtw_mesh_hwmp.o
  CC [M]  /opt/rtl8188eus/core/rtw_xmit.o
  CC [M]  /opt/rtl8188eus/core/rtw_p2p.o
  CC [M]  /opt/rtl8188eus/core/rtw_rson.o
  CC [M]  /opt/rtl8188eus/core/rtw_tdls.o
  CC [M]  /opt/rtl8188eus/core/rtw_br_ext.o
  CC [M]  /opt/rtl8188eus/core/rtw_iol.o
  CC [M]  /opt/rtl8188eus/core/rtw_sreset.o
  CC [M]  /opt/rtl8188eus/core/rtw_btcoex_wifionly.o
  CC [M]  /opt/rtl8188eus/core/rtw_btcoex.o
  CC [M]  /opt/rtl8188eus/core/rtw_beamforming.o
  CC [M]  /opt/rtl8188eus/core/rtw_odm.o
  CC [M]  /opt/rtl8188eus/core/rtw_rm.o
  CC [M]  /opt/rtl8188eus/core/rtw_rm_fsm.o
  CC [M]  /opt/rtl8188eus/core/efuse/rtw_efuse.o
/opt/rtl8188eus/core/efuse/rtw_efuse.c: In function ‘rtw_efuse_analyze’:
/opt/rtl8188eus/core/efuse/rtw_efuse.c:714:17: warning: this ‘if’ clause does not guard... [-Wmisleading-indentation]
  714 |                 if (i % 16 == 0)
      |                 ^~
In file included from /opt/rtl8188eus/include/drv_types.h:59,
                 from /opt/rtl8188eus/core/efuse/rtw_efuse.c:17:
/opt/rtl8188eus/include/rtw_debug.h:194:9: note: ...this statement, but the latter is misleadingly indented as if it were guarded by the ‘if’
  194 |         do {\
      |         ^~
/opt/rtl8188eus/core/efuse/rtw_efuse.c:716:25: note: in expansion of macro ‘_RTW_PRINT_SEL’
  716 |                         _RTW_PRINT_SEL(RTW_DBGDUMP, "%02X%s"
      |                         ^~~~~~~~~~~~~~
  CC [M]  /opt/rtl8188eus/os_dep/osdep_service.o
/opt/rtl8188eus/os_dep/osdep_service.c: In function ‘thread_exit’:
/opt/rtl8188eus/os_dep/osdep_service.c:1271:9: error: implicit declaration of function ‘complete_and_exit’ [-Werror=implicit-function-declaration]
 1271 |         complete_and_exit(comp, 0);
      |         ^~~~~~~~~~~~~~~~~
/opt/rtl8188eus/os_dep/osdep_service.c: In function ‘rtw_change_ifname’:
/opt/rtl8188eus/os_dep/osdep_service.c:2450:28: warning: passing argument 1 of ‘_rtw_memcpy’ discards ‘const’ qualifier from pointer target type [-Wdiscarded-qualifiers]
 2450 |         _rtw_memcpy(pnetdev->dev_addr, adapter_mac_addr(padapter), ETH_ALEN);
      |                     ~~~~~~~^~~~~~~~~~
/opt/rtl8188eus/os_dep/osdep_service.c:899:24: note: expected ‘void *’ but argument is of type ‘const unsigned char *’
  899 | void _rtw_memcpy(void *dst, const void *src, u32 sz)
      |                  ~~~~~~^~~
cc1: some warnings being treated as errors
make[2]: *** [/usr/src/linux-headers-5.17.0-kali3-common/scripts/Makefile.build:293: /opt/rtl8188eus/os_dep/osdep_service.o] Error 1
make[1]: *** [/usr/src/linux-headers-5.17.0-kali3-common/Makefile:1855: /opt/rtl8188eus] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.17.0-kali3-amd64'
make: *** [Makefile:2058: modules] Error 2
  

dkms-install.sh also failed, with less detailed messages:
└─$ sudo ./dkms-install.sh                     
About to run dkms install steps...
Error! DKMS tree already contains: 8188eu-5.3.9
You cannot add the same module/version combo more than once.
Deprecated feature: REMAKE_INITRD

Building module:
cleaning build area...
'make' -j4 KVER=5.17.0-kali3-amd64 KSRC=/lib/modules/5.17.0-kali3-amd64/build.........(bad exit status: 2)
Error! Bad return status for module build on kernel: 5.17.0-kali3-amd64 (x86_64)
Consult /var/lib/dkms/8188eu/5.3.9/build/make.log for more information.
Deprecated feature: REMAKE_INITRD

Building module:
cleaning build area...
'make' -j4 KVER=5.17.0-kali3-amd64 KSRC=/lib/modules/5.17.0-kali3-amd64/build.........(bad exit status: 2)
Error! Bad return status for module build on kernel: 5.17.0-kali3-amd64 (x86_64)
Consult /var/lib/dkms/8188eu/5.3.9/build/make.log for more information.
Finished running dkms install steps.

i assume "Error! DKMS tree already contains: 8188eu-5.3.9" refers to the modules built for previous kernels.

Batu33TR, 2022-06-12:
Manually add changes from here
https://github.com/aircrack-ng/rtl8188eus/pull/174/commits/13cc38399eb44f234a52dde5f94509e54a09c71f

mcars, 2022-06-14:
Thank you. And hopefully next time I'll remember to check the pull requests tab. >_>

closed 2022-06-14 by mcars