aircrack-ng/rtl8188eus #180: kali 2022.2 (error in sudo make && make install)

imn1ma, 2022-06-14:
`┌──(kali㉿kali)-[~/rtl8188eus]
└─$ sudo make && make install
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.17.0-kali3-amd64/build M=/home/kali/rtl8188eus  modules
make[1]: Entering directory '/usr/src/linux-headers-5.17.0-kali3-amd64'
  CC [M]  /home/kali/rtl8188eus/os_dep/osdep_service.o
/home/kali/rtl8188eus/os_dep/osdep_service.c: In function ‘thread_exit’:
/home/kali/rtl8188eus/os_dep/osdep_service.c:1271:9: error: implicit declaration of function ‘complete_and_exit’ [-Werror=implicit-function-declaration]
 1271 |         complete_and_exit(comp, 0);
      |         ^~~~~~~~~~~~~~~~~
/home/kali/rtl8188eus/os_dep/osdep_service.c: In function ‘rtw_change_ifname’:
/home/kali/rtl8188eus/os_dep/osdep_service.c:2450:28: warning: passing argument 1 of ‘_rtw_memcpy’ discards ‘const’ qualifier from pointer target type [-Wdiscarded-qualifiers]
 2450 |         _rtw_memcpy(pnetdev->dev_addr, adapter_mac_addr(padapter), ETH_ALEN);
      |                     ~~~~~~~^~~~~~~~~~
/home/kali/rtl8188eus/os_dep/osdep_service.c:899:24: note: expected ‘void *’ but argument is of type ‘const unsigned char *’
  899 | void _rtw_memcpy(void *dst, const void *src, u32 sz)
      |                  ~~~~~~^~~
cc1: some warnings being treated as errors
make[2]: *** [/usr/src/linux-headers-5.17.0-kali3-common/scripts/Makefile.build:293: /home/kali/rtl8188eus/os_dep/osdep_service.o] Error 1
make[1]: *** [/usr/src/linux-headers-5.17.0-kali3-common/Makefile:1855: /home/kali/rtl8188eus] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.17.0-kali3-amd64'
make: *** [Makefile:2058: modules] Error 2

`

WolfHash, 2022-06-15:
https://github.com/KanuX-14/rtl8188eus/tree/patch-1

ltcdata, 2022-06-16:
> https://github.com/KanuX-14/rtl8188eus kernel 5.17+ update

I have the same error and this solution did not help. In any case, thanks for the contribution.

CreepyMathematician, 2022-06-16:
I'm having the same problem, anyone got a solution?

CreepyMathematician, 2022-06-17:
> Here is the driver with a full set of patches. Works on kernel version 5.17.x and does not work on 5.18.x. [rtl8188eus.tar.gz](https://github.com/aircrack-ng/rtl8188eus/files/8928556/rtl8188eus.tar.gz)

Let me try this, I have v3 of tp link tl-wn722n,

imn1ma, 2022-06-17:
> Here is the driver with a full set of patches. Works on kernel version 5.17.x and does not work on 5.18.x. [rtl8188eus.tar.gz](https://github.com/aircrack-ng/rtl8188eus/files/8928556/rtl8188eus.tar.gz)

it's working
thank you

Forsaken-stranger, 2022-06-17:
> Here is the driver with a full set of patches. Works on kernel version 5.17.x and does not work on 5.18.x. [rtl8188eus.tar.gz](https://github.com/aircrack-ng/rtl8188eus/files/8928556/rtl8188eus.tar.gz)

Thank you,
may both sides of your pillow be cool

CreepyMathematician, 2022-06-17:
> > Here is the driver with a full set of patches. Works on kernel version 5.17.x and does not work on 5.18.x. [rtl8188eus.tar.gz](https://github.com/aircrack-ng/rtl8188eus/files/8928556/rtl8188eus.tar.gz)
> 
> Thank you, may both sides of your pillow be cool

Tell me how to install the driver when I try to clone from this link it says repository not found. 🙏

Forsaken-stranger, 2022-06-17:
> Tell me how to install the driver when I try to clone from this link it says repository not found. 🙏

Well I'm no expert, but you don't need to clone it, all u need to do is download it
I can show u how did it...
First remove version u have already installed (if u have one) with 
rm -r rtl8188eus

then download the file simply by clicking this link: [rtl8188eus.tar.gz](https://github.com/aircrack-ng/rtl8188eus/files/8928556/rtl8188eus.tar.gz)

unzip it with:
gzip -d Downloads/rtl8188eus.tar.gz 
tar -xf rtl8188eus.tar 
cp Downloads/rtl8188eus rtl8188eus

you can then delete the file in download, for now all you need to do is:
cd rtl8188eus
sudo make  
sudo make install
sudo modprobe 8188eu
sudo reboot 

if it gave u no errors, it should work
hope it helps

CreepyMathematician, 2022-06-17:
> Well I'm no expert, but you don't need to clone it, all u need to do is download it I can show u how did it... First remove version u have already installed (if u have one) with rm -r rtl8188eus [...]

Thanks for the fast reply. 
So the driver was installed correctly with no errors and i managed to change the adapter mode to monitor, and now I can inject packets (airmon ng confirmed it) but wifite and other monitoring application can't detect any interface with monitoring mode on. Even when iwconfig shows that mode is Monitor. 
Any solution will be much appreciated.

imn1ma, 2022-06-17:
> Thanks for the fast reply. So the driver was installed correctly with no errors and i managed to change the adapter mode to monitor, and now I can inject packets (airmon ng confirmed it) but wifite [...]

it was my problem too
but when I reboot the kali it fixed

so reboot and try again

CreepyMathematician, 2022-06-18:
> it was my problem too but when I reboot the kali it fixed
>
> so reboot and try again
After

> it was my problem too but when I reboot the kali it fixed
>
> so reboot and try again

After rebooting several times wifite cannot use the adapter in monitor mode,

CreepyMathematician, 2022-06-18:
This worked for me 
sudo ifconfig wlanx down
sudo airmon-ng check kill
sudo airmon-ng start wlanx

sudo service networking restart
sudo service NetworkManager restart

ltcdata, 2022-06-18:
Thank you guys! It finally worked.

moiz813, 2022-06-24:
@quickreflex you my brother are a hero who has saved countless lives,cant thank you enough

moiz813, 2022-06-24:
@CreepyMathematician iam still getting an error "in function 'is zero mac addr'
And in function 'is broadcast mac address'

moiz813, 2022-06-24:
@quickreflex i think it has smthing to do witt my linux header module can someone please help

matrixangel1, 2022-06-26:
> I am also ready with the driver running on the kernel 5.18.x. Try it. [rtl8188eus_new.tar.gz](https://github.com/aircrack-ng/rtl8188eus/files/8977331/rtl8188eus_new.tar.gz)

Grateful to you ❤

rahul12398, 2022-06-27:
> I am also ready with the driver running on the kernel 5.18.x. Try it. [rtl8188eus_new.tar.gz](https://github.com/aircrack-ng/rtl8188eus/files/8977331/rtl8188eus_new.tar.gz)

Thank you bro 🤩🤩🤩

ECUKmillar5, 2022-07-01:
> I am also ready with the driver running on the kernel 5.18.x. Try it. [rtl8188eus_new.tar.gz](https://github.com/aircrack-ng/rtl8188eus/files/8977331/rtl8188eus_new.tar.gz)

Can confirm this works on 5.18.x - Thanks!

lord47zeus, 2022-07-06:
doesn't work for me, 

ld: /home/kali/Downloads/rtl8188eus/PLM/P/rtl8188eus/os_dep/linux/wifi_regd.o: in function `is_zero_mac_addr':
/home/kali/Downloads/rtl8188eus/PLM/P/rtl8188eus/include/ieee80211.h:1544: multiple definition of `is_zero_mac_addr'; /home/kali/Downloads/rtl8188eus/PLM/P/rtl8188eus/core/rtw_cmd.o:/home/kali/Downloads/rtl8188eus/PLM/P/rtl8188eus/include/ieee80211.h:1544: first defined here
ld: /home/kali/Downloads/rtl8188eus/PLM/P/rtl8188eus/os_dep/linux/rtw_android.o: in function `is_multicast_mac_addr':
/home/kali/Downloads/rtl8188eus/PLM/P/rtl8188eus/include/ieee80211.h:1533: multiple definition of `is_multicast_mac_addr'; /home/kali/Downloads/rtl8188eus/PLM/P/rtl8188eus/core/rtw_cmd.o:/home/kali/Downloads/rtl8188eus/PLM/P/rtl8188eus/include/ieee80211.h:1533: first defined here
ld: /home/kali/Downloads/rtl8188eus/PLM/P/rtl8188eus/os_dep/linux/rtw_android.o: in function `is_broadcast_mac_addr':
/home/kali/Downloads/rtl8188eus/PLM/P/rtl8188eus/include/ieee80211.h:1538: multiple definition of `is_broadcast_mac_addr'; /home/kali/Downloads/rtl8188eus/PLM/P/rtl8188eus/core/rtw_cmd.o:/home/kali/Downloads/rtl8188eus/PLM/P/rtl8188eus/include/ieee80211.h:1538: first defined here
ld: /home/kali/Downloads/rtl8188eus/PLM/P/rtl8188eus/os_dep/linux/rtw_android.o: in function `is_zero_mac_addr':
/home/kali/Downloads/rtl8188eus/PLM/P/rtl8188eus/include/ieee80211.h:1544: multiple definition of `is_zero_mac_addr'; /home/kali/Downloads/rtl8188eus/PLM/P/rtl8188eus/core/rtw_cmd.o:/home/kali/Downloads/rtl8188eus/PLM/P/rtl8188eus/include/ieee80211.h:1544: first defined here

this is the error can you help?

bob69420x, 2022-07-08:
> I am also ready with the driver running on the kernel 5.18.x. Try it. [rtl8188eus_new.tar.gz](https://github.com/aircrack-ng/rtl8188eus/files/8977331/rtl8188eus_new.tar.gz)

You're a legend man<3 
Thank you for taking time & effort to tweak this into a working solution.

**The ONLY thing** that worked. 7/8/2022 - 5.18.x kernel. kali 2022.2

Much love @quickreflex @imn1ma @CreepyMathematician @ltcdata  !!

lord47zeus, 2022-07-08:
now it's working, thank you @quickreflex

cxkeeley, 2022-07-15:
WOW thats do amazing, finally after search for 2 days it works, thanks @quickreflex

GoldDeniel, 2022-08-07:
> I am also ready with the driver running on the kernel 5.18.x. Try it. [rtl8188eus_new.tar.gz](https://github.com/aircrack-ng/rtl8188eus/files/8977331/rtl8188eus_new.tar.gz)

It just works. It took me literally a day to find your comment.
I just simply installed it as it's written for the other versions as well, and it works like charm.

johnfoxii, 2022-08-15:
Outstanding
```

└─$ sudo airmon-ng start wlan0

PHY     Interface       Driver          Chipset

phy0    wlan0           8188eu          TP-Link TL-WN722N v2/v3 [Realtek RTL8188EUS]
                (monitor mode enabled)
                
```
it is a v2 device				
Thank you, quickreflex

black-null, 2022-08-19:
Thank a lot @quickreflex, worked for kernel 5.18.x. kali 2022.2

archinmodi, 2022-08-25:
@WolfHash , it's working fine, thanks a lot. For more than 5 months I'm struggling with the wifi adapter issue.

bob69420x, 2022-10-16:
> Here is the driver with a full set of patches. Works on kernel version 5.17.x and does not work on 5.18.x. [rtl8188eus.tar.gz](https://github.com/aircrack-ng/rtl8188eus/files/8928556/rtl8188eus.tar.gz)

Greetings,
This was working well on 5.17.x not long ago, recently updated to 5.19, so it doesnt work anymore, is there any fix for 5.19?

Best regards

Rajmsr, 2022-11-05:
any one have some solution for kernel 5.19.0

lab37, 2022-11-05:
just do:
apt  instsll  realtek-rtl8188eus-dkms

Ayanokouji-Only, 2023-03-09:
[image]
got this problem after 'make'

Selva-TS-cube, 2023-03-13:
> * wget https://github.com/aircrack-ng/rtl8188eus/files/8977331/rtl8188eus_new.tar.gz
> * tar xvf rtl8188eus_new.tar.gz
> * cd rtl8188eus_new
> * sudo make
> * sudo make install
> * sudo modprobe 8188eu
> * sudo reboot

Vasudev109, 2023-03-17:
Today kali linux comes with new kernel update with vertion 6.1.0-kali5-amd64 .
 I installed kali yesterday and I am trying to install tplink wn722n monitor mode supporting driver which were uploaded 3 days ago BY [Selva-TS-cube](https://github.com/Selva-TS-cube) for vertion 6.0.0-kali3  which is giving me COMPILATION ERROR...

after (sudo make) 

[image]

gglluukk, 2023-03-17:
try this for 6.1/6.2 kernels (still in pull requests, not merged yet):
https://github.com/gglluukk/rtl8188eus

hubgit115, 2023-04-17:
plz give some solution of above images

Vasudev109, 2023-04-17:
😁Thanks bro my issue is already solved
From your help via last update😄

hubgit115, 2023-04-17:
> grinThanks bro my issue is already solved From your help via last updatesmile
> […](#)

plz share it to me
i also want the solution . I am trying it for many days but  I am not able to solve it. I have watched many youtube videos but i got failed . plz share

Termin7, 2023-05-01:
> `┌──(kali㉿kali)-[~/rtl8188eus] └─$ sudo make && make install make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.17.0-kali3-amd64/build M=/home/kali/rtl8188eus modules make[1]: Entering directory [...]

Hi bro, I was getting a different but similar kind of error . It is not solving with current driver. Can anyone help me to sort out this issue. I am pasting the error below.
Kali VM version : 2022

Error : rtl8188eus/os_dep/osdep_service.c:2582:16: error: implicit declaration of function ‘prandom_u32’; did you mean ‘prandom_u32_max’? [-Werror=implicit-function-declaration]
2582 | return prandom_u32();

Termin7, 2023-05-01:
Hi bro please tell me the solution am also facing this issue

Termin7, 2023-05-01:
> try this for 6.1/6.2 kernels (still in pull requests, not merged yet): https://github.com/gglluukk/rtl8188eus

Thanks bro. Its working

957070, 2023-07-05:
> please solve this new error for new kali linux kernal ... [image] Got this error after sudo make command I AM WAITING BROTHERS ......

Akhilesh-prob, 2024-07-24:
> Well I'm no expert, but you don't need to clone it, all u need to do is download it I can show u how did it... First remove version u have already installed (if u have one) with rm -r rtl8188eus [...]

Yo dude can please please help by telling how to apply all these things I am literally noob at this and want to enable monitor mode and packet injection and also I have same adaptor please help me like someone spoon feeding a baby

Akhilesh-prob, 2024-07-24:
@quickreflex didn't worke for me 
@Forsaken-stranger

Akhilesh-prob, 2024-07-24:
> [image] got this problem after 'make'

i also got same issue u got any solution ? 
@quickreflex @Forsaken-stranger