aircrack-ng/rtl8188eus #184: command failed: Operation not permitted (-1)

m-desouki, 2022-07-01:
┌──(root💀kali)-[~]
└─# iw dev wlan0 scan
command failed: Operation not permitted (-1)

`Bus 003 Device 004: ID 2357:010c TP-Link TL-WN722N v2/v3 [Realtek RTL8188EUS]
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x2357 TP-Link
  idProduct          0x010c TL-WN722N v2/v3 [Realtek RTL8188EUS]
  bcdDevice            0.00
  iManufacturer           1 Realtek
  iProduct                2 802.11n NIC
  iSerial                 3 00E04C0001
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0027
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           3
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass    255 Vendor Specific Subclass
      bInterfaceProtocol    255 Vendor Specific Protocol
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x03  EP 3 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
Device Qualifier (for other device speed):
  bLength                10
  bDescriptorType         6
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  bNumConfigurations      1
Device Status:     0x0002
  (Bus Powered)
  Remote Wakeup Enabled
`

quickreflex, 2022-07-03:
Use the alternate command:
iwlist wlan0 scanning

m-desouki, 2022-07-03:
> 
┌──(root💀kali)-[~]
└─# iwlist wlan0 scanning
wlan0     No scan results

quickreflex, 2022-07-03:
You can't scan in monitor mode, so switch to mode first managed:
iw dev wlan0 set type managed
iwlist wlan0 scanning

Waqar-Wani, 2023-01-31:
┌──(root㉿kali)-[~]
└─# iwconfig                 
lo        no wireless extensions.

wlan0     unassociated  ESSID:""  Nickname:"<WIFI@REALTEK>"
          Mode:Managed  Frequency=2.412 GHz  Access Point: Not-Associated   
          Sensitivity:0/0  
          Retry:off   RTS thr:off   Fragment thr:off
          Encryption key:off
          Power Management:off
          Link Quality:0  Signal level:0  Noise level:0
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:0   Missed beacon:0

                                                                                                                                                                      
┌──(root㉿kali)-[~]
└─# ip link set wlan0 down      
                                                                                                                                                                      
┌──(root㉿kali)-[~]
└─# iw wlan0 set monitor control
command failed: Operation not permitted (-1)
                                                                                                                                                                      
┌──(root㉿kali)-[~]
└─# iwlist wlan0 scanning       
wlan0     No scan results

                                                                                                                                                                      
┌──(root㉿kali)-[~]
└─# iw dev wlan0 set type managed 
                                                                                                                                                                      
┌──(root㉿kali)-[~]
└─# iwlist wlan0 scanning        
wlan0     No scan results

Milor123, 2024-09-02:
I have this problem in docker, how can i resolver it? 
i am using it:
```
version: '3'
services:
  MrKali:
    image: kalilinux/kali-rolling
    container_name: kali-net-v3
    network_mode: host
    cap_add:
      - NET_ADMIN
      - NET_RAW
      - SYS_ADMIN
    devices:
      - "/dev/net/tun:/dev/net/tun"
      - "/dev/rfkill:/dev/rfkill"
    privileged: true
    stdin_open: true  # equivalente a -i
    tty: true         # equivalente a -t
    pid: host
    security_opt:
      - apparmor:unconfined
      - seccomp:unconfined
    #command: /bin/bash -c "apt-get update && apt-get install -y ca-certificates && apt-get install -y pciutils kali-linux-headless"
    #command: tail -f /dev/null  # Mantiene el contenedor en ejecución

```