aircrack-ng/rtl8188eus #185: Kernel error & crashes after waking up from suspend & re-loading the driver: shift-out-of-bounds in phydm_phystatus.c. shift exponent 63 is too large for 32-bit type 'int'

navid-zamani, 2022-07-01:
Since the last update I got serveral system hangs (display output normal, audio loops, input does not respond), specifically not long after coming back out of suspend. Also, the wifi (via the usb stick using the 8188eu chip) seems very slow, and after the forced shutdown and reboot, is very quick again.
And I found this in dmesg, when re-loading the driver (`modprobe -r 8188eu; modprobe 8188eu`):

```
[  315.655229] ================================================================================
[  315.655301] UBSAN: shift-out-of-bounds in /var/lib/dkms/8188eu/5.3.9/build/hal/phydm/phydm_phystatus.c:1717:69
[  315.655318] shift exponent 63 is too large for 32-bit type 'int'
[  315.655329] CPU: 1 PID: 0 Comm: swapper/1 Tainted: G           OE     5.15.0-33-generic #34~20.04.1-Ubuntu
[  315.655341] Hardware name: LENOVO 4293AF4/4293AF4, BIOS 8DET76WW (1.46 ) 06/21/2018
[  315.655347] Call Trace:
[  315.655354]  <IRQ>
[  315.655361]  dump_stack_lvl+0x4a/0x5f
[  315.655382]  dump_stack+0x10/0x12
[  315.655391]  ubsan_epilogue+0x9/0x45
[  315.655400]  __ubsan_handle_shift_out_of_bounds.cold+0x61/0x10e
[  315.655416]  ? phydm_parsing_cfo+0x2d/0xd9 [8188eu]
[  315.655573]  phydm_process_rssi_for_dm+0x1be/0x2fb [8188eu]
[  315.655714]  odm_phy_status_query+0xee/0x14b [8188eu]
[  315.655852]  rx_query_phy_status+0x21e/0x3ac [8188eu]
[  315.656006]  pre_recv_entry+0x2f/0x6f [8188eu]
[  315.656158]  recvbuf2recvframe+0xa6/0x293 [8188eu]
[  315.656303]  usb_recv_tasklet+0x61/0xd6 [8188eu]
[  315.656470]  tasklet_action_common.isra.0+0xea/0xf0
[  315.656482]  tasklet_action+0x22/0x30
[  315.656489]  __do_softirq+0xdd/0x2ea
[  315.656501]  ? ioapic_ir_ack_level+0x22/0x30
[  315.656513]  irq_exit_rcu+0x79/0xa0
[  315.656520]  common_interrupt+0x8a/0xa0
[  315.656528]  </IRQ>
[  315.656530]  <TASK>
[  315.656533]  asm_common_interrupt+0x1e/0x40
[  315.656543] RIP: 0010:cpuidle_enter_state+0xd9/0x620
[  315.656555] Code: 3d b4 44 a1 6a e8 37 b2 6f ff 49 89 c7 0f 1f 44 00 00 31 ff e8 e8 bd 6f ff 80 7d d0 00 0f 85 5a 01 00 00 fb 66 0f 1f 44 00 00 <45> 85 f6 0f 88 66 01 00 00 4d 63 ee 49 83 fd 09 0f 87 08 04 00 00
[  315.656562] RSP: 0018:ffffb134800bfe18 EFLAGS: 00000246
[  315.656571] RAX: ffff908c52270f00 RBX: ffffd1347fc5ce00 RCX: 0000000000000000
[  315.656576] RDX: 0000000000000027 RSI: 00000000335dc76b RDI: 0000000000000000
[  315.656580] RBP: ffffb134800bfe68 R08: 000000497e83781b R09: 00000000000230c0
[  315.656585] R10: 0000000000000001 R11: 000000000000000f R12: ffffffff96cd2780
[  315.656589] R13: 0000000000000005 R14: 0000000000000005 R15: 000000497e83781b
[  315.656597]  ? cpuidle_enter_state+0xc8/0x620
[  315.656607]  cpuidle_enter+0x2e/0x40
[  315.656614]  call_cpuidle+0x23/0x40
[  315.656626]  do_idle+0x1f0/0x270
[  315.656637]  cpu_startup_entry+0x20/0x30
[  315.656648]  start_secondary+0x12a/0x180
[  315.656655]  secondary_startup_64_no_verify+0xc2/0xcb
[  315.656669]  </TASK>
[  315.656675] ================================================================================
[  317.381502] ================================================================================
[  317.381566] UBSAN: shift-out-of-bounds in /var/lib/dkms/8188eu/5.3.9/build/hal/phydm/phydm_phystatus.c:1754:67
[  317.381579] shift exponent 63 is too large for 32-bit type 'int'
[  317.381588] CPU: 1 PID: 0 Comm: swapper/1 Tainted: G           OE     5.15.0-33-generic #34~20.04.1-Ubuntu
[  317.381597] Hardware name: LENOVO 4293AF4/4293AF4, BIOS 8DET76WW (1.46 ) 06/21/2018
[  317.381601] Call Trace:
[  317.381607]  <IRQ>
[  317.381614]  dump_stack_lvl+0x4a/0x5f
[  317.381631]  dump_stack+0x10/0x12
[  317.381637]  ubsan_epilogue+0x9/0x45
[  317.381645]  __ubsan_handle_shift_out_of_bounds.cold+0x61/0x10e
[  317.381653]  ? __alloc_skb+0x92/0x1d0
[  317.381667]  phydm_process_rssi_for_dm+0x206/0x2fb [8188eu]
[  317.381797]  odm_phy_status_query+0xee/0x14b [8188eu]
[  317.381909]  rx_query_phy_status+0x21e/0x3ac [8188eu]
[  317.382033]  pre_recv_entry+0x2f/0x6f [8188eu]
[  317.382155]  recvbuf2recvframe+0xa6/0x293 [8188eu]
[  317.382272]  usb_recv_tasklet+0x61/0xd6 [8188eu]
[  317.382407]  tasklet_action_common.isra.0+0xea/0xf0
[  317.382416]  tasklet_action+0x22/0x30
[  317.382422]  __do_softirq+0xdd/0x2ea
[  317.382432]  ? ioapic_ir_ack_level+0x22/0x30
[  317.382442]  irq_exit_rcu+0x79/0xa0
[  317.382447]  common_interrupt+0x8a/0xa0
[  317.382453]  </IRQ>
[  317.382455]  <TASK>
[  317.382458]  asm_common_interrupt+0x1e/0x40
[  317.382466] RIP: 0010:cpuidle_enter_state+0xd9/0x620
[  317.382475] Code: 3d b4 44 a1 6a e8 37 b2 6f ff 49 89 c7 0f 1f 44 00 00 31 ff e8 e8 bd 6f ff 80 7d d0 00 0f 85 5a 01 00 00 fb 66 0f 1f 44 00 00 <45> 85 f6 0f 88 66 01 00 00 4d 63 ee 49 83 fd 09 0f 87 08 04 00 00
[  317.382481] RSP: 0018:ffffb134800bfe18 EFLAGS: 00000246
[  317.382488] RAX: ffff908c52270f00 RBX: ffffd1347fc5ce00 RCX: 0000000000000000
[  317.382493] RDX: 0000000000000027 RSI: 00000000335dc76b RDI: 0000000000000000
[  317.382496] RBP: ffffb134800bfe68 R08: 00000049e5684c81 R09: 00000049e5ac7daf
[  317.382500] R10: 00000049e5326baf R11: 0000000000000006 R12: ffffffff96cd2780
[  317.382503] R13: 0000000000000005 R14: 0000000000000005 R15: 00000049e5684c81
[  317.382509]  ? cpuidle_enter_state+0xc8/0x620
[  317.382517]  cpuidle_enter+0x2e/0x40
[  317.382524]  call_cpuidle+0x23/0x40
[  317.382533]  do_idle+0x1f0/0x270
[  317.382542]  cpu_startup_entry+0x20/0x30
[  317.382551]  start_secondary+0x12a/0x180
[  317.382557]  secondary_startup_64_no_verify+0xc2/0xcb
[  317.382568]  </TASK>
[  317.382573] ================================================================================

```
I don’t know if the crashes and slowness after wake up and this error message are related, but of course this error is not how things should be, and maybe it solves other things in the process. 

It seem that the error message contains everything necessary, but if you need anything more, I’m happy to provide it.

quickreflex, 2022-07-02:
Add this line to sysctl.conf and reboot so it doesn't go into mode suspend:
usbcore.autosuspend = -1
Or upgrade your current kernel from the first version 5.15.0 to the current 5.15.51 or later.

navid-zamani, 2022-07-02:
Since this is a laptop, not going to suspend is not an option, as the battery would drain too much.
I am surprised with how old the kernels in Mint are. So I will go the route of updating to the latest, using [Ubuntu Mainline Kernel Installer](https://github.com/bkw777/mainline) (since UKUU has committed suicide).
Thank you, and sorry for bothering you with an oversight (old kernel) on my part. :)

closed 2022-07-02 by navid-zamani