aircrack-ng/rtl8188eus #186: the usb adapter it's not recognized now

Lparis201, 2022-07-10:
hi, i follow all the instructions, but when i do the reboot after the make and make install, kali did not recognize the wifi adapter, what can i do? 
i am in kali v5.15.18

LuccaDale, 2022-07-16:
Hi,
I'm having the same issue. I been using this driver for a while but today my pc froze and I had to restart, after that ubuntu won't recognize the wifi adapter.

I'm in Ubuntu 22.04, and my adapter is TP-Link TL-WN722N v2/v3 [Realtek RTL8188EUS]

quickreflex, 2022-07-16:
Give me the output of the dmesg command. There might be an error in there somewhere.

MAKOSTO7, 2022-07-16:
me too same problem

quickreflex, 2022-07-16:
What is your platform, is it powerpc (PPC)?

LuccaDale, 2022-07-16:
> Give me the output of the dmesg command. There might be an error in there somewhere.

Here is the relevant output for dmesg after disconnecting and reconnecting the device:
```
[59235.555857] usb 1-3: USB disconnect, device number 7
[59246.459705] usb 1-3: new high-speed USB device number 10 using xhci_hcd
[59246.628196] usb 1-3: New USB device found, idVendor=2357, idProduct=010c, bcdDevice= 0.00
[59246.628203] usb 1-3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[59246.628205] usb 1-3: Product: 802.11n NIC
[59246.628207] usb 1-3: Manufacturer: Realtek
[59246.628208] usb 1-3: SerialNumber: 00E04C0001
[59246.648750] 8188eu: disagrees about version of symbol module_layout
[59667.783986] usb 1-3: USB disconnect, device number 10
[59680.215174] usb 1-3: new high-speed USB device number 11 using xhci_hcd
[59680.383935] usb 1-3: New USB device found, idVendor=2357, idProduct=010c, bcdDevice= 0.00
[59680.383943] usb 1-3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[59680.383945] usb 1-3: Product: 802.11n NIC
[59680.383946] usb 1-3: Manufacturer: Realtek
[59680.383948] usb 1-3: SerialNumber: 00E04C0001
[59680.405715] 8188eu: disagrees about version of symbol module_layout
```

> What is your platform, is it powerpc (PPC)?

I'm using a desktop computer with an x64 processor from AMD, hope that answers you question.

LuccaDale, 2022-07-17:
Okey, so after googling the "8188eu: disagrees about version of symbol module_layout" message from dmesg I saw other similar issues being attributed to a mismatch between the kernel and the compilation headers of the driver. 

Since the driver was working before, I downgraded my kernel from 5.15.0-41 to 5.15.0-40 and now its working.
I guess the real issue here is compatibility with the new kernel version.

Thank you all for your help!

quickreflex, 2022-07-17:
This issue is discussed here - [kernel, headers, configuration and compiler dependencies](https://github.com/lwfinger/rtl8188eu/issues/102).

KanuX-14, 2022-07-21:
> I downgraded my kernel from 5.15.0-41 to 5.15.0-40 and now its working.

This should be investigated more deeply. The reason that we keep this repository alive is to give compatibility and optimizations to all users.