aircrack-ng/rtl8188eus #187: How can i solve this issue

System-Error44, 2022-07-10:
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.18.0-kali5-amd64/build M=/home/hidden/Desktop/tools/rtl8188eus  modules
make[1]: Entering directory '/usr/src/linux-headers-5.18.0-kali5-amd64'
  CC [M]  /home/hidden/Desktop/tools/rtl8188eus/os_dep/osdep_service.o
/home/hidden/Desktop/tools/rtl8188eus/os_dep/osdep_service.c: In function ‘thread_exit’:
/home/hidden/Desktop/tools/rtl8188eus/os_dep/osdep_service.c:1271:9: error: implicit declaration of function ‘complete_and_exit’ [-Werror=implicit-function-declaration]
 1271 |         complete_and_exit(comp, 0);
      |         ^~~~~~~~~~~~~~~~~
/home/hidden/Desktop/tools/rtl8188eus/os_dep/osdep_service.c: In function ‘rtw_change_ifname’:
/home/hidden/Desktop/tools/rtl8188eus/os_dep/osdep_service.c:2450:28: warning: passing argument 1 of ‘_rtw_memcpy’ discards ‘const’ qualifier from pointer target type [-Wdiscarded-qualifiers]
 2450 |         _rtw_memcpy(pnetdev->dev_addr, adapter_mac_addr(padapter), ETH_ALEN);
      |                     ~~~~~~~^~~~~~~~~~
/home/hidden/Desktop/tools/rtl8188eus/os_dep/osdep_service.c:899:24: note: expected ‘void *’ but argument is of type ‘const unsigned char *’
  899 | void _rtw_memcpy(void *dst, const void *src, u32 sz)
      |                  ~~~~~~^~~
cc1: some warnings being treated as errors
make[2]: *** [/usr/src/linux-headers-5.18.0-kali5-common/scripts/Makefile.build:293: /home/hidden/Desktop/tools/rtl8188eus/os_dep/osdep_service.o] Error 1
make[1]: *** [/usr/src/linux-headers-5.18.0-kali5-common/Makefile:1858: /home/hidden/Desktop/tools/rtl8188eus] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.18.0-kali5-amd64'
make: *** [Makefile:2058: modules] Error 2

quickreflex, 2022-07-13:
There is a solution to the problem. Read the other issues below where the patched driver file is.

flash3636, 2022-07-17:
hi sir i have error on compiling please help me

quickreflex, 2022-07-17:
Copy the output of the make command and paste it here or attach it as a text file.

flash3636, 2022-07-17:
[image]
Please help me

quickreflex, 2022-07-17:
It works, but you have to use my driver archive where all the bugs are fixed and you need to install kernel headers. Run the following sequence of commands in an empty directory:

wget [https://github.com/aircrack-ng/rtl8188eus/files/8977331/rtl8188eus_new.tar.gz](https://github.com/aircrack-ng/rtl8188eus/files/8977331/rtl8188eus_new.tar.gz)
tar xvf rtl8188eus_new.tar.gz
cd rtl8188eus_new
sudo make
sudo make install
sudo modprobe 8188eu
sudo reboot

flash3636, 2022-07-17:
THANK YOU BRO..... FOR YOUR HELP THANK YOU 
THANK YOU VERY VERY 
HAVE FRIENDS....?

flash3636, 2022-07-17:
I AM BEGINNER MY AGE IS 17th TO 18th 
HAVE FRIENDS....?

quickreflex, 2022-07-19:
You have not installed the driver. You must login and use the root user account to run the "make install" command without error.

System-Error44, 2022-07-19:
> It works, but you have to use my driver archive where all the bugs are fixed and you need to install kernel headers. Run the following sequence of commands in an empty directory:
>
> wget [...]

Its works 😃

KanuX-14, 2022-07-21:
Maybe it is a feature since kernel 5.19 did not show any error messages stated at my commit:
https://github.com/KanuX-14/rtl8188eus/commit/0f1905259ec9b85fd1453be3abd322ff543e12bc
It appears to be that from kernel 5.18 the `extern` function breaks some declarations, making it have definitions in multiple files.

quickreflex, 2022-07-21:
Here's a quote from a kernel developer letter:

With compilers which follow the C99 standard (like modern versions of
gcc and clang), "extern inline" does the opposite thing from older
versions of gcc (emits code for an externally linkable version of the
inline function).

"static inline" does the intended behavior in all cases instead.

KanuX-14, 2022-07-21:
So it is really the newer C version that they are implementing.
This will break a lot of codes.

Fokz1, 2022-08-06:
> It works, but you have to use my driver archive where all the bugs are fixed and you need to install kernel headers. Run the following sequence of commands in an empty directory:
>
> wget [...]

ty so much, i love you

AbdullahTanveer, 2022-08-15:
> It works, but you have to use my driver archive where all the bugs are fixed and you need to install kernel headers. Run the following sequence of commands in an empty directory:
>
> wget [...]

does that work on k5.18 on kali linux 2022.2?

quickreflex, 2022-08-16:
Yes, it works.

AbdullahTanveer, 2022-08-18:
> Yes, it works.

Thank you so much, brother
it worked!!!

MrRezax85, 2022-10-26:
> It works, but you have to use my driver archive where all the bugs are fixed and you need to install kernel headers. Run the following sequence of commands in an empty directory:
>
> wget [...]

Please help me
[image]

KanuX-14, 2022-10-26:
@MrRezax85 Use [this](https://github.com/KanuX-14/rtl8188eus) instead.

MrRezax85, 2022-10-29:
> @MrRezax85 Use [this](https://github.com/KanuX-14/rtl8188eus) instead.

thank you bro 
this work

Lucksinia, 2022-11-13:
Hi, I have a question. If I can't use wget (because I don't have Ethernet or WiFi connections, obviously...)
Where do I need to unzip/untar archive?
/lib/modules/3.4.39-01-lobo/kernel/drivers?
Because make is like:

make [1]: *** /lib/modules/3.4.39-1-lobo/build: no such file or directory. Stop.
Makefile:2061: recipe for target 'modules' failed
make: [modules] Error 2

But build file IS here, I can clearly see it.
Can you help me? (Posting here, because make error 2. If I need to open a new issue, I will)
Thank you in advance.

KanuX-14, 2022-11-13:
@Lucksinia
You **need** at least the kernel source in order to build the kernel modules.

If you don't have internet connected to your device, get your kernel source code from your phone or a public wifi.
Send it to your device, and then compile it using your already installed configuration file under `/boot/`.

Do a `modules_install` and a `headers_install` and you are good to go. It should install the build driver.

Lucksinia, 2022-11-15:
Okay... I know that will be idiotic question, but I got no Idea how to do it after messing with it for quite a while.

KanuX-14, 2022-11-18:
The configuration file is under `/boot/`. The reference of the current directory is `./boot/`.<br>
It is needed to copy it to the current kernel repository and rename it to `.config`. Example:
```sh
cp -v /boot/config-x.y.z .config
```

Sorry for not being clear of it, as you are not familiar with kernel building.<br>
You need to make the files in order to install:
```sh
make olddefconfig
make modules
make headers
```

And then install it **as root**:
```sh
sudo make install modules_install
sudo make install headers_install
sudo update-initramfs -u
```
If `sudo` is not installed, it may be needed to use `su` and do from the 'make' part. Or use your own installed super-user manager.

These instructions are for Debian-based or itself. Use [this script](https://github.com/KanuX-14/buildme/blob/main/buildme.sh) as reference.