aircrack-ng/rtl8188eus #194: TL-WN722N V3

Mian-Nouman, 2022-08-02:
cc1: some warnings being treated as errors
make[2]: *** [/usr/src/linux-headers-5.18.0-kali5-common/scripts/Makefile.build:293: /home/mian/rtl8188eus/os_dep/osdep_service.o] Error 1
make[1]: *** [/usr/src/linux-headers-5.18.0-kali5-common/Makefile:1858: /home/mian/rtl8188eus] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.18.0-kali5-amd64'
make: *** [Makefile:2058: modules] Error 2

quickreflex, 2022-08-03:
Does this errors happen with this driver?
[https://github.com/aircrack-ng/rtl8188eus/files/8977331/rtl8188eus_new.tar.gz](https://github.com/aircrack-ng/rtl8188eus/files/8977331/rtl8188eus_new.tar.gz)

Eweeeee, 2022-08-04:
> Does this errors happen with this driver? https://github.com/aircrack-ng/rtl8188eus/files/8977331/rtl8188eus_new.tar.gz

this works for me but if i reboot the machine, the adapter doesn't work and i need every time to reinstall the drivers.
someone help me?

abdullayevrajabboy756, 2023-03-26:
/home/abdullayev/rtl8188eus_new/os_dep/osdep_service.c: In function ‘rtw_random32’:
/home/abdullayev/rtl8188eus_new/os_dep/osdep_service.c:2582:16: error: implicit declaration of function ‘prandom_u32’; did you mean ‘prandom_u32_max’? [-Werror=implicit-function-declaration]
 2582 |         return prandom_u32();
      |                ^~~~~~~~~~~
      |                prandom_u32_max
cc1: some warnings being treated as errors
make[2]: *** [/usr/src/linux-headers-6.1.0-kali5-common/scripts/Makefile.build:255: /home/abdullayev/rtl8188eus_new/os_dep/osdep_service.o] Error 1
make[1]: *** [/usr/src/linux-headers-6.1.0-kali5-common/Makefile:2030: /home/abdullayev/rtl8188eus_new] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-6.1.0-kali5-amd64'
make: *** [Makefile:2059: modules] Error 2

abdullayevrajabboy756, 2023-03-26:
i have this problem
please help me

gglluukk, 2023-03-26:
try this for 6.1/6.2 kernels (still in pull requests, not merged yet):
https://github.com/gglluukk/rtl8188eus