aircrack-ng/rtl8188eus #199: Does it support android 12?

shashankx86, 2022-09-01:


drygdryg, 2022-09-07:
The driver depends on the kernel version. Android 12 can run on a wide range of kernel versions, so you need to find out the Linux kernel version.

NickTheNeko2002, 2022-09-18:
> _No description provided._

Download Termux and write in it `uname -a` so we can give you a straight answer.

shashankx86, 2022-09-19:
> > _No description provided._
> 
> Download Termux and write in it `uname -a` and send us the info so we can give you a straight answer.

### Kernel version
4.19.157-perf-g3eee292b027c

NickTheNeko2002, 2022-09-24:
> > > _No description provided._
> > 
> > 
> > Download Termux and write in it `uname -a` and send us the info so we can give you a straight answer.
> 
> ### Kernel version
> 4.19.157-perf-g3eee292b027c

i'd say it's a hit or miss with that one to be honest, if you try and dig around the forks of this driver there might be one that has support for 4.19.xxx
tho you should try to see by yourself and compile and run the driver tho the real question is does the driver support ARM based CPU's

shashankx86, 2022-09-25:
> i'd say it's a hit or miss with that one to be honest, if you try and dig around the forks of this driver there might be one that has support for 4.19.xxx tho you should try to see by yourself and [...]

Thnx