aircrack-ng/rtl8188eus #204: Does not Work on 5.19.9-arch1-1 kernel

JuanVega315, 2022-09-17:
The driver doesn't work with arch linux (kernel version 5.19.9 arch1-1)

Please add support for 5.19.9-arch1-1 kernel 
I would be thankful!!!

NickTheNeko2002, 2022-09-18:
> The driver doesn't work with arch linux (kernel version 5.19.9 arch1-1)
> 
> Please add support for 5.19.9-arch1-1 kernel I would be thankful!!!

just use an older version of the linux kernel, you can use multiple kernels
 > if on Linux Arch Do the(You'll need an AUR helper so yeah, i use yay tho paru works just as well) Folowing:
  > yay -Syu
  > yay -S linux-lts linux-lts-headers
  > Restart, Just to be on the safe side
  > yay -S downgrade
  > sudo downgrade linux-lts
  > Pick a version like 5.10 or 1.15 check this image to decide what you like kernel you like
[image]

  > Restart again and pick the LTS version that was downgraded
  > Once done, re-try installing the driver
  > You should be done and ready to use the driver+USB
if this doesn't work then reply to me and give me feedback

JuanVega315, 2022-09-22:
I tried downgrading, I couldn't make it to appear in grub the downgraded kernel version, I will continue trying

NickTheNeko2002, 2022-09-24:
> Thanks! It worked!!! grinsmilesmile

if this doesn't work try this one with the latest Linux ZEN kernel

`sudo pacman -S linux-zen`
After that do `sudo grub-mkconfig -o /boot/grub/grub.cfg`
and restart
then once the restart is done and you went and are using Linux-zen
do the following
 `git clone https://github.com/kimocoder/realtek_rtwifi`
 then do `nano Makefile`
 change the very first line of code from `obj-$(CONFIG_RTL8XXXU) += rtwifi.o` to `obj-m += rtwifi.o`
 then do `CTRL+S` and `CTRL+C`
 once finished do `sudo make && make install`

NickTheNeko2002, 2022-09-24:
for future problems feel free to write me

KanuX-14, 2022-10-03:
[Try this repository](https://github.com/KanuX-14/rtl8188eus). There are some among the [PRs](https://github.com/aircrack-ng/rtl8188eus/pulls).

TheGr3atJosh, 2022-12-03:
I tried all three steps mentioned in this thread but none worked for me. I'm using an ASUSTek Computer, Inc. Realtek 8188EUS [USB-N10 Nano] and no matter what I do, the driver always shows up using lsmod but the device never shows up using ifconfig even though it does show up using lsusb.