aircrack-ng/rtl8188eus #206: Can't Compile gcc error

NickTheNeko2002, 2022-09-18:
when i do `sudo make && make install` i get the following
> `make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.10.90-1-lts/build M=/home/nick/rtl8188eus  modules`
> `make[1]: Entering directory '/usr/lib/modules/5.10.90-1-lts/build'`
> `  CC [M]  /home/nick/rtl8188eus/core/rtw_cmd.o`
> `cc1: error: incompatible gcc/plugin versions`
> `cc1: error: failed to initialize plugin ./scripts/gcc-plugins/structleak_plugin.so`
> `make[2]: *** [scripts/Makefile.build:280: /home/nick/rtl8188eus/core/rtw_cmd.o] Error 1`
> `make[1]: *** [Makefile:1822: /home/nick/rtl8188eus] Error 2`
> `make[1]: Leaving directory '/usr/lib/modules/5.10.90-1-lts/build'`
> `make: *** [Makefile:2058: modules] Error 2`

OS Info:

[image]

komradehieu, 2022-11-19:
Yeah i got the exact error like yours. I'm on 5.10.89-lts kernel. Hoping that this problem can be solved soon. 
[image]

BerejMan, 2022-12-11:
I solved. Need to install linux-headers

komradehieu, 2022-12-13:
> I solved. Need to install linux-headers

Bro how does this happen ? I have already installed Linux-headers. Screenshot down below
[image]

komradehieu, 2022-12-13:
I even have both linux-headers and linux -lts-headers. Still didnt work

[image]

NickTheNeko2002, 2022-12-13:
try installing these 3
`yay -Syu dkms`
`yay -S linux-headers`
and
`yay -S linux-firmware`

NickTheNeko2002, 2022-12-13:
> > I solved. Need to install linux-headers
> 
> Bro how does this happen ? I have already installed Linux-headers. Screenshot down below [image]

btw what DE is that?

komradehieu, 2022-12-14:
> > > I solved. Need to install linux-headers
> > 
> > 
> > Bro how does this happen ? I have already installed Linux-headers. Screenshot down below [image]
> 
> btw what DE is that?

no actually im using no DE. This is just a tiling window manager. And my case im using BSPWM tiling window manager

NickTheOnlyNeko, 2023-01-10:
> > > > I solved. Need to install linux-headers
> > > 
> > > 
> > > Bro how does this happen ? I have already installed Linux-headers. Screenshot down below [image]
> > 
> > 
> > btw what DE is that?
> 
> no actually im using no DE. This is just a tiling window manager. And my case im using BSPWM tiling window manager

use latest stable Linux version (for me it's 6.1.4-zen2-1-zen) and retry

komradehieu, 2023-01-10:
> > > > > I solved. Need to install linux-headers
> > > > 
> > > > 
> > > > Bro how does this happen ? I have already installed Linux-headers. Screenshot down below [image]
> > > 
> > > 
> > > btw what DE is that?
> > 
> > no actually im using no DE. This is just a tiling window manager. And my case im using BSPWM tiling window manager
> 
> use latest stable Linux version (for me it's 6.1.4) and retry
> 
> would recommend this one to use [this one](https://github.com/KanuX-14/rtl8188eus)
> 

update: im using latest lts kernel version and its working fine also. Monitor mode, aireplay-ng,... everything works, but sometimes its a little bit buggy. I tried zen kernel on my machine and ended up in booting directly to uefi instead of grub, have to chroot into arch again to delete zen kernel  and reinstall grub