aircrack-ng/rtl8188eus #208: new compile error on kernel 5.19.10 arch linux

voldmort21, 2022-09-24:
so i have got a new compile error even so it should compile after the latest commit but any way i got this - > 

https://pastebin.com/H24JxEak

[hussin@hussin-z600 rtl8188eus]$ sudo make -j$(nproc --all) all
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.19.10-arch1-1/build M=/home/hussin/build/rtl8188eus  modules
make[1]: Entering directory '/usr/lib/modules/5.19.10-arch1-1/build'
  CC [M]  /home/hussin/build/rtl8188eus/core/rtw_cmd.o
  CC [M]  /home/hussin/build/rtl8188eus/core/rtw_security.o
  CC [M]  /home/hussin/build/rtl8188eus/core/rtw_debug.o
  CC [M]  /home/hussin/build/rtl8188eus/core/rtw_io.o
  CC [M]  /home/hussin/build/rtl8188eus/core/rtw_ioctl_query.o
  CC [M]  /home/hussin/build/rtl8188eus/core/rtw_ieee80211.o
  CC [M]  /home/hussin/build/rtl8188eus/core/rtw_ioctl_set.o
  CC [M]  /home/hussin/build/rtl8188eus/core/rtw_mlme.o
  CC [M]  /home/hussin/build/rtl8188eus/core/rtw_mlme_ext.o
  CC [M]  /home/hussin/build/rtl8188eus/core/rtw_mi.o
  CC [M]  /home/hussin/build/rtl8188eus/core/rtw_wlan_util.o
  CC [M]  /home/hussin/build/rtl8188eus/core/rtw_vht.o
  CC [M]  /home/hussin/build/rtl8188eus/core/rtw_pwrctrl.o
  CC [M]  /home/hussin/build/rtl8188eus/core/rtw_rf.o
  CC [M]  /home/hussin/build/rtl8188eus/core/rtw_recv.o
  CC [M]  /home/hussin/build/rtl8188eus/core/rtw_sta_mgt.o
  CC [M]  /home/hussin/build/rtl8188eus/core/rtw_ap.o
  CC [M]  /home/hussin/build/rtl8188eus/core/mesh/rtw_mesh.o
  CC [M]  /home/hussin/build/rtl8188eus/core/mesh/rtw_mesh_pathtbl.o
  CC [M]  /home/hussin/build/rtl8188eus/core/rtw_xmit.o
  CC [M]  /home/hussin/build/rtl8188eus/core/rtw_p2p.o
  CC [M]  /home/hussin/build/rtl8188eus/core/mesh/rtw_mesh_hwmp.o
  CC [M]  /home/hussin/build/rtl8188eus/core/rtw_tdls.o
  CC [M]  /home/hussin/build/rtl8188eus/core/rtw_rson.o
  CC [M]  /home/hussin/build/rtl8188eus/core/rtw_br_ext.o
  CC [M]  /home/hussin/build/rtl8188eus/core/rtw_iol.o
  CC [M]  /home/hussin/build/rtl8188eus/core/rtw_sreset.o
  CC [M]  /home/hussin/build/rtl8188eus/core/rtw_btcoex_wifionly.o
  CC [M]  /home/hussin/build/rtl8188eus/core/rtw_btcoex.o
  CC [M]  /home/hussin/build/rtl8188eus/core/rtw_beamforming.o
  CC [M]  /home/hussin/build/rtl8188eus/core/rtw_odm.o
  CC [M]  /home/hussin/build/rtl8188eus/core/rtw_rm.o
  CC [M]  /home/hussin/build/rtl8188eus/core/rtw_rm_fsm.o
  CC [M]  /home/hussin/build/rtl8188eus/core/efuse/rtw_efuse.o
  CC [M]  /home/hussin/build/rtl8188eus/os_dep/osdep_service.o
  CC [M]  /home/hussin/build/rtl8188eus/os_dep/linux/os_intfs.o
/home/hussin/build/rtl8188eus/core/rtw_mlme_ext.c: In function ‘mgt_dispatcher’:
/home/hussin/build/rtl8188eus/core/rtw_mlme_ext.c:1838:25: warning: attribute ‘fallthrough’ not preceding a case label or default label
 1838 |                         __attribute__ ((__fallthrough__));
      |                         ^~~~~~~~~~~~~
  CC [M]  /home/hussin/build/rtl8188eus/os_dep/linux/usb_intf.o
/home/hussin/build/rtl8188eus/os_dep/osdep_service.c: In function ‘rtw_change_ifname’:
/home/hussin/build/rtl8188eus/os_dep/osdep_service.c:2454:28: warning: passing argument 1 of ‘_rtw_memcpy’ discards ‘const’ qualifier from pointer target type [-Wdiscarded-qualifiers]
 2454 |         _rtw_memcpy(pnetdev->dev_addr, adapter_mac_addr(padapter), ETH_ALEN);
      |                     ~~~~~~~^~~~~~~~~~
/home/hussin/build/rtl8188eus/os_dep/osdep_service.c:899:24: note: expected ‘void *’ but argument is of type ‘const unsigned char *’
  899 | void _rtw_memcpy(void *dst, const void *src, u32 sz)
      |                  ~~~~~~^~~
  CC [M]  /home/hussin/build/rtl8188eus/os_dep/linux/usb_ops_linux.o
/home/hussin/build/rtl8188eus/core/efuse/rtw_efuse.c: In function ‘rtw_efuse_analyze’:
/home/hussin/build/rtl8188eus/core/efuse/rtw_efuse.c:714:17: warning: this ‘if’ clause does not guard... [-Wmisleading-indentation]
  714 |                 if (i % 16 == 0)
      |                 ^~
In file included from /home/hussin/build/rtl8188eus/include/drv_types.h:59,
                 from /home/hussin/build/rtl8188eus/core/efuse/rtw_efuse.c:17:
/home/hussin/build/rtl8188eus/include/rtw_debug.h:194:9: note: ...this statement, but the latter is misleadingly indented as if it were guarded by the ‘if’
  194 |         do {\
      |         ^~
/home/hussin/build/rtl8188eus/core/efuse/rtw_efuse.c:716:25: note: in expansion of macro ‘_RTW_PRINT_SEL’
  716 |                         _RTW_PRINT_SEL(RTW_DBGDUMP, "%02X%s"
      |                         ^~~~~~~~~~~~~~
  CC [M]  /home/hussin/build/rtl8188eus/os_dep/linux/ioctl_linux.o
  CC [M]  /home/hussin/build/rtl8188eus/os_dep/linux/xmit_linux.o
[... 53 lines cut ...]
/home/hussin/build/rtl8188eus/os_dep/linux/ioctl_cfg80211.c:9379:20: note: (near initialization for ‘rtw_cfg80211_ops.stop_ap’)
/home/hussin/build/rtl8188eus/os_dep/linux/ioctl_cfg80211.c: In function ‘rtw_wdev_unregister’:
/home/hussin/build/rtl8188eus/os_dep/linux/ioctl_cfg80211.c:9637:17: error: ‘struct wireless_dev’ has no member named ‘current_bss’
 9637 |         if (wdev->current_bss) {
      |                 ^~
  CC [M]  /home/hussin/build/rtl8188eus/hal/rtl8188e/rtl8188e_cmd.o
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:249: /home/hussin/build/rtl8188eus/os_dep/linux/ioctl_cfg80211.o] Error 1
make[2]: *** Waiting for unfinished jobs....
/home/hussin/build/rtl8188eus/os_dep/linux/ioctl_linux.c: In function ‘rtw_mp_efuse_set’:
/home/hussin/build/rtl8188eus/os_dep/linux/ioctl_linux.c:9792:46: warning: passing argument 1 of ‘_rtw_memcpy’ discards ‘const’ qualifier from pointer target type [-Wdiscarded-qualifiers]
 9792 |                 _rtw_memcpy(padapter->pnetdev->dev_addr, get_hal_mac_addr(padapter), ETH_ALEN); /* set mac addr to net_device */
      |                             ~~~~~~~~~~~~~~~~~^~~~~~~~~~
In file included from /home/hussin/build/rtl8188eus/include/drv_types.h:26,
                 from /home/hussin/build/rtl8188eus/os_dep/linux/ioctl_linux.c:17:
/home/hussin/build/rtl8188eus/include/osdep_service.h:292:35: note: expected ‘void *’ but argument is of type ‘const unsigned char *’
  292 | extern void     _rtw_memcpy(void *dec, const void *sour, u32 sz);
      |                             ~~~~~~^~~
/home/hussin/build/rtl8188eus/hal/hal_hci/hal_usb.c: In function ‘usb_init_recv_priv’:
/home/hussin/build/rtl8188eus/hal/hal_hci/hal_usb.c:28:22: warning: cast between incompatible function types from ‘void (*)(void *)’ to ‘void (*)(long unsigned int)’ [-Wcast-function-type]
   28 |                      (void(*)(unsigned long))usb_recv_tasklet,
      |                      ^
make[1]: *** [Makefile:1846: /home/hussin/build/rtl8188eus] Error 2
make[1]: Leaving directory '/usr/lib/modules/5.19.10-arch1-1/build'
make: *** [Makefile:2058: modules] Error 2

KanuX-14, 2022-10-03:
Could you [try my repository](https://github.com/KanuX-14/rtl8188eus)? It has patches for new kernels.

voldmort21, 2022-10-06:
i removed the kernel and install 5.10 again

closed 2022-10-06 by voldmort21

bob69420x, 2022-10-18:
> Could you [try my repository](https://github.com/KanuX-14/rtl8188eus)? It has patches for new kernels.

Godsend! Worked! Best regards!

carloshoffmann, 2022-10-25:
it worked!!!