aircrack-ng/rtl8188eus #212: Compiling Error with rtl8188eus_new, kernel version 5.19.0-kali2

DeSlater, 2022-10-25:
Failing to compile during "sudo make"

/home/kali/rtl8188eus_new/os_dep/linux/ioctl_cfg80211.c: In function ‘rtw_cfg80211_ch_switch_notify’:
/home/kali/rtl8188eus_new/os_dep/linux/ioctl_cfg80211.c:420:9: error: too few arguments to function ‘cfg80211_ch_switch_notify’
  420 |         cfg80211_ch_switch_notify(adapter->pnetdev, &chdef);
      |         ^~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/kali/rtl8188eus_new/include/osdep_service_linux.h:93,
                 from /home/kali/rtl8188eus_new/include/osdep_service.h:50,
                 from /home/kali/rtl8188eus_new/include/drv_types.h:26,
                 from /home/kali/rtl8188eus_new/os_dep/linux/ioctl_cfg80211.c:17:
/usr/src/linux-headers-5.19.0-kali2-common/include/net/cfg80211.h:7967:6: note: declared here
 7967 | void cfg80211_ch_switch_notify(struct net_device *dev,
      |      ^~~~~~~~~~~~~~~~~~~~~~~~~
/home/kali/rtl8188eus_new/os_dep/linux/ioctl_cfg80211.c: At top level:
/home/kali/rtl8188eus_new/os_dep/linux/ioctl_cfg80211.c:9379:20: error: initialization of ‘int (*)(struct wiphy *, struct net_device *, unsigned int)’ from incompatible pointer type ‘int (*)(struct wiphy *, struct net_device *)’ [-Werror=incompatible-pointer-types]
 9379 |         .stop_ap = cfg80211_rtw_stop_ap,
      |                    ^~~~~~~~~~~~~~~~~~~~
/home/kali/rtl8188eus_new/os_dep/linux/ioctl_cfg80211.c:9379:20: note: (near initialization for ‘rtw_cfg80211_ops.stop_ap’)
/home/kali/rtl8188eus_new/os_dep/linux/ioctl_cfg80211.c: In function ‘rtw_wdev_unregister’:
/home/kali/rtl8188eus_new/os_dep/linux/ioctl_cfg80211.c:9637:17: error: ‘struct wireless_dev’ has no member named ‘current_bss’
 9637 |         if (wdev->current_bss) {
      |                 ^~
cc1: some warnings being treated as errors
make[2]: *** [/usr/src/linux-headers-5.19.0-kali2-common/scripts/Makefile.build:254: /home/kali/rtl8188eus_new/os_dep/linux/ioctl_cfg80211.o] Error 1
make[1]: *** [/usr/src/linux-headers-5.19.0-kali2-common/Makefile:1870: /home/kali/rtl8188eus_new] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.19.0-kali2-amd64'
make: *** [Makefile:2059: modules] Error 2

KanuX-14, 2022-10-25:
For the moment, I am updating the repository for my own until this get updated.
This is a new problem within kernel versions above 5.19.
[Try compiling with this one](https://github.com/KanuX-14/rtl8188eus).

VidCC, 2022-10-27:
[image]

A alguien mas no le compila?

LamTruong-SystemAdmin, 2022-10-28:
Please remove the kernel version 5.19 from the machine and try again:

LamTruong-SystemAdmin, 2022-10-28:
I fixed the error. Please remove the kernel version 5.19 from the machine and try again:
1. sudo apt-get remove --purge linux-image-5.19* -y
2. sudo apt-get remove --purge linux-headers-5.19* -y

LamTruong-SystemAdmin, 2022-10-28:
I fixed the error. If you have updated kernel to version 5.19. Please remove the kernel version 5.19 from the machine and try again:
sudo apt-get remove --purge linux-image-5.19* -y
sudo apt-get remove --purge linux-headers-5.19* -y
reboot
...

lucky-zwx, 2022-11-01:
> 

[image]

lucky-zwx, 2022-11-01:
> For the moment, I am updating the repository for my own until this get updated. This is a new problem within kernel versions above 5.19. [Try compiling with this one](https://github.com/KanuX-14/rtl8188eus).

[image]

kardgnik, 2022-11-01:
> For the moment, I am updating the repository for my own until this get updated. This is a new problem within kernel versions above 5.19. [Try compiling with this one](https://github.com/KanuX-14/rtl8188eus).

ty worked

KanuX-14, 2022-11-01:
@lucky-zwx Indeed it compiled, but I guess it is trying to apply the patch to the main Android kernel.
But thanks though. I might add a manual installation of the module without the `make install` command.

lucky-zwx, 2022-11-02:
> @lucky-zwx它确实编译了，但我猜它是试图应用补丁到主要的Android内核。不过，还是谢谢你。我可以添加模块的手动安装，而不使用`make install` command.

What should i do?

Rockyman101, 2022-11-06:
> For the moment, I am updating the repository for my own until this get updated. This is a new problem within kernel versions above 5.19. [Try compiling with this one](https://github.com/KanuX-14/rtl8188eus).

thaks bro helped so muchh 🚀🚀🚀