aircrack-ng/rtl8188eus #220: Error with compiling [fatal error: drv_types.h]

KacperKsiez, 2022-12-19:
**Errors after make command**

`/home/kacper/rtl8188eus/core/rtw_cmd.c:17:10: fatal error: drv_types.h: No such file or directory
   17 | #include <drv_types.h>
      |          ^~~~~~~~~~~~~
compilation terminated.`

Full log

`make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/6.0.12-arch1-1/build M=/home/kacper/rtl8188eus  modules
make[1]: Entering directory '/usr/lib/modules/6.0.12-arch1-1/build'
  CC [M]  /home/kacper/rtl8188eus/core/rtw_cmd.o
/home/kacper/rtl8188eus/core/rtw_cmd.c:17:10: fatal error: drv_types.h: No such file or directory
   17 | #include <drv_types.h>
      |          ^~~~~~~~~~~~~
compilation terminated.
make[2]: *** [scripts/Makefile.build:249: /home/kacper/rtl8188eus/core/rtw_cmd.o] Error 1
make[1]: *** [Makefile:1852: /home/kacper/rtl8188eus] Error 2
make[1]: Leaving directory '/usr/lib/modules/6.0.12-arch1-1/build'
make: *** [Makefile:2062: modules] Error 2`

bzass, 2022-12-19:
I'm having the same problem

a520ass, 2022-12-20:
git reset --hard f79dffb49b567a854c4f510a93a6d901caf63342

kimocoder, 2022-12-20:
Yes, need to revert the latest PR.

MrRob0-X, 2022-12-20:
> **Errors after make command**
>
> `/home/kacper/rtl8188eus/core/rtw_cmd.c:17:10: fatal error: drv_types.h: No such file or directory 17 | #include <drv_types.h> | ^~~~~~~~~~~~~ compilation [...]

try this in driver dir then try build again and show screenshot
`sed -i 's/$(srctree)\/$(src)/$(pwd)/'g`

MrRob0-X, 2022-12-20:
> git reset --hard [f79dffb](https://github.com/aircrack-ng/rtl8188eus/commit/f79dffb49b567a854c4f510a93a6d901caf63342)

won't help

LoaiEsam37, 2022-12-20:
yea i have the same error too

MrRob0-X, 2022-12-20:
> yea i have the same error too

try the solution in my comment

LoaiEsam37, 2022-12-20:
```
┌──(loai㉿kali)-[~/rtl8188eus]
└─$ sed -i 's/$(srctree)\/$(src)/$(pwd)/'g Makefile

┌──(loai㉿kali)-[~/rtl8188eus]
└─$ sudo make
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.16.0-kali7-amd64/build M=/home/loai/rtl8188eus  modules
make[1]: Entering directory '/usr/src/linux-headers-5.16.0-kali7-amd64'
/bin/sh: 1: /usr/src/linux-headers-5.16.0-kali7-common/scripts/pahole-flags.sh: not found
/bin/sh: 1: /usr/src/linux-headers-5.16.0-kali7-common/scripts/pahole-flags.sh: not found
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: gcc-11 (Debian 11.2.0-19) 11.2.0
  You are using:           gcc-11 (Debian 11.3.0-8) 11.3.0
/bin/sh: 1: /usr/src/linux-headers-5.16.0-kali7-common/scripts/pahole-flags.sh: not found
  CC [M]  /home/loai/rtl8188eus/core/rtw_cmd.o
/home/loai/rtl8188eus/core/rtw_cmd.c:17:10: fatal error: drv_types.h: No such file or directory
   17 | #include <drv_types.h>
      |          ^~~~~~~~~~~~~
compilation terminated.
make[2]: *** [/usr/src/linux-headers-5.16.0-kali7-common/scripts/Makefile.build:292: /home/loai/rtl8188eus/core/rtw_cmd.o] Error 1
make[1]: *** [/usr/src/linux-headers-5.16.0-kali7-common/Makefile:1870: /home/loai/rtl8188eus] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.16.0-kali7-amd64'
make: *** [Makefile:2062: modules] Error 2
```
**Same Error**

MrRob0-X, 2022-12-20:
> ```
> ┌──(loai㉿kali)-[~/rtl8188eus]
> └─$ sed -i 's/$(srctree)\/$(src)/$(pwd)/'g Makefile
>
> ┌──(loai㉿kali)-[~/rtl8188eus]
> └─$ sudo make
> make ARCH=x86_64 CROSS_COMPILE= -C [...]

`sed -i 's/$(srctree)\/$(src)/$(pwd)\/$(src)/'g Makefile`

LoaiEsam37, 2022-12-20:
### Still the same Error
```
┌──(loai㉿kali)-[~/rtl8188eus]
└─$ sed -i 's/$(srctree)\/$(src)/$(pwd)/'g Makefile
```
i had just do it before make cmd

MrRob0-X, 2022-12-20:
> ### Still the same Error
> ```
> ┌──(loai㉿kali)-[~/rtl8188eus]
> └─$ sed -i 's/$(srctree)\/$(src)/$(pwd)/'g Makefile
> ```
> 
> i had just do it before make cmd

it's different, show me current bug

LoaiEsam37, 2022-12-20:
```
┌──(loai㉿kali)-[~/rtl8188eus]
└─$ sudo make
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.16.0-kali7-amd64/build M=/home/loai/rtl8188eus  modules
make[1]: Entering directory '/usr/src/linux-headers-5.16.0-kali7-amd64'
/bin/sh: 1: /usr/src/linux-headers-5.16.0-kali7-common/scripts/pahole-flags.sh: not found
/bin/sh: 1: /usr/src/linux-headers-5.16.0-kali7-common/scripts/pahole-flags.sh: not found
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: gcc-11 (Debian 11.2.0-19) 11.2.0
  You are using:           gcc-11 (Debian 11.3.0-8) 11.3.0
/bin/sh: 1: /usr/src/linux-headers-5.16.0-kali7-common/scripts/pahole-flags.sh: not found
  CC [M]  /home/loai/rtl8188eus/core/rtw_cmd.o
/home/loai/rtl8188eus/core/rtw_cmd.c:17:10: fatal error: drv_types.h: No such file or directory
   17 | #include <drv_types.h>
      |          ^~~~~~~~~~~~~
compilation terminated.
make[2]: *** [/usr/src/linux-headers-5.16.0-kali7-common/scripts/Makefile.build:292: /home/loai/rtl8188eus/core/rtw_cmd.o] Error 1
make[1]: *** [/usr/src/linux-headers-5.16.0-kali7-common/Makefile:1870: /home/loai/rtl8188eus] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.16.0-kali7-amd64'
make: *** [Makefile:2062: modules] Error 2
```

LoaiEsam37, 2022-12-20:
# No It`s not.

MrRob0-X, 2022-12-20:
> ```
> ┌──(loai㉿kali)-[~/rtl8188eus]
> └─$ sudo make
> make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.16.0-kali7-amd64/build M=/home/loai/rtl8188eus  modules
> make[1]: Entering directory [...]

ok let's this default path for now, i also try something to avoid this bug for 5.4 kernels

`sed -i 's/$(pwd)/$(srctree)\/$(src)/'g Makefile`

MrRob0-X, 2022-12-20:
> > ```
> > ┌──(loai㉿kali)-[~/rtl8188eus]
> > └─$ sudo make
> > make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.16.0-kali7-amd64/build M=/home/loai/rtl8188eus  modules
> > make[1]: Entering directory '/usr/src/linux-headers-5.16.0-kali7-amd64'
> > /bin/sh: 1: /usr/src/linux-headers-5.16.0-kali7-common/scripts/pahole-flags.sh: not found
> > /bin/sh: 1: /usr/src/linux-headers-5.16.0-kali7-common/scripts/pahole-flags.sh: not found
> > warning: the compiler differs from the one used to build the kernel
> >   The kernel was built by: gcc-11 (Debian 11.2.0-19) 11.2.0
> >   You are using:           gcc-11 (Debian 11.3.0-8) 11.3.0
> > /bin/sh: 1: /usr/src/linux-headers-5.16.0-kali7-common/scripts/pahole-flags.sh: not found
> >   CC [M]  /home/loai/rtl8188eus/core/rtw_cmd.o
> > /home/loai/rtl8188eus/core/rtw_cmd.c:17:10: fatal error: drv_types.h: No such file or directory
> >    17 | #include <drv_types.h>
> >       |          ^~~~~~~~~~~~~
> > compilation terminated.
> > make[2]: *** [/usr/src/linux-headers-5.16.0-kali7-common/scripts/Makefile.build:292: /home/loai/rtl8188eus/core/rtw_cmd.o] Error 1
> > make[1]: *** [/usr/src/linux-headers-5.16.0-kali7-common/Makefile:1870: /home/loai/rtl8188eus] Error 2
> > make[1]: Leaving directory '/usr/src/linux-headers-5.16.0-kali7-amd64'
> > make: *** [Makefile:2062: modules] Error 2
> > ```
> 
> ok let's use default path for now, i also try something to avoid this bug for 5.4 kernels
> 
> `sed -i 's/$(pwd)/$(srctree)\/$(src)/'g Makefile`

MrRob0-X, 2022-12-20:
> > > ```
> > > ┌──(loai㉿kali)-[~/rtl8188eus]
> > > └─$ sudo make
> > > make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.16.0-kali7-amd64/build M=/home/loai/rtl8188eus  modules
> > > make[1]: Entering directory '/usr/src/linux-headers-5.16.0-kali7-amd64'
> > > /bin/sh: 1: /usr/src/linux-headers-5.16.0-kali7-common/scripts/pahole-flags.sh: not found
> > > /bin/sh: 1: /usr/src/linux-headers-5.16.0-kali7-common/scripts/pahole-flags.sh: not found
> > > warning: the compiler differs from the one used to build the kernel
> > >   The kernel was built by: gcc-11 (Debian 11.2.0-19) 11.2.0
> > >   You are using:           gcc-11 (Debian 11.3.0-8) 11.3.0
> > > /bin/sh: 1: /usr/src/linux-headers-5.16.0-kali7-common/scripts/pahole-flags.sh: not found
> > >   CC [M]  /home/loai/rtl8188eus/core/rtw_cmd.o
> > > /home/loai/rtl8188eus/core/rtw_cmd.c:17:10: fatal error: drv_types.h: No such file or directory
> > >    17 | #include <drv_types.h>
> > >       |          ^~~~~~~~~~~~~
> > > compilation terminated.
> > > make[2]: *** [/usr/src/linux-headers-5.16.0-kali7-common/scripts/Makefile.build:292: /home/loai/rtl8188eus/core/rtw_cmd.o] Error 1
> > > make[1]: *** [/usr/src/linux-headers-5.16.0-kali7-common/Makefile:1870: /home/loai/rtl8188eus] Error 2
> > > make[1]: Leaving directory '/usr/src/linux-headers-5.16.0-kali7-amd64'
> > > make: *** [Makefile:2062: modules] Error 2
> > > ```
> > 
> > 
> > ok let's use default path for now, I'll also try something to avoid this bug for 5.4 kernels
> > `sed -i 's/$(pwd)/$(srctree)\/$(src)/'g Makefile`

LoaiEsam37, 2022-12-20:
bro, i have executed this ``sed -i 's/$(pwd)/$(src)/'g Makefile``  for like 20 times in ``~/rtl8188eus`` directory and nothing works

MrRob0-X, 2022-12-20:
> `sed -i 's/$(srctree)\/$(src)/$(pwd)\/$(src)/'g Makefile`

wait i think this command is wrong, i added $(srctree)/$(src) but it's aready removed by previous command https://github.com/aircrack-ng/rtl8188eus/issues/220#issuecomment-1358935037

MrRob0-X, 2022-12-20:
> bro, i have executed this `sed -i 's/$(pwd)/$(src)/'g Makefile` for like 20 times in `~/rtl8188eus` directory and nothing works

you didn't read it, there are different stuff if you take a look

LoaiEsam37, 2022-12-20:
ok , i will remove the ``repo`` and try to ``git clone`` again and try the ``sed -i 's/$(srctree)\/$(src)/$(pwd)\/$(src)/'g Makefile``

MrRob0-X, 2022-12-20:
> ok , i will remove the `repo` and try to `git clone` again and try the `sed -i 's/$(srctree)\/$(src)/$(pwd)\/$(src)/'g Makefile`

don't lol, do `sed -i 's//$(pwd)/$(pwd)\/$(src)/'g Makefile`

LoaiEsam37, 2022-12-20:
i have already delete it 😁

LoaiEsam37, 2022-12-20:
Thanks a lot. It worked

MrRob0-X, 2022-12-20:
> Thanks a lot. It worked
do `grep -r "/include" Makefile` and show me screenshot

LoaiEsam37, 2022-12-20:
```
┌──(loai㉿kali)-[~/rtl8188eus]
└─$ grep -r "/include" Makefile
EXTRA_CFLAGS += -I$(pwd)/$(src)/include

```

Quang-vip, 2022-12-20:
Hello, coincidentally I'm having the same problem as I'm too trying to compile the driver with a nethunter kernel.
I tried your suggestion of  `sed -i 's/$(pwd)/$(pwd)\/$(src)/'g Makefile`  but it's yielding the same error of:
`rtl8188eus/core/rtw_cmd.c:17:23: fatal error: drv_types.h: No such file or directory`

The full compilation log:

```
paimon@buildsrv:~/rtl8188eus$ make KSRC=~/santoni_headers/lib/modules/*/build EXTRA_CFLAGS="-Wno-error"
make ARCH=arm64 CROSS_COMPILE=/home/paimon/santoni_kernel_nethunter/toolchain/bin/aarch64-linux-android- -C /home/paimon/santoni_headers/lib/modules/*/build M=/home/paimon/rtl8188eus  modules
make[1]: Entering directory '/home/paimon/santoni_kernel_nethunter'

  WARNING: Symbol version dump ./Module.symvers
           is missing; modules will have no dependencies and modversions.

  CC [M]  /home/paimon/rtl8188eus/core/rtw_cmd.o
/home/paimon/rtl8188eus/core/rtw_cmd.c:17:23: fatal error: drv_types.h: No such file or directory
 #include <drv_types.h>
                       ^
compilation terminated.
make[2]: *** [scripts/Makefile.build:258: /home/paimon/rtl8188eus/core/rtw_cmd.o] Error 1
make[1]: *** [Makefile:1408: _module_/home/paimon/rtl8188eus] Error 2
make[1]: Leaving directory '/home/paimon/santoni_kernel_nethunter'
make: *** [Makefile:2062: modules] Error 2
```

Any idea how I can fix that?
Would really appreciate it if there's a remedy to this :D

MrRob0-X, 2022-12-20:
> Hello, coincidentally I'm having the same problem as I'm too trying to compile the driver with a nethunter kernel. I tried your suggestion of `sed -i 's/$(pwd)/$(pwd)\/$(src)/'g Makefile` but it's [...]

`sed -i 's/$(srctree)\/$(src)/$(pwd)\/$(src)/'g Makefile`

Quang-vip, 2022-12-20:
> `sed -i 's/$(srctree)\/$(src)/$(pwd)\/$(src)/'g Makefile`

Thanks for the help :DD
Still the same thing apparently:

```
paimon@buildsrv:~/rtl8188eus$ sed -i 's/$(srctree)\/$(src)/$(pwd)\/$(src)/'g Makefile
paimon@buildsrv:~/rtl8188eus$ make KSRC=~/santoni_headers/lib/modules/*/build EXTRA_CFLAGS="-Wno-error"
make ARCH=arm64 CROSS_COMPILE=/home/paimon/santoni_kernel_nethunter/toolchain/bin/aarch64-linux-android- -C /home/paimon/santoni_headers/lib/modules/*/build M=/home/paimon/rtl8188eus  modules
make[1]: Entering directory '/home/paimon/santoni_kernel_nethunter'

  WARNING: Symbol version dump ./Module.symvers
           is missing; modules will have no dependencies and modversions.

  CC [M]  /home/paimon/rtl8188eus/core/rtw_cmd.o
/home/paimon/rtl8188eus/core/rtw_cmd.c:17:23: fatal error: drv_types.h: No such file or directory
 #include <drv_types.h>
                       ^
compilation terminated.
make[2]: *** [scripts/Makefile.build:258: /home/paimon/rtl8188eus/core/rtw_cmd.o] Error 1
make[1]: *** [Makefile:1408: _module_/home/paimon/rtl8188eus] Error 2
make[1]: Leaving directory '/home/paimon/santoni_kernel_nethunter'
make: *** [Makefile:2062: modules] Error 2

```

MrRob0-X, 2022-12-20:
> Thanks for the help :DD Still the same thing apparently:
>
> ```
> paimon@buildsrv:~/rtl8188eus$ sed -i 's/$(srctree)\/$(src)/$(pwd)\/$(src)/'g Makefile
> paimon@buildsrv:~/rtl8188eus$ make [...]

dot this https://github.com/aircrack-ng/rtl8188eus/issues/220#issuecomment-1359117072

Quang-vip, 2022-12-20:
What it shows:

`paimon@buildsrv:~/rtl8188eus$ grep -r "/include" Makefile
EXTRA_CFLAGS += -I$(src)/include
`

MrRob0-X, 2022-12-20:
> What it shows:
> 
> `paimon@buildsrv:~/rtl8188eus$ grep -r "/include" Makefile EXTRA_CFLAGS += -I$(src)/include `

`sed -i 's/$(src)/$(pwd)\/$(src)/'g Makefile`

Quang-vip, 2022-12-20:
Still yields the same:

```
paimon@buildsrv:~/rtl8188eus$ sed -i 's/$(src)/$(pwd)\/$(src)/'g Makefile
paimon@buildsrv:~/rtl8188eus$ make KSRC=~/santoni_headers/lib/modules/*/build EXTRA_CFLAGS="-Wno-error"
make ARCH=arm64 CROSS_COMPILE=/home/paimon/santoni_kernel_nethunter/toolchain/bin/aarch64-linux-android- -C /home/paimon/santoni_headers/lib/modules/*/build M=/home/paimon/rtl8188eus  modules
make[1]: Entering directory '/home/paimon/santoni_kernel_nethunter'

  WARNING: Symbol version dump ./Module.symvers
           is missing; modules will have no dependencies and modversions.

  CC [M]  /home/paimon/rtl8188eus/core/rtw_cmd.o
/home/paimon/rtl8188eus/core/rtw_cmd.c:17:23: fatal error: drv_types.h: No such file or directory
 #include <drv_types.h>
                       ^
compilation terminated.
make[2]: *** [scripts/Makefile.build:258: /home/paimon/rtl8188eus/core/rtw_cmd.o] Error 1
make[1]: *** [Makefile:1408: _module_/home/paimon/rtl8188eus] Error 2
make[1]: Leaving directory '/home/paimon/santoni_kernel_nethunter'
make: *** [Makefile:2062: modules] Error 2
```

the output of:
```
paimon@buildsrv:~/rtl8188eus$ grep -r "/include" Makefile                                            
EXTRA_CFLAGS += -I$(pwd)/$(pwd)/$(src)/include
```

MrRob0-X, 2022-12-20:
> Still yields the same:
>
> ```
> paimon@buildsrv:~/rtl8188eus$ sed -i 's/$(src)/$(pwd)\/$(src)/'g Makefile
> paimon@buildsrv:~/rtl8188eus$ make KSRC=~/santoni_headers/lib/modules/*/build [...]
you broken the path dude

`sed -i 's/$(pwd)/$(pwd)$(src)/$(pwd)\/$(src)/'g Makefile`  be carefully when typing it this time

Quang-vip, 2022-12-20:
I double-checked it and welp, still no progress

This time though:
```
paimon@buildsrv:~/rtl8188eus$ grep -r "/include" Makefile                                             
EXTRA_CFLAGS += -I$(pwd)/$(src)/include
```

MrRob0-X, 2022-12-20:
> I double-checked it and welp, still no progress
> 
> This time though:
> 
> ```
> paimon@buildsrv:~/rtl8188eus$ grep -r "/include" Makefile                                             
> EXTRA_CFLAGS += -I$(pwd)/$(src)/include
> ```

tried build now?

Quang-vip, 2022-12-20:
> > I double-checked it and welp, still no progress
> > This time though:
> > ```
> > paimon@buildsrv:~/rtl8188eus$ grep -r "/include" Makefile                                             
> > EXTRA_CFLAGS += -I$(pwd)/$(src)/include
> > ```
> 
> tried build now?

Yes I did and the result is still the same

MrRob0-X, 2022-12-20:
> > > I double-checked it and welp, still no progress
> > > This time though:
> > > ```
> > > paimon@buildsrv:~/rtl8188eus$ grep -r "/include" Makefile                                             
> > > EXTRA_CFLAGS += -I$(pwd)/$(src)/include
> > > ```
> > 
> > 
> > tried build now?
> 
> Yes I did and the result is still the same

try this https://github.com/aircrack-ng/rtl8188eus/issues/220#issuecomment-1359104073

don't forget to show me /include output also

MrRob0-X, 2022-12-20:
@Quang-vip @JpvMaking

try apply this patch and build again, i will also try it on kernel source

`wget https://github.com/MrRob0-X/rtl8188eus/commit/5e3cfb4be0258dbca0ad553be5dc04d0829dbd04.patch && patch -p1 < *patch && rm -rf *patch`

Giuseppe7887, 2022-12-20:
> @Quang-vip @JpvMaking
> 
> try apply this patch and build again, i will also try it on kernel source
> 
> `wget https://github.com/MrRob0-X/rtl8188eus/commit/5e3cfb4be0258dbca0ad553be5dc04d0829dbd04.patch && patch -p1 < *patch && rm -rf *patch`

it worked for me!

sans404-dev, 2022-12-20:
> @Quang-vip @JpvMaking
> 
> try apply this patch and build again, i will also try it on kernel source
> 
> `wget https://github.com/MrRob0-X/rtl8188eus/commit/5e3cfb4be0258dbca0ad553be5dc04d0829dbd04.patch && patch -p1 < *patch && rm -rf *patch`

Thanks, it worked

Quang-vip, 2022-12-20:
> @Quang-vip @JpvMaking
> 
> try apply this patch and build again, i will also try it on kernel source
> 
> `wget https://github.com/MrRob0-X/rtl8188eus/commit/5e3cfb4be0258dbca0ad553be5dc04d0829dbd04.patch && patch -p1 < *patch && rm -rf *patch`

frustratingly enough, still the same issue after I had repulled the source and applied your patch
perhaps it's something wrong with my kernel headers?

MrRob0-X, 2022-12-20:
> > @Quang-vip @JpvMaking
> > try apply this patch and build again, i will also try it on kernel source
> > `wget https://github.com/MrRob0-X/rtl8188eus/commit/5e3cfb4be0258dbca0ad553be5dc04d0829dbd04.patch && patch -p1 < *patch && rm -rf *patch`
> 
> frustratingly enough, still the same issue after I had repulled the source and applied your patch perhaps it's something wrong with my kernel headers?
> 
> the output of /include after the patch :
> 
> ```
> paimon@paimon:~/rtl8188eus$ grep -r "/include" Makefile                                             
> EXTRA_CFLAGS += -I$(src)/include
> EXTRA_CFLAGS += -I$(srctree)/$(src)/include
> ```

try build with default kernel headers not phone one

Quang-vip, 2022-12-20:
> try build with default kernel headers not phone one

now it succeeds, so it's my nethunter kernel headers' fault

MrRob0-X, 2022-12-20:
> > try build with default kernel headers not phone one
> 
> now it succeeds, so it's my nethunter kernel headers' fault
> 
> nevertheless, thanks a lot for your contribution :D seems like my problem is much deeper than that

you can give this a try with your phone kernel header
https://github.com/aircrack-ng/rtl8188eus/pull/222/commits/3a4ce2bde940b02a476577acf87087dacbffe34c

Magnumjackal, 2022-12-21:
> wget https://github.com/MrRob0-X/rtl8188eus/commit/5e3cfb4be0258dbca0ad553be5dc04d0829dbd04.patch && patch -p1 < *patch && rm -rf *patch

Thank you so much, patch worked perfectly!

Quang-vip, 2022-12-21:
> > > try build with default kernel headers not phone one
> > 
> > 
> > now it succeeds, so it's my nethunter kernel headers' fault
> > nevertheless, thanks a lot for your contribution :D seems like my problem is much deeper than that
> 
> you can give this a try with your phone kernel header [3a4ce2b](https://github.com/aircrack-ng/rtl8188eus/commit/3a4ce2bde940b02a476577acf87087dacbffe34c)

if I remove the `EXTRA_CFLAGS="-Wno-error"`, it'd go smoothly until this point:

```
...
CC [M]  /home/paimon/rtl8188eus/os_dep/linux/xmit_linux.o
CC [M]  /home/paimon/rtl8188eus/os_dep/linux/mlme_linux.o
CC [M]  /home/paimon/rtl8188eus/os_dep/linux/recv_linux.o
CC [M]  /home/paimon/rtl8188eus/os_dep/linux/ioctl_cfg80211.o
/home/paimon/rtl8188eus/os_dep/linux/ioctl_cfg80211.c: In function 'cfg80211_rtw_change_iface':
/home/paimon/rtl8188eus/os_dep/linux/ioctl_cfg80211.c:2376:2: warning: empty declaration
error, forbidden warning: ioctl_cfg80211.c:2376
make[2]: *** [scripts/Makefile.build:258: /home/paimon/rtl8188eus/os_dep/linux/ioctl_cfg80211.o] Error 1
make[1]: *** [Makefile:1408: _module_/home/paimon/rtl8188eus] Error 2
make[1]: Leaving directory '/home/paimon/santoni_kernel_nethunter'
make: *** [Makefile:2062: modules] Error 2
```

I googled this and that's the purpose of the `-Wno-error` flag I suppose
but then it wouldn't be able to find _drv_types.h_ if I pass that flag in
really confusing tbh

masharia-john, 2022-12-23:
┌──(kali㉿kali)-[~/Desktop/rtl8188eus-5.3.9]
└─$ make
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.18.0-kali5-amd64/build M=/home/kali/Desktop/rtl8188eus-5.3.9  modules
make[1]: Entering directory '/usr/src/linux-headers-5.18.0-kali5-amd64'
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: gcc-11 (Debian 11.3.0-3) 11.3.0
  You are using:           gcc-11 (Debian 11.3.0-8) 11.3.0
  CC [M]  /home/kali/Desktop/rtl8188eus-5.3.9/core/rtw_cmd.o
/home/kali/Desktop/rtl8188eus-5.3.9/core/rtw_cmd.c:17:10: fatal error: drv_types.h: No such file or directory
   17 | #include <drv_types.h>
      |          ^~~~~~~~~~~~~
compilation terminated.
make[2]: *** [/usr/src/linux-headers-5.18.0-kali5-common/scripts/Makefile.build:293: /home/kali/Desktop/rtl8188eus-5.3.9/core/rtw_cmd.o] Error 1
make[1]: *** [/usr/src/linux-headers-5.18.0-kali5-common/Makefile:1858: /home/kali/Desktop/rtl8188eus-5.3.9] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.18.0-kali5-amd64'
make: *** [Makefile:2062: modules] Error 2
                                                                                                                                                                      
┌──(kali㉿kali)-[~/Desktop/rtl8188eus-5.3.9]
└─$ sed -i 's/$(srctree)\/$(src)/$(pwd)/'g Makefile
                                                                                                                                                                      
┌──(kali㉿kali)-[~/Desktop/rtl8188eus-5.3.9]
└─$ make
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.18.0-kali5-amd64/build M=/home/kali/Desktop/rtl8188eus-5.3.9  modules
make[1]: Entering directory '/usr/src/linux-headers-5.18.0-kali5-amd64'
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: gcc-11 (Debian 11.3.0-3) 11.3.0
  You are using:           gcc-11 (Debian 11.3.0-8) 11.3.0
  CC [M]  /home/kali/Desktop/rtl8188eus-5.3.9/core/rtw_cmd.o
/home/kali/Desktop/rtl8188eus-5.3.9/core/rtw_cmd.c:17:10: fatal error: drv_types.h: No such file or directory
   17 | #include <drv_types.h>
      |          ^~~~~~~~~~~~~
compilation terminated.
make[2]: *** [/usr/src/linux-headers-5.18.0-kali5-common/scripts/Makefile.build:293: /home/kali/Desktop/rtl8188eus-5.3.9/core/rtw_cmd.o] Error 1
make[1]: *** [/usr/src/linux-headers-5.18.0-kali5-common/Makefile:1858: /home/kali/Desktop/rtl8188eus-5.3.9] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.18.0-kali5-amd64'
make: *** [Makefile:2062: modules] Error 2
                                                                                                                                                                      
┌──(kali㉿kali)-[~/Desktop/rtl8188eus-5.3.9]
└─$ 
So i tried the above suggestions and i still get the same error ### messages.So i tried the patch and got...

┌──(kali㉿kali)-[~/Desktop/rtl8188eus-5.3.9]
└─$ wget https://github.com/MrRob0-X/rtl8188eus/commit/5e3cfb4be0258dbca0ad553be5dc04d0829dbd04.patch && patch -p1 < *patch && rm -rf *patch
--2022-12-23 22:39:22--  https://github.com/MrRob0-X/rtl8188eus/commit/5e3cfb4be0258dbca0ad553be5dc04d0829dbd04.patch
Resolving github.com (github.com)... 140.82.121.3
Connecting to github.com (github.com)|140.82.121.3|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 1113 (1.1K) [text/plain]
Saving to: ‘5e3cfb4be0258dbca0ad553be5dc04d0829dbd04.patch.1’

5e3cfb4be0258dbca0ad553be5dc04d0829dbd04. 100%[===================================================================================>]   1.09K  --.-KB/s    in 0s      

2022-12-23 22:39:23 (11.1 MB/s) - ‘5e3cfb4be0258dbca0ad553be5dc04d0829dbd04.patch.1’ saved [1113/1113]

patching file Makefile
Hunk #1 FAILED at 19.
Hunk #2 FAILED at 226.
2 out of 2 hunks FAILED -- saving rejects to file Makefile.rej
so tried the flag call and got...

┌──(kali㉿kali)-[~/Desktop/rtl8188eus-5.3.9]
└─$ grep -r "/include" Makefile                                             
EXTRA_CFLAGS += -I$(pwd)/$(src)/include

EXTRA_CFLAGS += -I$(srctree)/$(pwd)/$(src)/include
EXTRA_CFLAGS: command not found

MrRob0-X, 2022-12-23:
> ┌──(kali㉿kali)-[~/Desktop/rtl8188eus-5.3.9] └─$ make make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.18.0-kali5-amd64/build M=/home/kali/Desktop/rtl8188eus-5.3.9 modules make[1]: Entering directory '/usr/src/linux-headers-5.18.0-kali5-amd64' warning: the compiler differs from the one used to build the kernel The kernel was built by: gcc-11 (Debian 11.3.0-3) 11.3.0 You are using: gcc-11 (Debian 11.3.0-8) 11.3.0 CC [M] /home/kali/Desktop/rtl8188eus-5.3.9/core/rtw_cmd.o /home/kali/Desktop/rtl8188eus-5.3.9/core/rtw_cmd.c:17:10: fatal error: drv_types.h: No such file or directory 17 | #include <drv_types.h> | ^~~~~~~~~~~~~ compilation terminated. make[2]: *** [/usr/src/linux-headers-5.18.0-kali5-common/scripts/Makefile.build:293: /home/kali/Desktop/rtl8188eus-5.3.9/core/rtw_cmd.o] Error 1 make[1]: *** [/usr/src/linux-headers-5.18.0-kali5-common/Makefile:1858: /home/kali/Desktop/rtl8188eus-5.3.9] Error 2 make[1]: Leaving directory '/usr/src/linux-headers-5.18.0-kali5-amd64' make: *** [Makefile:2062: modules] Error 2
> 
> ┌──(kali㉿kali)-[~/Desktop/rtl8188eus-5.3.9] └─$ sed -i 's/$(srctree)/$(src)/$(pwd)/'g Makefile
> 
> ┌──(kali㉿kali)-[~/Desktop/rtl8188eus-5.3.9] └─$ make make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.18.0-kali5-amd64/build M=/home/kali/Desktop/rtl8188eus-5.3.9 modules make[1]: Entering directory '/usr/src/linux-headers-5.18.0-kali5-amd64' warning: the compiler differs from the one used to build the kernel The kernel was built by: gcc-11 (Debian 11.3.0-3) 11.3.0 You are using: gcc-11 (Debian 11.3.0-8) 11.3.0 CC [M] /home/kali/Desktop/rtl8188eus-5.3.9/core/rtw_cmd.o /home/kali/Desktop/rtl8188eus-5.3.9/core/rtw_cmd.c:17:10: fatal error: drv_types.h: No such file or directory 17 | #include <drv_types.h> | ^~~~~~~~~~~~~ compilation terminated. make[2]: *** [/usr/src/linux-headers-5.18.0-kali5-common/scripts/Makefile.build:293: /home/kali/Desktop/rtl8188eus-5.3.9/core/rtw_cmd.o] Error 1 make[1]: *** [/usr/src/linux-headers-5.18.0-kali5-common/Makefile:1858: /home/kali/Desktop/rtl8188eus-5.3.9] Error 2 make[1]: Leaving directory '/usr/src/linux-headers-5.18.0-kali5-amd64' make: *** [Makefile:2062: modules] Error 2
> 
> ┌──(kali㉿kali)-[~/Desktop/rtl8188eus-5.3.9] └─$ So i tried the above suggestions and i still get the same error ### messages.So i tried the patch and got...
> 
> ┌──(kali㉿kali)-[~/Desktop/rtl8188eus-5.3.9] └─$ wget https://github.com/MrRob0-X/rtl8188eus/commit/5e3cfb4be0258dbca0ad553be5dc04d0829dbd04.patch && patch -p1 < *patch && rm -rf *patch --2022-12-23 22:39:22-- https://github.com/MrRob0-X/rtl8188eus/commit/5e3cfb4be0258dbca0ad553be5dc04d0829dbd04.patch Resolving github.com (github.com)... 140.82.121.3 Connecting to github.com (github.com)|140.82.121.3|:443... connected. HTTP request sent, awaiting response... 200 OK Length: 1113 (1.1K) [text/plain] Saving to: ‘5e3cfb4be0258dbca0ad553be5dc04d0829dbd04.patch.1’
> 
> [5e3cfb4](https://github.com/aircrack-ng/rtl8188eus/commit/5e3cfb4be0258dbca0ad553be5dc04d0829dbd04). 100%[===================================================================================>] 1.09K --.-KB/s in 0s
> 
> 2022-12-23 22:39:23 (11.1 MB/s) - ‘5e3cfb4be0258dbca0ad553be5dc04d0829dbd04.patch.1’ saved [1113/1113]
> 
> patching file Makefile Hunk #1 FAILED at 19. Hunk #2 FAILED at 226. 2 out of 2 hunks FAILED -- saving rejects to file Makefile.rej so tried the flag call and got...
> 
> ┌──(kali㉿kali)-[~/Desktop/rtl8188eus-5.3.9] └─$ grep -r "/include" Makefile EXTRA_CFLAGS += -I$(pwd)/$(src)/include
> 
> EXTRA_CFLAGS += -I$(srctree)/$(pwd)/$(src)/include EXTRA_CFLAGS: command not found

you broke Makefile, reclone rtl8188eus and do this
`wget https://github.com/aircrack-ng/rtl8188eus/commit/3a4ce2bde940b02a476577acf87087dacbffe34c.patch && patch -p1 < *patch && rm -rf *patch`

masharia-john, 2022-12-23:
Thank you....it works

anand-darkvision, 2022-12-31:
thanks bro it works

linkvapeluckyman, 2023-01-17:
make && sudo make install                                                                                                               
/bin/sh: 1: bc: not found
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/6.0.0-kali3-amd64/build M=/home/san/rtl8188eus  modules
make[1]: *** /lib/modules/6.0.0-kali3-amd64/build: No such file or directory.  Stop.
make: *** [Makefile:2065: modules] Error 2

linkvapeluckyman, 2023-01-17:
wget https://github.com/aircrack-ng/rtl8188eus/commit/3a4ce2bde940b02a476577acf87087dacbffe34c.patch && patch -p1 < *patch && rm -rf *patch
--2023-01-18 00:33:27--  https://github.com/aircrack-ng/rtl8188eus/commit/3a4ce2bde940b02a476577acf87087dacbffe34c.patch
Resolving github.com (github.com)... 20.207.73.82
Connecting to github.com (github.com)|20.207.73.82|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 1294 (1.3K) [text/plain]
Saving to: ‘3a4ce2bde940b02a476577acf87087dacbffe34c.patch’

3a4ce2bde940b02a476577acf87087dacbffe34c.patch              100%[=========================================================================================================================================>]   1.26K  --.-KB/s    in 0s      

2023-01-18 00:33:28 (45.1 MB/s) - ‘3a4ce2bde940b02a476577acf87087dacbffe34c.patch’ saved [1294/1294]

patching file Makefile
Hunk #1 FAILED at 19.
Hunk #2 FAILED at 226.
2 out of 2 hunks FAILED -- saving rejects to file Makefile.rej
patching file Makefile
Hunk #1 FAILED at 19.
Hunk #2 FAILED at 226.
2 out of 2 hunks FAILED -- saving rejects to file Makefile.rej

MrRob0-X, 2023-01-18:
> wget https://github.com/aircrack-ng/rtl8188eus/commit/3a4ce2bde940b02a476577acf87087dacbffe34c.patch && patch -p1 < *patch && rm -rf *patch --2023-01-18 00:33:27-- https://github.com/aircrack-ng/rtl8188eus/commit/3a4ce2bde940b02a476577acf87087dacbffe34c.patch Resolving github.com (github.com)... 20.207.73.82 Connecting to github.com (github.com)|20.207.73.82|:443... connected. HTTP request sent, awaiting response... 200 OK Length: 1294 (1.3K) [text/plain] Saving to: ‘3a4ce2bde940b02a476577acf87087dacbffe34c.patch’
> 
> [3a4ce2b](https://github.com/aircrack-ng/rtl8188eus/commit/3a4ce2bde940b02a476577acf87087dacbffe34c).patch 100%[=========================================================================================================================================>] 1.26K --.-KB/s in 0s
> 
> 2023-01-18 00:33:28 (45.1 MB/s) - ‘3a4ce2bde940b02a476577acf87087dacbffe34c.patch’ saved [1294/1294]
> 
> patching file Makefile Hunk #1 FAILED at 19. Hunk #2 FAILED at 226. 2 out of 2 hunks FAILED -- saving rejects to file Makefile.rej patching file Makefile Hunk #1 FAILED at 19. Hunk #2 FAILED at 226. 2 out of 2 hunks FAILED -- saving rejects to file Makefile.rej

No need to apply the patch, aready merged in repo

MrRob0-X, 2023-01-18:
> make && sudo make install /bin/sh: 1: bc: not found make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/6.0.0-kali3-amd64/build M=/home/san/rtl8188eus modules make[1]: *** /lib/modules/6.0.0-kali3-amd64/build: No such file or directory. Stop. make: *** [Makefile:2065: modules] Error 2

`sudo apt-get update && sudo apt-get install bc -y`

mrcodering, 2025-06-28:
> Thank you....it works

i'm using arch kernel 6.15.3 and itsnot working for me. Any idea?

gglluukk, 2025-06-28:
use builtin `rtl8xxxu` or try: https://github.com/gglluukk/rtl8188eus

cpermq, 2025-07-06:
### Error is from #include
In this part of rwt_cmd.c

#define _RTW_CMD_C_

#include <drv_types.h>
#include <hal_data.h>

#ifndef DBG_CMD_EXECUTE
	#define DBG_CMD_EXECUTE 0
#endif

It use <> to get headers from system include path,but the headers that it used is in the "include" folder in the source folder.
The complier couldn't get headers,so it causes error.
One of the way to solve the problem is copy all the headers to the core folder,and change <> to "" in all the .c and .h files.
But there are also some problems that I can't repair like "-Wimplicit-function-declaration".
This is my first comment,hope it will help you.