aircrack-ng/rtl8188eus #223: Fatal error: drv_types.h: No such file or directory

Med1ummm, 2022-12-20:
Hey Guys who can help me after i typing sudo make thats what happens 
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/6.0.0-kali5-amd64/build M=/home/kali/rtl8188eus  modules
make[1]: Entering directory '/usr/src/linux-headers-6.0.0-kali5-amd64'
  CC [M]  /home/kali/rtl8188eus/core/rtw_cmd.o
/home/kali/rtl8188eus/core/rtw_cmd.c:17:10: fatal error: drv_types.h: No such file or directory
   17 | #include <drv_types.h>
      |          ^~~~~~~~~~~~~
compilation terminated.
make[2]: *** [/usr/src/linux-headers-6.0.0-kali5-common/scripts/Makefile.build:254: /home/kali/rtl8188eus/core/rtw_cmd.o] Error 1
make[1]: *** [/usr/src/linux-headers-6.0.0-kali5-common/Makefile:1876: /home/kali/rtl8188eus] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-6.0.0-kali5-amd64'
make: *** [Makefile:2062: modules] Error 2

scucu, 2022-12-21:
Try `sed -i 's/$(srctree)\/$(src)/$(pwd)\/$(src)/'g Makefile`

Med1ummm, 2022-12-21:
> Try `sed -i 's/$(srctree)\/$(src)/$(pwd)\/$(src)/'g Makefile`

nop . come back with error 

make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/6.0.0-kali5-amd64/build M=/home/kali/rtl8188eus  modules
make[1]: Entering directory '/usr/src/linux-headers-6.0.0-kali5-amd64'
  CC [M]  /home/kali/rtl8188eus/core/rtw_cmd.o
/home/kali/rtl8188eus/core/rtw_cmd.c:17:10: fatal error: drv_types.h: No such file or directory
   17 | #include <drv_types.h>
      |          ^~~~~~~~~~~~~
compilation terminated.
make[2]: *** [/usr/src/linux-headers-6.0.0-kali5-common/scripts/Makefile.build:254: /home/kali/rtl8188eus/core/rtw_cmd.o] Error 1
make[1]: *** [/usr/src/linux-headers-6.0.0-kali5-common/Makefile:1876: /home/kali/rtl8188eus] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-6.0.0-kali5-amd64'
make: *** [Makefile:2062: modules] Error 2

scenegrapher, 2022-12-21:
wget https://github.com/MrRob0-X/rtl8188eus/commit/5e3cfb4be0258dbca0ad553be5dc04d0829dbd04.patch && patch -p1 < *patch && rm -rf *patch

Apply this patch

Med1ummm, 2022-12-21:
https://github.com/MrRob0-X/rtl8188eus/commit/5e3cfb4be0258dbca0ad553be5dc04d0829dbd04.patch && patch -p1 < *patch && rm -rf *patch
--2022-12-21 16:05:20--  https://github.com/MrRob0-X/rtl8188eus/commit/5e3cfb4be0258dbca0ad553be5dc04d0829dbd04.patch
Resolving github.com (github.com)... 140.82.121.4
Connecting to github.com (github.com)|140.82.121.4|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 1113 (1.1K) [text/plain]
Saving to: ‘5e3cfb4be0258dbca0ad553be5dc04d0829dbd04.patch’

5e3cfb4be0258dbca0ad553be5dc04d0829dbd 100%[===========================================================================>]   1.09K  --.-KB/s    in 0s      

2022-12-21 16:05:21 (16.3 MB/s) - ‘5e3cfb4be0258dbca0ad553be5dc04d0829dbd04.patch’ saved [1113/1113]

can't find file to patch at input line 15
Perhaps you used the wrong -p or --strip option?
The text leading up to this was:
--------------------------
|From 5e3cfb4be0258dbca0ad553be5dc04d0829dbd04 Mon Sep 17 00:00:00 2001
|From: MrRob0-X <hanya8493@gmail.com>
|Date: Tue, 20 Dec 2022 13:14:03 +0000
|Subject: [PATCH] Makefile: set check for two way path's
|
|this will avoid `drv_types.h not found` error when building standalone or in kernel versions like 5.4
|---
| Makefile | 7 +++++++
| 1 file changed, 7 insertions(+)
|
|diff --git a/Makefile b/Makefile
|index 5f2f680..9c594f9 100755
|--- a/Makefile
|+++ b/Makefile
--------------------------
File to patch: install
install: No such file or directory
 **how i can fix that ?**

Med1ummm, 2022-12-21:
same story make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/6.0.0-kali5-amd64/build M=/home/kali/rtl8188eus  modules
make[1]: Entering directory '/usr/src/linux-headers-6.0.0-kali5-amd64'
  CC [M]  /home/kali/rtl8188eus/core/rtw_cmd.o
/home/kali/rtl8188eus/core/rtw_cmd.c:17:10: fatal error: drv_types.h: No such file or directory
   17 | #include <drv_types.h>
      |          ^~~~~~~~~~~~~
compilation terminated.
make[2]: *** [/usr/src/linux-headers-6.0.0-kali5-common/scripts/Makefile.build:254: /home/kali/rtl8188eus/core/rtw_cmd.o] Error 1
make[1]: *** [/usr/src/linux-headers-6.0.0-kali5-common/Makefile:1876: /home/kali/rtl8188eus] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-6.0.0-kali5-amd64'
make: *** [Makefile:2062: modules] Error 2

sensoredspecialist, 2022-12-22:
can the problem be with the kernel itself, which indeed here is version 6.0.0

MrRob0-X, 2022-12-22:
> https://github.com/MrRob0-X/rtl8188eus/commit/5e3cfb4be0258dbca0ad553be5dc04d0829dbd04.patch && patch -p1 < *patch && rm -rf *patch --2022-12-21 16:05:20-- https://github.com/MrRob0-X/rtl8188eus/commit/5e3cfb4be0258dbca0ad553be5dc04d0829dbd04.patch Resolving github.com (github.com)... 140.82.121.4 Connecting to github.com (github.com)|140.82.121.4|:443... connected. HTTP request sent, awaiting response... 200 OK Length: 1113 (1.1K) [text/plain] Saving to: ‘5e3cfb4be0258dbca0ad553be5dc04d0829dbd04.patch’
> 
> [5e3cfb4](https://github.com/aircrack-ng/rtl8188eus/commit/5e3cfb4be0258dbca0ad553be5dc04d0829dbd04) 100%[===========================================================================>] 1.09K --.-KB/s in 0s
> 
> 2022-12-21 16:05:21 (16.3 MB/s) - ‘5e3cfb4be0258dbca0ad553be5dc04d0829dbd04.patch’ saved [1113/1113]
> 
> ## can't find file to patch at input line 15
> Perhaps you used the wrong -p or --strip option?
> The text leading up to this was:
> |From [5e3cfb4](https://github.com/aircrack-ng/rtl8188eus/commit/5e3cfb4be0258dbca0ad553be5dc04d0829dbd04) Mon Sep 17 00:00:00 2001 |From: MrRob0-X [hanya8493@gmail.com](mailto:hanya8493@gmail.com) |Date: Tue, 20 Dec 2022 13:14:03 +0000 |Subject: [PATCH] Makefile: set check for two way path's |
> 
> this will avoid `drv_types.h not found` error when building standalone or in kernel versions like 5.4
> Makefile
> 1 file changed, 7 insertions(+)
> ## |
> |diff --git a/Makefile b/Makefile
> |index 5f2f680..9c594f9 100755
> |--- a/Makefile
> |+++ b/Makefile
> File to patch: install install: No such file or directory **how i can fix that ?**

do this and show output
`grep -r "/include" Makefile`

MrRob0-X, 2022-12-22:
> can the problem be with the kernel itself, which indeed here is version 6.0.0

it's the path, i opened a pr and waiting to merge the solution

Med1ummm, 2022-12-22:
> 

──(kali㉿kali)-[~/rtl8188eus]
└─$ grep -r "/include" Makefile
EXTRA_CFLAGS += -I$(pwd)/include

MrRob0-X, 2022-12-22:
> > 
> 
> ──(kali㉿kali)-[~/rtl8188eus] └─$ grep -r "/include" Makefile EXTRA_CFLAGS += -I$(pwd)/include

`sed -i 's/$(pwd)/$(srctree)\/$(src)/'g && rm -rf *patch && wget https://github.com/MrRob0-X/rtl8188eus/commit/5e3cfb4be0258dbca0ad553be5dc04d0829dbd04.patch && patch -p1 < *patch && rm -rf *patch`

Med1ummm, 2022-12-23:
> > > 
> > 
> > 
> > ──(kali㉿kali)-[~/rtl8188eus] └─$ grep -r "/include" Makefile EXTRA_CFLAGS += -I$(pwd)/include
> 
> `sed -i 's/$(pwd)/$(srctree)\/$(src)/'g && rm -rf *patch && wget https://github.com/MrRob0-X/rtl8188eus/commit/5e3cfb4be0258dbca0ad553be5dc04d0829dbd04.patch && patch -p1 < *patch && rm -rf *patch`

sed -i 's/$(pwd)/$(srctree)\/$(src)/'g && rm -rf *patch && wget https://github.com/MrRob0-X/rtl8188eus/commit/5e3cfb4be0258dbca0ad553be5dc04d0829dbd04.patch && patch -p1 < *patch && rm -rf *patch 
sed: no input files

MrRob0-X, 2022-12-23:
> sed -i 's/$(pwd)/$(srctree)/$(src)/'g && rm -rf *patch && wget https://github.com/MrRob0-X/rtl8188eus/commit/5e3cfb4be0258dbca0ad553be5dc04d0829dbd04.patch && patch -p1 < *patch && rm -rf *patch [...]

fixed it, check my comment again

Med1ummm, 2022-12-23:
> fixed it, check my comment again

thank you so much it's worked ;)

sensoredspecialist, 2022-12-25:
> > > 
> > 
> > 
> > ──(kali㉿kali)-[~/rtl8188eus] └─$ grep -r "/include" Makefile EXTRA_CFLAGS += -I$(pwd)/include
> 
> `sed -i 's/$(pwd)/$(srctree)\/$(src)/'g Makefile && rm -rf *patch && wget https://github.com/MrRob0-X/rtl8188eus/commit/5e3cfb4be0258dbca0ad553be5dc04d0829dbd04.patch && patch -p1 < *patch && rm -rf *patch`

Helped me a lot, Thank you!

el-aprendis24, 2022-12-27:
> > > > 
> > > 
> > > 
> > > ──(kali㉿kali)-[~/rtl8188eus] └─$ grep -r "/include" Makefile EXTRA_CFLAGS += -I$(pwd)/include
> > 
> > 
> > `sed -i 's/$(pwd)/$(srctree)\/$(src)/'g Makefile && rm -rf *patch && wget https://github.com/MrRob0-X/rtl8188eus/commit/5e3cfb4be0258dbca0ad553be5dc04d0829dbd04.patch && patch -p1 < *patch && rm -rf *patch`
> 
> me ayudo mucho, **gracias**

muchas gracias, tenia dos dias trabado en ese lado, tu solucion fue muy asectada.

fsantanderb, 2023-02-15:
patching file Makefile
Hunk #1 FAILED at 19.
Hunk #2 FAILED at 226.
2 out of 2 hunks FAILED -- saving rejects to file Makefile.rej

└─$ cat Makefile.rej 
```
--- Makefile
+++ Makefile
@@ -19,6 +19,8 @@ ifeq ($(GCC_VER_49),1)
 EXTRA_CFLAGS += -Wno-date-time # Fix compile error && warning on gcc 4.9 and later
 endif
 
+EXTRA_CFLAGS += -I$(src)/include
+#else
 EXTRA_CFLAGS += -I$(srctree)/$(src)/include
 
 EXTRA_LDFLAGS += --strip-debug
@@ -226,9 +228,14 @@ _HAL_INTFS_FILES :=        hal/hal_intf.o \
                        hal/led/hal_$(HCI_NAME)_led.o
 
 
+EXTRA_CFLAGS += -I$(src)/platform
+#else
 EXTRA_CFLAGS += -I$(srctree)/$(src)/platform
 _PLATFORM_FILES := platform/platform_ops.o
 
+EXTRA_CFLAGS += -I$(src)/hal/btc
+EXTRA_CFLAGS += -I$(src)/hal/phydm
+#else
 EXTRA_CFLAGS += -I$(srctree)/$(src)/hal/btc
 EXTRA_CFLAGS += -I$(srctree)/$(src)/hal/phydm
```