aircrack-ng/rtl8188eus #224: Build error for 6.1 kernel version

sergiall, 2022-12-28:
Have a nice day! I try to build this module v5.3.9 branch with MrRob0-X patches within the kernel tree, but get errors:

`
drivers/staging/rtl8188/os_dep/linux/os_intfs.c:1621:9: error: too many arguments to function ‘netif_napi_add’
 1621 |         netif_napi_add(ndev, &adapter->napi, rtw_recv_napi_poll, RTL_NAPI_WEIGHT);
      |         ^~~~~~~~~~~~~~
In file included from drivers/staging/rtl8188/include/osdep_service_linux.h:30,
                 from drivers/staging/rtl8188/include/osdep_service.h:50,
                 from drivers/staging/rtl8188/include/drv_types.h:26,
                 from drivers/staging/rtl8188/os_dep/linux/os_intfs.c:17:
./include/linux/netdevice.h:2562:1: note: declared here
 2562 | netif_napi_add(struct net_device *dev, struct napi_struct *napi,
`

Then i'm try to change the "netif_napi_add" defenition with "netif_napi_add_weight" and file "os_dep/linux/os_intfs.c" compiled without error, and module 8188eu too, but:

`
make KERNELRELEASE=6.1.1-lmk-060101-generic++ ARCH=x86 	KBUILD_BUILD_VERSION=060101.202212250000 -f ./Makefile
  CALL    scripts/checksyscalls.sh
  CC [M]  drivers/staging/rtl8188/os_dep/osdep_service.o
  LD [M]  drivers/staging/rtl8188/8188eu.o
  MODPOST Module.symvers
ERROR: modpost: "prandom_u32" [drivers/staging/rtl8188/8188eu.ko] undefined!
`
As i investigate, 6.1 kernel have some changes in {include/linux/netdevice.h, include/linux/prandom.h}, but i dont know how resolve this causes here, please help me (__

hussinHelal, 2023-01-01:
i have the same issue to, looks like we are going to have errors every kernel update

jinjiachen, 2023-01-23:
i was compiling the  different module: rtl8188eus. but i have run into the similar problem. My kernel version is 6.1.7-arch1-1 and my console output is following:
`/tmp/rtl8188eus_new/rtl8188eus_new/os_dep/osdep_service.c: 在函数‘rtw_random32’中:
/tmp/rtl8188eus_new/rtl8188eus_new/os_dep/osdep_service.c:2582:16: 错误：implicit declaration of function ‘prandom_u32’; did you mean ‘prandom_u32_max’? [-Werror=implicit-function-declaration]
 2582 |         return prandom_u32();
      |                ^~~~~~~~~~~
      |                prandom_u32_max
cc1：有些警告被当作是错误
make[2]: *** [scripts/Makefile.build:250: /tmp/rtl8188eus_new/rtl8188eus_new/os_dep/osdep_service.o] Error 1
make[1]: *** [Makefile:1992: /tmp/rtl8188eus_new/rtl8188eus_new] Error 2
make[1]: 离开目录“/usr/lib/modules/6.1.7-arch1-1/build”`

Masrkai, 2023-01-26:
even from Aur build using paru got this error :-
 Bad return status for module build on kernel: 6.1.6-zen1-1-zen (x86_64)
Consult /var/lib/dkms/8188eu/5.3.9.r184.f8ead57/build/make.log for more information
idk wth changes every kernel update wth is wrong xD

gglluukk, 2023-01-26:
check pull requests, there are waiting for accepts patches

whoim2, 2023-02-17:
> i was compiling the different module: rtl8188eus. but i have run into the similar problem. My kernel version is 6.1.7-arch1-1 and my console output is following: [...]

in osdep_service.c:
`#ifdef PLATFORM_LINUX
#if (LINUX_VERSION_CODE >= KERNEL_VERSION(6, 0, 0))
        return get_random_u32();
#elif (LINUX_VERSION_CODE >= KERNEL_VERSION(3, 8, 0))
        return prandom_u32();
#elif (LINUX_VERSION_CODE <= KERNEL_VERSION(2, 6, 18))
        u32 random_int;
        get_random_bytes(&random_int , 4);
        return random_int;
`

Gabrycraft20, 2023-02-24:
are there any news?

hazem22123, 2023-02-25:
### Same error on kali linux latest kernel

┌──(root㉿kali)-[/home/afrakosh/tools/rtl8188eus]
└─# uname -a            
Linux kali 6.1.0-kali5-amd64 #1 SMP PREEMPT_DYNAMIC Debian 6.1.12-1kali1 (2023-02-20) x86_64 GNU/Linux

/home/afrakosh/tools/rtl8188eus/os_dep/osdep_service.c: In function ‘rtw_change_ifname’:
/home/afrakosh/tools/rtl8188eus/os_dep/osdep_service.c:2454:28: warning: passing argument 1 of ‘_rtw_memcpy’ discards ‘const’ qualifier from pointer target type [-Wdiscarded-qualifiers]
 2454 |         _rtw_memcpy(pnetdev->dev_addr, adapter_mac_addr(padapter), ETH_ALEN);
      |                     ~~~~~~~^~~~~~~~~~
/home/afrakosh/tools/rtl8188eus/os_dep/osdep_service.c:899:24: note: expected ‘void *’ but argument is of type ‘const unsigned char *’
  899 | void _rtw_memcpy(void *dst, const void *src, u32 sz)
      |                  ~~~~~~^~~
/home/afrakosh/tools/rtl8188eus/os_dep/osdep_service.c: In function ‘rtw_random32’:
/home/afrakosh/tools/rtl8188eus/os_dep/osdep_service.c:2578:16: error: implicit declaration of function ‘prandom_u32’; did you mean ‘prandom_u32_max’? [-Werror=implicit-function-declaration]
 2578 |         return prandom_u32();
      |                ^~~~~~~~~~~
      |                prandom_u32_max
At top level:
cc1: note: unrecognized command-line option ‘-Wno-sometimes-uninitialized’ may have been intended to silence earlier diagnostics
cc1: some warnings being treated as errors
make[2]: *** [/usr/src/linux-headers-6.1.0-kali5-common/scripts/Makefile.build:255: /home/afrakosh/tools/rtl8188eus/os_dep/osdep_service.o] Error 1
make[1]: *** [/usr/src/linux-headers-6.1.0-kali5-common/Makefile:2030: /home/afrakosh/tools/rtl8188eus] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-6.1.0-kali5-amd64'
make: *** [Makefile:2065: modules] Error 2

gglluukk, 2023-02-26:
try this for 6.1.x kernels (still in pull requests, not merged yet):
https://github.com/gglluukk/rtl8188eus

zeshankhalid22, 2023-03-06:
it works on lts kernel with some warnings.

Waelhh, 2023-03-06:
> try this for 6.1.x kernels (still in pull requests, not merged yet): https://github.com/gglluukk/rtl8188eus

i don't know how to thanks you, more than million thanks

charon19d, 2023-03-11:
> try this for 6.1.x kernels (still in pull requests, not merged yet): https://github.com/gglluukk/rtl8188eus

you are a god for me yrrr litrally i spend 4 days to solve it but today i see your conservation i solved really your are good for mee

bragos123123, 2023-03-14:
> 

How to fix this problem

bragos123123, 2023-03-14:
> ### Same error on kali linux latest kernel
> ┌──(root㉿kali)-[/home/afrakosh/tools/rtl8188eus] └─# uname -a Linux kali 6.1.0-kali5-amd64 #1 SMP PREEMPT_DYNAMIC Debian 6.1.12-1kali1 (2023-02-20) x86_64 GNU/Linux
> 
> ### Output
> /home/afrakosh/tools/rtl8188eus/os_dep/osdep_service.c: In function ‘rtw_change_ifname’: /home/afrakosh/tools/rtl8188eus/os_dep/osdep_service.c:2454:28: warning: passing argument 1 of ‘_rtw_memcpy’ discards ‘const’ qualifier from pointer target type [-Wdiscarded-qualifiers] 2454 | _rtw_memcpy(pnetdev->dev_addr, adapter_mac_addr(padapter), ETH_ALEN); | ~~~~~~~^~~~~~~~~~ /home/afrakosh/tools/rtl8188eus/os_dep/osdep_service.c:899:24: note: expected ‘void *’ but argument is of type ‘const unsigned char *’ 899 | void _rtw_memcpy(void *dst, const void *src, u32 sz) | ~~~~~~^~~ /home/afrakosh/tools/rtl8188eus/os_dep/osdep_service.c: In function ‘rtw_random32’: /home/afrakosh/tools/rtl8188eus/os_dep/osdep_service.c:2578:16: error: implicit declaration of function ‘prandom_u32’; did you mean ‘prandom_u32_max’? [-Werror=implicit-function-declaration] 2578 | return prandom_u32(); | ^~~~~~~~~~~ | prandom_u32_max At top level: cc1: note: unrecognized command-line option ‘-Wno-sometimes-uninitialized’ may have been intended to silence earlier diagnostics cc1: some warnings being treated as errors make[2]: *** [/usr/src/linux-headers-6.1.0-kali5-common/scripts/Makefile.build:255: /home/afrakosh/tools/rtl8188eus/os_dep/osdep_service.o] Error 1 make[1]: *** [/usr/src/linux-headers-6.1.0-kali5-common/Makefile:2030: /home/afrakosh/tools/rtl8188eus] Error 2 make[1]: Leaving directory '/usr/src/linux-headers-6.1.0-kali5-amd64' make: *** [Makefile:2065: modules] Error 2

How to fix this?

bragos123123, 2023-03-14:
How to fix this?

Syncure, 2023-04-11:
> try this for 6.1.x kernels (still in pull requests, not merged yet): https://github.com/gglluukk/rtl8188eus

Thanks a lot!!!

PanDa-Dax, 2023-04-12:
I get the following error in the screenshot 
[image]

Terracuk, 2023-11-10:
I get same error as above with v6.1.21 can anyone help? Or release a v6.1.21 version please?

gglluukk, 2023-11-10:
@Terracuk 
```sudo make install```

Terracuk, 2023-11-10:
I get:

Install: cannot stat 8188eu.ko : no such file or directory 

Make *** makefile install error 1

gglluukk, 2023-11-10:
i meant that error which @PanDa-Dax had.

if you're unable to compile: update your distro, delete driver's source and download it again.

other option is to use rtl8xxxu coming with newer kernels which supports monitor mode too.

markyangcc, 2024-01-16:
Since kernel API changed, some not modifoed files may failed when building.

Replace `netif_napi_add` to `netif_napi_add_weight` to fix it

https://git.kernel.org/pub/scm/linux/kernel/git/netdev/net-next.git/commit/?id=899b8cd0d392

gglluukk, 2024-01-16:
even more: from kernel version at least 6.0 call to `netif_napi_add()` and/or `netif_napi_add_weight()` can be safely removed since it does nothing.

see: https://elixir.bootlin.com/linux/v6.0.19/source/include/linux/netdevice.h#L2553

but let it keep as it is for being compatible with older versions *_^

Pytholearn, 2024-06-21:
PLS Help Me Bax
=======
┌──(root㉿kali)-[/home/kali/rtl8188eus]
└─# uname -r 
6.1.0-kali9-amd64
=======