aircrack-ng/rtl8188eus #228: Compile Error

bennyHPR, 2023-02-12:
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.18.0-kali5-amd64/build M=/home/hpr/rtl8188eus  modules
make[1]: *** /lib/modules/5.18.0-kali5-amd64/build: No such file or directory.  Stop.
make: *** [Makefile:2058: modules] Error 2

Please Help

fawaz7, 2023-02-14:
I have the same problem too, I'm just waiting for the pull request to be merged.. hopefully soon :)

fawaz7, 2023-02-14:
I had the same error fixed, by following this guide here : 
https://github.com/davidbombal/Kali-Linux/blob/main/TP-Link%20TL-WN722N%20adapter

RadoslavL, 2023-02-17:
You must have linux-headers installed on your system.

bennyHPR, 2023-02-20:
> I had the same error, fixed by following this guide here : https://github.com/davidbombal/Kali-Linux/blob/main/TP-Link%20TL-WN722N%20adapter

That's Work broo....Thanks a lot.

actually I've written line sudo apt-get install linux-headers-`uname -r` but with bracket like this (-uname -r) from another post before, but no luck.

closed 2023-02-20 by bennyHPR