aircrack-ng/rtl8188eus #232: rtl8188eus driver compile error

satyauday112, 2023-02-27:
Hi, Iam trying to install realtek rtl8188eus driver in kali lilnux. But Iam getting the below error:
[image]

/home/Satya/rtl8188eus_new/os_dep/osdep_service.c: In function ‘rtw_random32’:
/home/Satya/rtl8188eus_new/os_dep/osdep_service.c:2582:16: error: too few arguments to function ‘prandom_u32_max’
 2582 |         return prandom_u32_max();
      |                ^~~~~~~~~~~~~~~
In file included from /usr/src/linux-headers-6.1.0-kali5-common/include/linux/random.h:172,
                 from /usr/src/linux-headers-6.1.0-kali5-common/include/linux/nodemask.h:97,
                 from /usr/src/linux-headers-6.1.0-kali5-common/include/linux/mmzone.h:17,
                 from /usr/src/linux-headers-6.1.0-kali5-common/include/linux/gfp.h:7,
                 from /usr/src/linux-headers-6.1.0-kali5-common/include/linux/umh.h:4,
                 from /usr/src/linux-headers-6.1.0-kali5-common/include/linux/kmod.h:9,
                 from /usr/src/linux-headers-6.1.0-kali5-common/include/linux/module.h:17,
                 from /home/Satya/rtl8188eus_new/include/basic_types.h:76,
                 from /home/Satya/rtl8188eus_new/include/drv_types.h:25,
                 from /home/Satya/rtl8188eus_new/os_dep/osdep_service.c:19:
/usr/src/linux-headers-6.1.0-kali5-common/include/linux/prandom.h:27:19: note: declared here
   27 | static inline u32 prandom_u32_max(u32 ep_ro)
      |                   ^~~~~~~~~~~~~~~
/home/Satya/rtl8188eus_new/os_dep/osdep_service.c:2595:1: error: control reaches end of non-void function [-Werror=return-type]
 2595 | }
      | ^
cc1: some warnings being treated as errors
make[2]: *** [/usr/src/linux-headers-6.1.0-kali5-common/scripts/Makefile.build:255: /home/Satya/rtl8188eus_new/os_dep/osdep_service.o] Error 1
make[1]: *** [/usr/src/linux-headers-6.1.0-kali5-common/Makefile:2030: /home/Satya/rtl8188eus_new] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-6.1.0-kali5-amd64'
make: *** [Makefile:2059: modules] Error 2

d4rkd3v1l, 2023-03-01:
Same issue here.
This fix worked for me: https://github.com/aircrack-ng/rtl8188eus/issues/224#issuecomment-1445275324

Therefore this issue here may actually be a duplicate.

RavitejaPegada, 2023-03-01:
[image]

satyauday112, 2023-03-02:
@R4vitej4 Go to issue #224 and find the comment by gglluukk. That worked for me.

closed 2023-03-02 by satyauday112