aircrack-ng/rtl8188eus #233: Kernel 6.2 make error

Masrkai, 2023-03-06:
[image]

Masrkai, 2023-03-06:
[image]

gglluukk, 2023-03-07:
try this for 6.2 kernel (still in pull requests, not merged yet):
https://github.com/gglluukk/rtl8188eus

Masrkai, 2023-03-19:
> try this for 6.2 kernel (still in pull requests, not merged yet): https://github.com/gglluukk/rtl8188eus
@gglluukk 
it worked for me in 6.1 thanks to your fork and until 6.2.6 kernel with no errors now 6.2.7 have a problem that the device not even showing

gglluukk, 2023-03-20:
it works:
```
lab ~/wifi # uname -a
Linux lab 6.2.7-LAB #1 SMP PREEMPT_DYNAMIC Mon Mar 20 18:53:55 MSK 2023 aarch64 GNU/Linux
lab ~/wifi # iwlist wlan scan | grep Freq | head
                    Frequency:2.412 GHz (Channel 1)
                    Frequency:2.412 GHz (Channel 1)
                    Frequency:2.412 GHz (Channel 1)
                    Frequency:2.422 GHz (Channel 3)
                    Frequency:2.412 GHz (Channel 1)
                    Frequency:2.412 GHz (Channel 1)
                    Frequency:2.412 GHz (Channel 1)
                    Frequency:2.417 GHz (Channel 2)
                    Frequency:2.412 GHz (Channel 1)
                    Frequency:2.422 GHz (Channel 3)
lab ~/wifi # ethtool -i wlan
driver: 8188eu
version: 6.2.7-LAB
firmware-version: 
expansion-rom-version: 
bus-info: 2-3:1.0
supports-statistics: no
supports-test: no
supports-eeprom-access: no
supports-register-dump: no
supports-priv-flags: no
```
make sure you did: 
```
make clean && make && make install
```

closed 2024-02-17 by Masrkai