aircrack-ng/rtl8188eus #234: everything is done but stuck on this now

Silentkiller987, 2023-03-09:
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/6.1.0-kali5-amd64/build M=/home/kali/Desktop/rtl8188fu  modules
make[1]: Entering directory '/usr/src/linux-headers-6.1.0-kali5-amd64'
  CC [M]  /home/kali/Desktop/rtl8188fu/os_dep/osdep_service.o
/home/kali/Desktop/rtl8188fu/os_dep/osdep_service.c: In function ‘rtw_random32’:
/home/kali/Desktop/rtl8188fu/os_dep/osdep_service.c:2350:16: error: implicit declaration of function ‘prandom_u32’; did you mean ‘prandom_u32_max’? [-Werror=implicit-function-declaration]
 2350 |         return prandom_u32();
      |                ^~~~~~~~~~~
      |                prandom_u32_max
cc1: some warnings being treated as errors
make[2]: *** [/usr/src/linux-headers-6.1.0-kali5-common/scripts/Makefile.build:255: /home/kali/Desktop/rtl8188fu/os_dep/osdep_service.o] Error 1
make[1]: *** [/usr/src/linux-headers-6.1.0-kali5-common/Makefile:2030: /home/kali/Desktop/rtl8188fu] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-6.1.0-kali5-amd64'
make: *** [Makefile:470: modules] Error 2

closed 2023-03-13 by Silentkiller987

santiagomd11, 2023-03-15:
I'm having the same issue, did you manage to solve it?

HackerTuhin, 2023-05-03:
@santiagomd11 thanks !!!!!!! it worked

lewixlabs, 2023-08-03:
I confirm, this fork works!