aircrack-ng/rtl8188eus #242: make error

FedericoFedeFede, 2023-04-07:
```
/bin/sh: 1: bc: not found
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/6.1.0-kali5-amd64/build M=/home/kali/Desktop/rtl8188eus-5.3.9  modules
make[1]: *** /lib/modules/6.1.0-kali5-amd64/build: No such file or directory.  Stop.
make: *** [Makefile:2065: modules] Error 2
                                           
```

m4nj4pe, 2023-04-10:
You do not have bc installed. Try:
sudo apt install bc -y

Arenas82, 2023-04-16:
If you speak spanish i found a video that could help

https://www.youtube.com/watch?v=ekDHoW14Rb4

I followed the step by step and finally got the solution for the monitor mode

AbhiJohnv, 2023-05-17:
Try this repository bro: https://github.com/varkmarker/Tp-linkWN722NV2.git