aircrack-ng/rtl8188eus #244: rtl8188eus driver installing issue

masudazad, 2023-04-25:
/home/masud/Desktop/realtek-rtl8188eus-dkms/os_dep/osdep_service.c: In function ‘rtw_change_ifname’:
/home/masud/Desktop/realtek-rtl8188eus-dkms/os_dep/osdep_service.c:2454:28: warning: passing argument 1 of ‘_rtw_memcpy’ discards ‘const’ qualifier from pointer target type [-Wdiscarded-qualifiers]
 2454 |         _rtw_memcpy(pnetdev->dev_addr, adapter_mac_addr(padapter), ETH_ALEN);
      |                     ~~~~~~~^~~~~~~~~~
/home/masud/Desktop/realtek-rtl8188eus-dkms/os_dep/osdep_service.c:899:24: note: expected ‘void *’ but argument is of type ‘const unsigned char *’
  899 | void _rtw_memcpy(void *dst, const void *src, u32 sz)
      |                  ~~~~~~^~~
/home/masud/Desktop/realtek-rtl8188eus-dkms/os_dep/osdep_service.c: In function ‘rtw_random32’:
/home/masud/Desktop/realtek-rtl8188eus-dkms/os_dep/osdep_service.c:2578:16: error: implicit declaration of function ‘prandom_u32’; did you mean ‘prandom_u32_max’? [-Werror=implicit-function-declaration]
 2578 |         return prandom_u32();
      |                ^~~~~~~~~~~
      |                prandom_u32_max
At top level:
cc1: note: unrecognized command-line option ‘-Wno-sometimes-uninitialized’ may have been intended to silence earlier diagnostics
cc1: some warnings being treated as errors
make[2]: *** [/usr/src/linux-headers-6.1.0-kali7-common/scripts/Makefile.build:255: /home/masud/Desktop/realtek-rtl8188eus-dkms/os_dep/osdep_service.o] Error 1
make[1]: *** [/usr/src/linux-headers-6.1.0-kali7-common/Makefile:2037: /home/masud/Desktop/realtek-rtl8188eus-dkms] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-6.1.0-kali7-amd64'
make: *** [Makefile:2065: modules] Error 2

masudazad, 2023-04-25:
i have used the same repo for 5.10 linux but its not working here. i have installed all the required packeges but still the same issue.

BLHodder, 2023-04-30:
I have the exact same issues, tried new vm and everything but doesn't seem to like it, tried adjusting the files to the supposed fixes given in the error but no luck

gglluukk, 2023-04-30:
try this for newer 6.x kernel (still in pull requests, not merged yet):
https://github.com/gglluukk/rtl8188eus

BLHodder, 2023-05-04:
Legend

Harmandeep01, 2023-05-05:
> try this for newer 6.x kernel (still in pull requests, not merged yet): https://github.com/gglluukk/rtl8188eus

Bro Thank you so much , this was huge headache for me since morning thanks a lot . May GOD Bless You!

sunr0e, 2023-05-12:
bingo！ thank you so much bro

diehard88, 2023-09-11:
@gglluukk 

what is the difference between aircrack-ng/rtl8188eus and gglluukk/rtl8188eus?

gglluukk, 2023-09-11:
you can see pull request with all the changes here:
https://github.com/aircrack-ng/rtl8188eus/pull/225

diehard88, 2023-10-20:
> you can see pull request with all the changes here: #225

we need to recompile for new kernel releases, correct?

SteamPunk89079, 2024-07-25:
can someone please make a recap of what needs to be done ? i am facing the same problem and can t find an answer .