aircrack-ng/rtl8188eus #246: how i fix it

Ahmed-idrees, 2023-05-04:
┌──(edree㉿kali)-[~]
└─$ cd Downloads 
                                                                                
┌──(edree㉿kali)-[~/Downloads]
└─$ cd rtl8188eus
                                                                                
┌──(edree㉿kali)-[~/Downloads/rtl8188eus]
└─$ make && sudo make install

make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/6.1.0-kali7-amd64/build M=/home/edree/Downloads/rtl8188eus  modules
make[1]: Entering directory '/usr/src/linux-headers-6.1.0-kali7-amd64'
  CC [M]  /home/edree/Downloads/rtl8188eus/os_dep/osdep_service.o
/home/edree/Downloads/rtl8188eus/os_dep/osdep_service.c: In function ‘rtw_change_ifname’:
/home/edree/Downloads/rtl8188eus/os_dep/osdep_service.c:2454:28: warning: passing argument 1 of ‘_rtw_memcpy’ discards ‘const’ qualifier from pointer target type [-Wdiscarded-qualifiers]
 2454 |         _rtw_memcpy(pnetdev->dev_addr, adapter_mac_addr(padapter), ETH_ALEN);
      |                     ~~~~~~~^~~~~~~~~~
/home/edree/Downloads/rtl8188eus/os_dep/osdep_service.c:899:24: note: expected ‘void *’ but argument is of type ‘const unsigned char *’
  899 | void _rtw_memcpy(void *dst, const void *src, u32 sz)
      |                  ~~~~~~^~~
/home/edree/Downloads/rtl8188eus/os_dep/osdep_service.c: In function ‘rtw_random32’:
/home/edree/Downloads/rtl8188eus/os_dep/osdep_service.c:2578:16: error: implicit declaration of function ‘prandom_u32’; did you mean ‘prandom_u32_max’? [-Werror=implicit-function-declaration]
 2578 |         return prandom_u32();
      |                ^~~~~~~~~~~
      |                prandom_u32_max
At top level:
cc1: note: unrecognized command-line option ‘-Wno-sometimes-uninitialized’ may have been intended to silence earlier diagnostics
cc1: some warnings being treated as errors
make[2]: *** [/usr/src/linux-headers-6.1.0-kali7-common/scripts/Makefile.build:255: /home/edree/Downloads/rtl8188eus/os_dep/osdep_service.o] Error 1
make[1]: *** [/usr/src/linux-headers-6.1.0-kali7-common/Makefile:2037: /home/edree/Downloads/rtl8188eus] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-6.1.0-kali7-amd64'
make: *** [Makefile:2065: modules] Error 2

mrnone-asr, 2023-05-16:
1. make sure you are using latest kali.
2. if still facing issue try using older version of driver

AbhiJohnv, 2023-05-17:
Try this repository bro: https://github.com/varkmarker/Tp-linkWN722NV2.git

kamesh7773, 2024-08-03:
Use this Kali Linux ISO File https://drive.google.com/file/d/1XvstnTnKlKm7lMqZbrff7h_qD67DNA4e/view?usp=sharing