aircrack-ng/rtl8188eus #256: i am faccing an issue can someone help me ?

C549, 2023-07-06:
home/kali/rtl8188eus_new/os_dep/osdep_service.c: In function ‘rtw_random32’:
/home/kali/rtl8188eus_new/os_dep/osdep_service.c:2582:16: error: implicit declaration of function ‘prandom_u32’; did you mean ‘prandom_u32_max’? [-Werror=implicit-function-declaration]
 2582 |         return prandom_u32();
      |                ^~~~~~~~~~~
      |                prandom_u32_max
cc1: some warnings being treated as errors
make[2]: *** [/usr/src/linux-headers-6.1.0-kali9-common/scripts/Makefile.build:255: /home/kali/rtl8188eus_new/os_dep/osdep_service.o] Error 1
make[1]: *** [/usr/src/linux-headers-6.1.0-kali9-common/Makefile:2037: /home/kali/rtl8188eus_new] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-6.1.0-kali9-amd64'
make: *** [Makefile:2059: modules] Error 2

gglluukk, 2023-07-06:
try this for newer kernel (still in pull requests, not merged yet):
https://github.com/gglluukk/rtl8188eus

C549, 2023-07-17:
Thank you @gglluukk for the help it worked

Mrpai2318, 2023-07-25:
[image]

Still getting the same error using your file as well please help @gglluukk

gglluukk, 2023-07-25:
@Mrpai2318 you're not using https://github.com/gglluukk/rtl8188eus since error related to `prandom_u32()` was fixed there

Mrpai2318, 2023-07-25:
[image]
Even now it is not working @gglluukk

gglluukk, 2023-07-25:
@Mrpai2318 errors you're getting are not connected to the driver, so please try to request help at other resources.

in your previous 'errors' you've been compiling under kernel version **6.3**  while trying to install under **6.1** kernel.

as well as you have to:
```
apt --yes install build-essential linux-headers-`uname -r`
```

Mrpai2318, 2023-07-25:
[image]
lsusb shows this soo does this mean i have the driver already

Mrpai2318, 2023-07-25:
Everything Works now. Thank you so much @gglluukk