aircrack-ng/rtl8188eus #257: Unknown symbol "net_log" (err 0)

AzeAstro, 2023-07-16:
I tried to compile rtl8188eus driver and while compiling I got 1 warning. Here is the full output(output was big): https://pastebin.com/Bh0ZL4Mc

Even if I got this warning, I still got 8188eu.ko and when I tried to load it to my Android, I got this error: `Unknown symbol "net_log" (err 0)`

Here you can find my kernel source: https://github.com/AzeAstro/samsung_kernel_sm7125/

Strange thing is that net_log function is defined both in header file (‎`include/linux/netlog.h‎`) and defined as function(`net/core/net_ipc_log.c`)

I thought that it might be missing somewhere and ran `grep -rnw . -e "netlog.h"` in the driver source directory and here is the output: https://pastebin.com/B52Gtn3L It is defined almost in every file.

I know it is more likely issue with kernel instead of driver but still.
Why this error is happening and how can I fix this?

gglluukk, 2023-07-16:
`Unknown symbol "function..."` in kernel log usually means for that function is missing or not exported (alike private inside the kernel).
I'd comment out any calls for `net_log` inside drivers' source and see what's going on next

AzeAstro, 2023-07-16:
> `Unknown symbol "function..."` in kernel log usually means for that function is missing or not exported (alike private inside the kernel).
> 
> Also you might missing `#include linux/netlog.h`
> 
> I'd try to add `include` or comment out any calls for `net_log` inside driver's source and see what's going on next

Strange thing is that there is no call for `net_log` function in driver source.
None. Completely zero
And the `netlog.h` only mentioned in *.cmd files.
Any possible ideas about where exactly it might happen?

AzeAstro, 2023-07-16:
Alright, I found out that I can analyze module file using `objdump`. I disassembled the module file. You can find it at attachments.
`net_log` only gets called in `netdev_br_init` and it is defined in `os_dep/linux/os_initfs.c`
[objdump.txt](https://github.com/aircrack-ng/rtl8188eus/files/12063413/objdump.txt)

gglluukk, 2023-07-17:
btw, you're using some other driver because this one doesn't have any net_log() calls as well as vanilla kernel doesn't. 
it must be some samsung's addon to kernel's sources

gglluukk, 2023-07-17:
oh, didn't read your comment carefully since you found that there is no `net_log()` call in this source.
in that case i'd rename calls to `netdev_br_init()` in driver's sources to `my_netdev_br_init()` and add `my_netdev_br_init()` function to driver's sources as in:
https://elixir.bootlin.com/linux/v5.19.17/source/drivers/staging/r8188eu/os_dep/os_intfs.c#L600 :
```
void my_netdev_br_init(struct net_device *netdev)
{
        struct adapter *adapter = (struct adapter *)rtw_netdev_priv(netdev);

        rcu_read_lock();

        if (rcu_dereference(adapter->pnetdev->rx_handler_data)) {
                struct net_device *br_netdev;
                struct net *devnet = NULL;

                devnet = dev_net(netdev);
                br_netdev = dev_get_by_name(devnet, CONFIG_BR_EXT_BRNAME);
                if (br_netdev) {
                        memcpy(adapter->br_mac, br_netdev->dev_addr, ETH_ALEN);
                        dev_put(br_netdev);
                } else {
                        pr_info("%s()-%d: dev_get_by_name(%s) failed!",
                                __func__, __LINE__, CONFIG_BR_EXT_BRNAME);
                }
        }
        adapter->ethBrExtInfo.addPPPoETag = 1;

        rcu_read_unlock();
}

```

but you should use sources for `netdev_br_init()` function from samsung's kernel with version your device running but without that broken(?) `net_log()` call

gglluukk, 2023-07-17:
or define your own version of `net_log()` in driver's source <-- that's most easiest way for trying to fix it

AzeAstro, 2023-07-17:
> in that case i'd rename calls to `netdev_br_init()` in driver's sources to `my_netdev_br_init()` and add `my_netdev_br_init()` function to driver's sources as in: [...]

Thank you for your suggestion about making own `netdev_br_init()` function or just renaming it. Yesterday I asked this same question on stack overflow and comment from user named "Tsyvarev" worked. I took a look at file where `net_log` function got defined but not exported as symbol. Exporting that function as GPL symbol worked.
Here you can find the post:
https://stackoverflow.com/questions/76694952/unknown-symbol-net-log-err-0

closed 2023-07-17 by AzeAstro