aircrack-ng/rtl8188eus #258: Make error nethunter

Slizakjan, 2023-08-09:
Hi, I am using nethunter kernel version 4.19.111 and I get this error when using make command:

# make
make ARCH=arm64 CROSS_COMPILE= -C /lib/modules/4.19.111-26004678/build M=/root/rtl8188eus modules
make[1]: *** /lib/modules/4.19.111-26004678/build: No such file or directory. Stop.
make: *** [Makefile:2065: modules] Error 2

Phone is: Samsung galaxy A12

Ahmedzero800, 2023-08-09:
U need a custom kernal for this to work
Search about how to compile a nethunter kernal for ur device

Slizakjan, 2023-08-10:
> U need a custom kernal for this to work
> Search about how to compile a nethunter kernal for ur device 

Ok, thanks

lonzo00, 2023-10-08:
Same problem 

make ARCH=arm64 CROSS_COMPILE= -C /lib/modules/5.4.210-qgki-26213204-abG990BXXS6EWG9/build M=/root/rtl8188eus  modules
make[1]: *** /lib/modules/5.4.210-qgki-26213204-abG990BXXS6EWG9/build: No such file or directory.  Stop.
make: *** [Makefile:2065: modules] Error 2

I use à Samsung S21 SE

mahadidot, 2023-10-24:
i also have same issue.
uname -r
4.14.206-NetErnels-v7-NYSP
--------
cd rtl8188eus
make
make ARCH=arm64 CROSS_COMPILE= -C /lib/modules/4.14.206-NetErnels-v7-NYSP/build M=/rtl8188eus  modules
make[1]: Entering directory '/usr/lib/modules/4.14.206-NetErnels-v7-NYSP/build'
make[1]: *** No rule to make target 'modules'.  Stop.
make[1]: Leaving directory '/usr/lib/modules/4.14.206-NetErnels-v7-NYSP/build'
make: *** [Makefile:2065: modules] Error 2

my device : redmi note 10

if anyone fixed this problem help me. how i can fix for my device.