aircrack-ng/rtl8188eus #259: Make Error Kali kernal 6.3.0-kali1-amd64

Angu0083, 2023-08-15:
CC [M]  /root/rtl8188eus/core/rtw_beamforming.o
  CC [M]  /root/rtl8188eus/core/rtw_odm.o
  CC [M]  /root/rtl8188eus/core/rtw_rm.o
  CC [M]  /root/rtl8188eus/core/rtw_rm_fsm.o
  CC [M]  /root/rtl8188eus/core/efuse/rtw_efuse.o
  CC [M]  /root/rtl8188eus/os_dep/osdep_service.o
/root/rtl8188eus/os_dep/osdep_service.c: In function ‘rtw_change_ifname’:
/root/rtl8188eus/os_dep/osdep_service.c:2454:28: warning: passing argument 1 of ‘_rtw_memcpy’ discards ‘const’ qualifier from pointer target type [-Wdiscarded-qualifiers]
 2454 |         _rtw_memcpy(pnetdev->dev_addr, adapter_mac_addr(padapter), ETH_ALEN);
      |                     ~~~~~~~^~~~~~~~~~
/root/rtl8188eus/os_dep/osdep_service.c:899:24: note: expected ‘void *’ but argument is of type ‘const unsigned char *’
  899 | void _rtw_memcpy(void *dst, const void *src, u32 sz)
      |                  ~~~~~~^~~
/root/rtl8188eus/os_dep/osdep_service.c: In function ‘rtw_random32’:
/root/rtl8188eus/os_dep/osdep_service.c:2578:16: error: implicit declaration of function ‘prandom_u32’; did you mean ‘get_random_u32’? [-Werror=implicit-function-declaration]
 2578 |         return prandom_u32();
      |                ^~~~~~~~~~~
      |                get_random_u32
At top level:
cc1: note: unrecognized command-line option ‘-Wno-sometimes-uninitialized’ may have been intended to silence earlier diagnostics
cc1: some warnings being treated as errors
make[2]: *** [/usr/src/linux-headers-6.3.0-kali1-common/scripts/Makefile.build:257: /root/rtl8188eus/os_dep/osdep_service.o] Error 1
make[1]: *** [/usr/src/linux-headers-6.3.0-kali1-common/Makefile:2050: /root/rtl8188eus] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-6.3.0-kali1-amd64'
make: *** [Makefile:2065: modules] Error 2

Vicbrill44, 2023-08-16:
Getting similar issue

gglluukk, 2023-08-30:
try this for newer kernel (still in pull requests, not merged yet):
https://github.com/gglluukk/rtl8188eus

Dark-Cod3r, 2023-09-04:
> try this for newer kernel (still in pull requests, not merged yet): https://github.com/gglluukk/rtl8188eus

this worked on parrot os (and other 6.0+ kernel distros).. 
and as for kali. 
kali linux already has this patch.. not sure why OP want to install these drivers..!?