aircrack-ng/rtl8188eus #260: sudo make command error

Arafat05nov, 2023-08-29:
┌──(kali㉿kali)-[~/Downloads/rtl8188eus_new]
└─$ sudo make       
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/6.4.0-kali3-amd64/build M=/home/kali/Downloads/rtl8188eus_new  modules
make[1]: Entering directory '/usr/src/linux-headers-6.4.0-kali3-amd64'
  CC [M]  /home/kali/Downloads/rtl8188eus_new/os_dep/osdep_service.o
/home/kali/Downloads/rtl8188eus_new/os_dep/osdep_service.c: In function ‘rtw_random32’:
/home/kali/Downloads/rtl8188eus_new/os_dep/osdep_service.c:2582:16: error: implicit declaration of function ‘prandom_u32’; did you mean ‘get_random_u32’? [-Werror=implicit-function-declaration]
 2582 |         return prandom_u32();
      |                ^~~~~~~~~~~
      |                get_random_u32
cc1: some warnings being treated as errors
make[2]: *** [/usr/src/linux-headers-6.4.0-kali3-common/scripts/Makefile.build:257: /home/kali/Downloads/rtl8188eus_new/os_dep/osdep_service.o] Error 1
make[1]: *** [/usr/src/linux-headers-6.4.0-kali3-common/Makefile:2057: /home/kali/Downloads/rtl8188eus_new] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-6.4.0-kali3-amd64'
make: *** [Makefile:2059: modules] Error 2

gglluukk, 2023-08-29:
try this for newer kernel (still in pull requests, not merged yet):
https://github.com/gglluukk/rtl8188eus

Arafat05nov, 2023-08-29:
> try this for newer kernel (still in pull requests, not merged yet):
> https://github.com/gglluukk/rtl8188eus

Tried this also but getting same error

gglluukk, 2023-08-29:
it can't produce same error for
`return prandom_u32();`
since it got patched.

you probably try to compile old source again

Arafat05nov, 2023-08-29:
> it can't produce same error for
> `return prandom_u32();`
> since it got patched.
> 
> you probably try to compile old source again

How should I delete the old ones completely

Arafat05nov, 2023-08-29:
Please help brother I'm frustrated by the error

Arafat05nov, 2023-08-29:
> yes, remove files from: 
> https://github.com/aircrack-ng/rtl8188eus/
> 
> git clone / download:
> https://github.com/gglluukk/rtl8188eus
> 
> and compile happily =)

Ok will try and get back to you

Arafat05nov, 2023-08-30:
> yes, remove files from: 
> https://github.com/aircrack-ng/rtl8188eus/
> 
> git clone / download:
> https://github.com/gglluukk/rtl8188eus
> 
> and compile happily =)

Thanks alot brother, at last it all went well. Thank you very much

Hsin8203, 2024-01-17:
> yes, remove files from: https://github.com/aircrack-ng/rtl8188eus/
> 
> git clone / download: https://github.com/gglluukk/rtl8188eus
> 
> and compile happily =)

Hi gglluukk,

Sorry, I am new to Linux. I am confused about the step 1. How will it work by a given github address in removing files?

gglluukk, 2024-01-18:
Hi, it means to:
1. remove old sources you've downloaded from https://github.com/aircrack-ng/rtl8188eus/ 
2. download sources from https://github.com/gglluukk/rtl8188eus
3. compile it

P.S.: driver `rtl8xxxu` coming with newer kernels supports monitor mode so use it before trying this one

Hsin8203, 2024-01-19:
> Hi, it means to:
> 
> 1. remove old sources you've downloaded from https://github.com/aircrack-ng/rtl8188eus/ 
> 
> 2. download sources from https://github.com/gglluukk/rtl8188eus
> 
> 3. compile it
> 
> 
> 
> P.S.: driver `rtl8xxxu` coming with newer kernels supports monitor mode so use it before trying this one

Thank you so so much for the update😀