aircrack-ng/rtl8188eus #261: nl80211: Could not configure driver mode

trueToastedCode, 2023-09-11:
hostapd.conf
```
ctrl_interface=/var/run/hostapd
ctrl_interface_group=0
driver=nl80211
ssid=WPA3
country_code=EN
channel=6
beacon_int=100
dtim_period=1
fragm_threshold=2346
rts_threshold=2347
hw_mode=g
ieee80211n=1
max_num_sta=16
wmm_enabled=1
ht_capab=[HT40-][SHORT-GI-20][SHORT-GI-40][RX-STBC1][MAX-AMSDU-7935][DSSS_CCK-40]
ignore_broadcast_ssid=0

auth_algs=1
wpa=2
wpa_key_mgmt=SAE
wpa_pairwise=CCMP
rsn_pairwise=CCMP
ieee80211w=2
sae_groups=19 20 21
sae_require_mfp=1
wpa_passphrase=1234567890
```

(`airman-ng check kill` had been executed)

This appears to be an issue with newer linux kernel, since it had been working for me with older.

Used this fix for 6.3: https://github.com/gglluukk/rtl8188eus

gglluukk, 2023-09-11:
Hi, 

This simple config works for me:
```
interface=wlan0
hw_mode=g
channel=7
wmm_enabled=0
macaddr_acl=0
auth_algs=1
ignore_broadcast_ssid=0
wpa=2
wpa_key_mgmt=WPA-PSK
wpa_pairwise=TKIP
rsn_pairwise=CCMP
ssid=TEST
wpa_passphrase=12345678
```

Try to run with debug output:
```
hostapd -dd /etc/hostapd/hostapd.conf
```

trueToastedCode, 2023-09-12:
> Hi,
>
> This simple config works for me:
>
> ```
> interface=wlan0
> hw_mode=g
> channel=7
> wmm_enabled=0
> macaddr_acl=0
> auth_algs=1
> ignore_broadcast_ssid=0
> wpa=2
> wpa_key_mgmt=WPA-PSK
> wpa_pairwise=TKIP [...]

Still not working for me... perhaps #263 may has something to do with it.

trueToastedCode, 2023-09-12:
> Hi,
>
> This simple config works for me:
>
> ```
> interface=wlan0
> hw_mode=g
> channel=7
> wmm_enabled=0
> macaddr_acl=0
> auth_algs=1
> ignore_broadcast_ssid=0
> wpa=2
> wpa_key_mgmt=WPA-PSK
> wpa_pairwise=TKIP [...]

```
┌──(parallels㉿kali-gnu-linux-2023)-[~]
└─$ sudo hostapd -dd /etc/hostapd/hostapd.conf 
random: getrandom() support available
Configuration file: /etc/hostapd/hostapd.conf
nl80211: Supported cipher 00-0f-ac:1
nl80211: Supported cipher 00-0f-ac:5
nl80211: Supported cipher 00-0f-ac:2
nl80211: Supported cipher 00-0f-ac:4
nl80211: Supported cipher 00-0f-ac:10
nl80211: Supported cipher 00-0f-ac:8
nl80211: Supported cipher 00-0f-ac:9
nl80211: Using driver-based off-channel TX
nl80211: Driver-advertised extended capabilities (default) - hexdump(len=8): 00 00 00 00 00 00 00 40
nl80211: Driver-advertised extended capabilities mask (default) - hexdump(len=8): 00 00 00 00 00 00 00 40
nl80211: key_mgmt=0x1ff0f enc=0xef auth=0x7 flags=0x40005104303d0a0 rrm_flags=0x10 probe_resp_offloads=0x0 max_stations=0 max_remain_on_chan=5000 max_scan_ssids=4
nl80211: interface wlan0 in phy phy7
nl80211: Set mode ifindex 10 iftype 3 (AP)
nl80211: Failed to set interface 10 to mode 3: -95 (Operation not supported)
nl80211: Try mode change after setting interface down
nl80211: Set mode ifindex 10 iftype 3 (AP)
nl80211: Failed to set interface 10 to mode 3: -95 (Operation not supported)
nl80211: Interface mode change to 3 from 0 failed
nl80211: Could not configure driver mode
nl80211: deinit ifname=wlan0 disabled_11b_rates=0
nl80211: Remove monitor interface: refcount=0
netlink: Operstate: ifindex=10 linkmode=0 (kernel-control), operstate=6 (IF_OPER_UP)
nl80211 driver initialization failed.
hostapd_interface_deinit_free(0xaaaac351c970)
hostapd_interface_deinit_free: num_bss=1 conf->num_bss=1
hostapd_interface_deinit(0xaaaac351c970)
wlan0: interface state UNINITIALIZED->DISABLED
hostapd_bss_deinit: deinit bss wlan0
wlan0: AP-DISABLED 
hostapd_cleanup(hapd=0xaaaac351e140 (wlan0))
wlan0: CTRL-EVENT-TERMINATING 
hostapd_free_hapd_data: Interface wlan0 wasn't started
hostapd_interface_deinit_free: driver=(nil) drv_priv=(nil) -> hapd_deinit
hostapd_interface_free(0xaaaac351c970)
hostapd_interface_free: free hapd 0xaaaac351e140
hostapd_cleanup_iface(0xaaaac351c970)
hostapd_cleanup_iface_partial(0xaaaac351c970)
hostapd_cleanup_iface: free iface=0xaaaac351c970
```

gglluukk, 2023-09-12:
try to:
`rmmod r8188eu rtl8xxxu 8188eu`
`modprobe 8188eu`
`ifconfig wlan0 down`
`hostapd -dd /etc/hostapd/hostapd.conf`

trueToastedCode, 2023-09-12:
> modprobe 8188eu

This finally worked! 🎉 Tho idk why this is necessary on my system

trueToastedCode, 2023-09-12:
> try to: `rmmod r8188eu rtl8xxxu 8188eu` `modprobe 8188eu` `ifconfig wlan0 down` `hostapd -dd /etc/hostapd/hostapd.conf`

This finally worked! 🎉 Tho idk why this is necessary on my system...

closed 2023-09-12 by trueToastedCode

trueToastedCode, 2023-09-13:
> > try to: `rmmod r8188eu rtl8xxxu 8188eu` `modprobe 8188eu` `ifconfig wlan0 down` `hostapd -dd /etc/hostapd/hostapd.conf`
> 
> This finally worked! 🎉 Tho idk why this is necessary on my system...

`/etc/modprobe.d/realtek.conf` should blacklist `rtl8xxxu` as well:
```
blacklist r8188eu
blacklist rtl8xxxu
```

now it will load `8188eu` instead.