aircrack-ng/rtl8188eus #262: make error on 6.4.0-kali3-arm64

skyper-cz, 2023-09-11:
Hi, I wanted to install these drivers on my virtual machine which connects to my usb dongle.
However Im getting following error.

[image]
Any idea how to solve it. It might me the ARM64 issue.

gglluukk, 2023-09-11:
Hi,

There are no issues for using driver under ARM64 (I use https://github.com/utmapp/UTM/ for Apple M1).

Try:
```
apt --yes install build-essential linux-headers-`uname -r` bc
```

skyper-cz, 2023-09-11:
It started working but now I got following error:

[image]

gglluukk, 2023-09-11:
same issue was described, read it here: https://github.com/aircrack-ng/rtl8188eus/issues/256

skyper-cz, 2023-09-11:
Thanks, using your variant it is working now.

closed 2023-09-11 by skyper-cz