aircrack-ng/rtl8188eus #263: make → /bin/sh: 1: ./tools/bpf/resolve_btfids/resolve_btfids: not found

trueToastedCode, 2023-09-12:
```
┌──(parallels㉿kali-gnu-linux-2023)-[~/rtl8188eus]
└─$ make
make ARCH=arm64 CROSS_COMPILE= -C /lib/modules/6.3.0-kali1-arm64/build M=/home/parallels/rtl8188eus  modules
make[1]: Entering directory '/usr/src/linux-headers-6.3.0-kali1-arm64'
  LD [M]  /home/parallels/rtl8188eus/8188eu.ko
  BTF [M] /home/parallels/rtl8188eus/8188eu.ko
/bin/sh: 1: ./tools/bpf/resolve_btfids/resolve_btfids: not found
make[2]: *** [/usr/src/linux-headers-6.3.0-kali1-common/scripts/Makefile.modfinal:63: /home/parallels/rtl8188eus/8188eu.ko] Error 127
make[2]: *** Deleting file '/home/parallels/rtl8188eus/8188eu.ko'
make[1]: *** [/usr/src/linux-headers-6.3.0-kali1-common/Makefile:1967: modules] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-6.3.0-kali1-arm64'
make: *** [Makefile:2065: modules] Error 2
```
(I've installed various packages using `sudo apt-get install linux-headers-$(uname -r) build-essential bc` and also used this solution [Skipping BTF generation](https://askubuntu.com/questions/1348250/skipping-btf-generation-xxx-due-to-unavailability-of-vmlinux-on-ubuntu-21-04))

gglluukk, 2023-09-12:
Also as I can see module is compiled and linked but removed here: `make[2]: *** Deleting file '/home/parallels/rtl8188eus/8188eu.ko'` because of that error.

Corresponding code is at /usr/src/YOUR_LINUX/scripts/Makefile.modfinal at string 63:
```
ifdef CONFIG_DEBUG_INFO_BTF_MODULES
        +$(if $(newer-prereqs),$(call cmd,btf_ko))
endif
```
So try to remove these 3 lines.

trueToastedCode, 2023-09-12:
> Also as I can see module is compiled and linked but removed here: `make[2]: *** Deleting file '/home/parallels/rtl8188eus/8188eu.ko'` because of that error.
>
> Corresponding code is at [...]

This works, thx 👍

closed 2023-09-12 by trueToastedCode