aircrack-ng/rtl8188eus #264: make problem

mayburs, 2023-09-14:
**Hi there.**

I installed kali on virtual box.
_Distributor ID: Kali
Description:    Kali GNU/Linux Rolling
Release:        2023.3
Codename:       kali-rolling_

I got tplink TL-WN722N network adapter.
after clonning in the make command i get this problem.

/home/kali/rtl8188eus/os_dep/osdep_service.c:2454:28: warning: passing argument 1 of ‘_rtw_memcpy’ discards ‘const’ qualifier from pointer target type [-Wdiscarded-qualifiers]
 2454 |         _rtw_memcpy(pnetdev->dev_addr, adapter_mac_addr(padapter), ETH_ALEN);
      |                     ~~~~~~~^~~~~~~~~~
/home/kali/rtl8188eus/os_dep/osdep_service.c:899:24: note: expected ‘void *’ but argument is of type ‘const unsigned char *’
  899 | void _rtw_memcpy(void *dst, const void *src, u32 sz)
      |                  ~~~~~~^~~
/home/kali/rtl8188eus/os_dep/osdep_service.c: In function ‘rtw_random32’:
/home/kali/rtl8188eus/os_dep/osdep_service.c:2578:16: error: implicit declaration of function ‘prandom_u32’; did you mean ‘get_random_u32’? [-Werror=implicit-function-declaration]
 2578 |         return prandom_u32();
      |                ^~~~~~~~~~~
      |                get_random_u32
At top level:
cc1: note: unrecognized command-line option ‘-Wno-sometimes-uninitialized’ may have been intended to silence earlier diagnostics
cc1: some warnings being treated as errors
make[2]: *** [/usr/src/linux-headers-6.4.0-kali3-common/scripts/Makefile.build:257: /home/kali/rtl8188eus/os_dep/osdep_service.o] Error 1
make[1]: *** [/usr/src/linux-headers-6.4.0-kali3-common/Makefile:2057: /home/kali/rtl8188eus] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-6.4.0-kali3-amd64'
make: *** [Makefile:2065: modules] Error 2

i used to clone to git clone  https://github.com/gglluukk/rtl8188eus/ but it made me some problems in monitor mode...

Would u please give an estimated time that it will be fixed hopefully.
and give me a soultion inorder to i can fix it?

gglluukk, 2023-09-14:
try:
```
rmmod r8188eu rtl8xxxu 8188eu
modprobe 8188eu
```
as well as:
```
echo 'blacklist r8188eu' | sudo tee -a '/etc/modprobe.d/realtek.conf'
echo 'blacklist rtl8xxxu' | sudo tee -a '/etc/modprobe.d/realtek.conf'
```
to keep changes through reboots

mayburs, 2023-09-14:
Thanks @gglluukk 
Actually I switeched to this  https://github.com/gglluukk/rtl8188eus/ again and did what you mentioned and for now it's fine

but another question:
I know it's not much important but inorder to keep things clean :
the interface name doesn't change from wlan0 to wlan0mon when i change the status from managed to monitor mode
although the mode has been changed and works.

and idea about it?

gglluukk, 2023-09-14:
there is no need to change interface's name, but you can do:
`ip link set wlan0 name wlan0mon`

gglluukk, 2023-09-20:
> make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/6.2.0-33-generic/build M=/home/vav4eg/rtl8188eus modules
> make[1]: вход в каталог «/usr/src/linux-headers-6.2.0-33-generic»
> warning: the compiler differs from the one used to build the kernel
> The kernel was built by: x86_64-linux-gnu-gcc-11 (Ubuntu 11.4.0-1ubuntu1~22.04) 11.4.0 You are using: gcc-11 (Ubuntu 11.4.0-1ubuntu1~22.04) 11.4.0
> CC [M] /home/vav4eg/rtl8188eus/core/rtw_cmd.o
> CC [M] /home/vav4eg/rtl8188eus/core/rtw_security.o
> CC [M] /home/vav4eg/rtl8188eus/core/rtw_debug.o
> CC [M] /home/vav4eg/rtl8188eus/core/rtw_io.o
> CC [M] /home/vav4eg/rtl8188eus/core/rtw_ioctl_query.o
> CC [M] /home/vav4eg/rtl8188eus/core/rtw_ioctl_set.o
> CC [M] /home/vav4eg/rtl8188eus/core/rtw_ieee80211.o
> CC [M] /home/vav4eg/rtl8188eus/core/rtw_mlme.o
> CC [M] /home/vav4eg/rtl8188eus/core/rtw_mlme_ext.o
> CC [M] /home/vav4eg/rtl8188eus/core/rtw_mi.o
> CC [M] /home/vav4eg/rtl8188eus/core/rtw_wlan_util.o
> CC [M] /home/vav4eg/rtl8188eus/core/rtw_vht.o
> CC [M] /home/vav4eg/rtl8188eus/core/rtw_pwrctrl.o
> CC [M] /home/vav4eg/rtl8188eus/core/rtw_rf.o
> CC [M] /home/vav4eg/rtl8188eus/core/rtw_recv.o
> CC [M] /home/vav4eg/rtl8188eus/core/rtw_sta_mgt.o
> CC [M] /home/vav4eg/rtl8188eus/core/rtw_ap.o
> CC [M] /home/vav4eg/rtl8188eus/core/mesh/rtw_mesh.o
> CC [M] /home/vav4eg/rtl8188eus/core/mesh/rtw_mesh_pathtbl.o
> CC [M] /home/vav4eg/rtl8188eus/core/mesh/rtw_mesh_hwmp.o
> CC [M] /home/vav4eg/rtl8188eus/core/rtw_xmit.o
> CC [M] /home/vav4eg/rtl8188eus/core/rtw_p2p.o
> CC [M] /home/vav4eg/rtl8188eus/core/rtw_rson.o
> CC [M] /home/vav4eg/rtl8188eus/core/rtw_tdls.o
> CC [M] /home/vav4eg/rtl8188eus/core/rtw_br_ext.o
> CC [M] /home/vav4eg/rtl8188eus/core/rtw_iol.o
> CC [M] /home/vav4eg/rtl8188eus/core/rtw_sreset.o
> CC [M] /home/vav4eg/rtl8188eus/core/rtw_btcoex_wifionly.o
> CC [M] /home/vav4eg/rtl8188eus/core/rtw_btcoex.o
> CC [M] /home/vav4eg/rtl8188eus/core/rtw_beamforming.o
> CC [M] /home/vav4eg/rtl8188eus/core/rtw_odm.o
> CC [M] /home/vav4eg/rtl8188eus/core/rtw_rm.o
> CC [M] /home/vav4eg/rtl8188eus/core/rtw_rm_fsm.o
> CC [M] /home/vav4eg/rtl8188eus/core/efuse/rtw_efuse.o
> CC [M] /home/vav4eg/rtl8188eus/os_dep/osdep_service.o
> /home/vav4eg/rtl8188eus/os_dep/osdep_service.c: In function ‘rtw_change_ifname’:
> /home/vav4eg/rtl8188eus/os_dep/osdep_service.c:2454:28: warning: passing argument 1 of ‘_rtw_memcpy’ discards ‘const’ qualifier from pointer target type [-Wdiscarded-qualifiers]
> 2454 | _rtw_memcpy(pnetdev->dev_addr, adapter_mac_addr(padapter), ETH_ALEN);
> | ~~~~~~~^~~~~~~~~~
> /home/vav4eg/rtl8188eus/os_dep/osdep_service.c:899:24: note: expected ‘void *’ but argument is of type ‘const unsigned char *’
> 899 | void _rtw_memcpy(void *dst, const void *src, u32 sz)
> | ~~~~~~^~~
> /home/vav4eg/rtl8188eus/os_dep/osdep_service.c: In function ‘rtw_random32’:
> /home/vav4eg/rtl8188eus/os_dep/osdep_service.c:2578:16: error: implicit declaration of function ‘prandom_u32’; did you mean ‘get_random_u32’? [-Werror=implicit-function-declaration]
> 2578 | return prandom_u32();
> | ^~~~~~~~~~~
> | get_random_u32
> /home/vav4eg/rtl8188eus/os_dep/osdep_service.c: At top level:
> cc1: note: unrecognized command-line option ‘-Wno-sometimes-uninitialized’ may have been intended to silence earlier diagnostics
> cc1: some warnings being treated as errors
> make[2]: *** [scripts/Makefile.build:260: /home/vav4eg/rtl8188eus/os_dep/osdep_service.o] Ошибка 1
> make[1]: *** [Makefile:2026: /home/vav4eg/rtl8188eus] Ошибка 2
> make[1]: выход из каталога «/usr/src/linux-headers-6.2.0-33-generic»
> make: *** [Makefile:2065: modules] Ошибка 2

use this repo: https://github.com/gglluukk/rtl8188eus/

arnav7633, 2023-09-21:
Hey @vav4eg what version of the 722N do you have? V1,2,3? Also can you use hcxdumptool -L and hcxdumptool -i <interface> .

I am looking to buy the same device and was wondering if everything is working.

vav4eg, 2023-09-21:
> use this repo: https://github.com/gglluukk/rtl8188eus/

On the specified driver, the signal is very bad, my home network, shows no more than one bar. The LED does not work.

gglluukk, 2023-09-21:
> > use this repo: https://github.com/gglluukk/rtl8188eus/
> 
> On the specified driver, the signal is very bad, my home network, shows no more than one bar. The LED does not work.

`rtl8188eus` driver is to be used mostly for monitor mode. 
for leds / client / ad-hoc you might switch back to `r8188eu` or `rtl8xxxu` and check signal strength with those drivers

arnav7633, 2023-09-22:
@vav4eg, sorry for that. Seems you have a older hcxdumptool. Was wondering if you could find out if the driver support packet injection, and active monitor mode. I am not trusting online articles since they were wrong about my current card.

If you update to latest hcxtools you can get that command running

vinimrs, 2023-11-05:
> use this repo: https://github.com/gglluukk/rtl8188eus/

Thank you @gglluukk, I have a problem with the drivers since I updated Ubuntu from 22.04 to 23.04 and your repository saved me, works correctly!

gglluukk, 2023-11-05:
thanks!
but there is no more need to use my repo since it's merged here