aircrack-ng/rtl8188eus #270: make error on 6.5.0-kali1-arm64 in UTM

skyper-cz, 2023-10-27:
Hello my friend
I wnated to try wifite after a while but my machine changed the driver for the original one. So I went to try recompile it again however I got this arror along the way.

[image]

Is there solution any at this moment?

SmillerMP, 2023-10-27:
You are using raspberry pi?

skyper-cz, 2023-11-30:
No, im using UTM for virtualization, but it was working before but not now.

gglluukk, 2023-12-01:
driver `rtl8xxxu` coming with newer kernels supports monitor mode so use it before trying this one

otherwise to compile this driver you need to fully update/upgrade your debian-based distro:
`apt update && apt -y upgrade && apt clean && apt autoremove && update-grub && reboot`

after that:
```
apt --yes install build-essential linux-headers-`uname -r`
```
and compile this driver