aircrack-ng/rtl8188eus #271: Make error

Diablooo999, 2023-11-01:
[image]
What should i do?

SmillerMP, 2023-11-06:
update to most recent version of possible kernel, new versions of linux kernel has support to rtl8xxxu for default

[image]

gglluukk, 2023-11-06:
@SmillerMP rtl8xxxu for newer kernels as well as rtl8188eu for older ones don’t support **monitor** mode, so that’s why this driver exists. even more: you’ll have to blacklist those two for not getting loaded and not interfere with this one

@Diablooo999 update/renew you kali and follow instructions in readme

SmillerMP, 2023-11-06:
[image]

I don't have any problems injecting packets or scanning with airodump with rtl8xxxu.

@kimocoder  commented that this repository would no longer be maintained due to the change to rtl8xxxu

gglluukk, 2023-11-06:
omg, you're right with rtl8xxxu. surprisingly for me it works now in monitor mode because before i tried it and it didn't work, so it might has been updated after i had tried it.
think that this repo should get archived with corresponding comments and instructions

SmillerMP, 2023-11-07:
I still use this driver for raspberry pi, but for other versions of Linux it works correctly rtl8xxxu

closed 2026-09-09 by Laaach