aircrack-ng/rtl8188eus #277: Mac address changes automatically

ItzKala, 2024-01-19:
I made a couple of changes to my virtual machine (debian 12) to run monitor mode on my tp-link tl-w722n. After I correctly installed your module (which worked perfectly btw) I noticed that the Mac address of my adapter changes on its own every like 10 minutes or something. I don't know what it is in your module  that did this but all I want is to control it (stop it for now, maybe activate it later) so guide me please. 
These are the exact commands I executed on my terminal and everything ran smoothly 

 `sudo apt update `

 `sudo apt install bc`

 `sudo rmmod r8188eu.ko`

 `git clone https://github.com/aircrack-ng/rtl8188eus`

`cd rtl8188eus `

`sudo -i`

`echo "blacklist r8188eu" > "/etc/modprobe.d/realtek.conf"`

`exit`

`make`

`sudo make install`

`sudo modprobe 8188eu`

Thank you for your time in advance.

gglluukk, 2024-01-23:
Hi,

try to use driver `rtl8xxxu` coming with newer kernels (it supports monitor mode) and check if device keeps changing its MAC

ItzKala, 2024-04-27:
@gglluukk  Hello there,

The driver `rtl8xxxu` is not working at all with me when the other drivers are blocked. is there anyway I can know what are the drivers I have installed on my machine? also is there a way I can remove a driver completely or at least undo everything I did in the comment above?

Thanks in advance.

dubhater, 2024-04-27:
``modinfo 8188eu`` will tell you where this driver is installed. Then you can look in that folder for other drivers.

``sudo make uninstall`` should remove it.

If you tell me what rtl8xxxu is doing, maybe I can fix it. Look at dmesg.

gglluukk, 2024-04-27:
use following commands to research your case:
- `lsusb` -- shows usb-attached devices
- `lsmod` -- lists loaded kernel modules
- `rmmod` -- removes kernel module
- `modprobe` -- loads kernel module
- `dmesg` -- shows kernel's output

try to:
- make sure `lsusb` shows your device attached
- remove `8188eu` and load `rtl8xxxu` and vise versa but don't let both modules being loaded in same time

ItzKala, 2024-04-28:
@dubhater  this is the result of `dmesg` along with `ifconfig`
`[  941.350974] usbcore: registered new interface driver rtl8xxxu
root@BlackRaven:~# ifconfig 
enp0s17: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 10.0.2.15  netmask 255.255.255.0  broadcast 10.0.2.255
        inet6 fe80::a00:27ff:fe27:37f2  prefixlen 64  scopeid 0x20<link>
        ether 08:00:27:27:37:f2  txqueuelen 1000  (Ethernet)
        RX packets 3027  bytes 2047249 (1.9 MiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 2278  bytes 433560 (423.3 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        inet6 ::1  prefixlen 128  scopeid 0x10<host>
        loop  txqueuelen 1000  (Local Loopback)
        RX packets 32  bytes 3635 (3.5 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 32  bytes 3635 (3.5 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
`
as you can see, the adapter's interface does not appear in `ifconfig` or `iwconfig`. even when I type `ifconfig -a`, it still wont work

also when I activate any of the other modules this appears in dmesg
`[  785.817097] r8188eu: module is from the staging directory, the quality is unknown, you have been warned.
[  785.905244] usbcore: registered new interface driver r8188eu
[  785.907632] r8188eu 1-1:1.0 wlx5ca6e6fb0b42: renamed from wlan0
[  785.930844] r8188eu 1-1:1.0: firmware: direct-loading firmware rtlwifi/rtl8188eufw.bin
[  785.930852] r8188eu 1-1:1.0: Firmware Version 11, SubVersion 1, Signature 0x88e1
[  792.052360] r8188eu 1-1:1.0: firmware: direct-loading firmware rtlwifi/rtl8188eufw.bin
[  920.772839] usbcore: deregistering interface driver r8188eu
[  921.135335] usb 1-1: reset high-speed USB device number 2 using xhci_hcd
`
first I wannna know why the interface name changes from wlan0 to this weird sh*t
I also wanna know what the firmware mesgs mean.
one last thing, when I type `rmmod 8188eu` and `modprobe r8188eu` then restart my machine, 8188eu loads again and r8188eu stops. when I tried to copy it from the  staging dir to the wireless dir, the mod wouldn't even load. how do I fix that?
 
I'm so sorry for the way too much questions but I'm really trying to learn.
and thank you in advance.

ItzKala, 2024-04-28:
@gglluukk 
Only one module runs at a time. yet rtl8xxxu does not work while the other two work fine

gglluukk, 2024-04-28:
that's strange if it's not got blacklisted via:
```
echo 'blacklist rtl8xxxu' | sudo tee -a '/etc/modprobe.d/realtek.conf'
```

if so just backup it and remove from kernel modules:
```
mv /usr/lib/modules/`uname` -r`/kernel/drivers/net/wireless/realtek/rtl8xxxu/rtl8xxxu.ko /root/
```

to make sure check if any other pieces of it exists:
```
find /lib/modules/ -name rtl8xxxu.ko
```

ItzKala, 2024-04-28:
@gglluukk 

I did exactly what you said and I got this

```
~# modprobe rtl8xxxu
modprobe: ERROR: could not insert 'rtl8xxxu': Unknown symbol in module, or unknown parameter (see dmesg)

```
result of dmesg:

[    4.250687] ==> rtl8188e_iol_efuse_patch
[    4.743518] [drm:vmw_msg_ioctl [vmwgfx]] *ERROR* Failed to open channel.
[    4.743547] [drm:vmw_msg_ioctl [vmwgfx]] *ERROR* Failed to open channel.
[    4.747399] IPv6: ADDRCONF(NETDEV_CHANGE): wlx5ca6e6fb0b42: link becomes ready
[    5.091564] [drm:vmw_msg_ioctl [vmwgfx]] *ERROR* Failed to open channel.
[    5.091594] [drm:vmw_msg_ioctl [vmwgfx]] *ERROR* Failed to open channel.
[    5.952946] rfkill: input handler disabled
[    6.127459] [drm:vmw_msg_ioctl [vmwgfx]] *ERROR* Failed to open channel.
[    6.127486] [drm:vmw_msg_ioctl [vmwgfx]] *ERROR* Failed to open channel.
[    7.054568] [drm:vmw_msg_ioctl [vmwgfx]] *ERROR* Failed to open channel.
[    7.054599] [drm:vmw_msg_ioctl [vmwgfx]] *ERROR* Failed to open channel.
[   86.832541] rfkill: input handler enabled
[   87.200923] [drm:vmw_msg_ioctl [vmwgfx]] *ERROR* Failed to open channel.
[   87.200949] [drm:vmw_msg_ioctl [vmwgfx]] *ERROR* Failed to open channel.
[   88.157956] rfkill: input handler disabled
[  118.978665] [drm:vmw_msg_ioctl [vmwgfx]] *ERROR* Failed to open channel.
[  118.978708] [drm:vmw_msg_ioctl [vmwgfx]] *ERROR* Failed to open channel.
[  118.983422] [drm:vmw_msg_ioctl [vmwgfx]] *ERROR* Failed to open channel.
[  118.983447] [drm:vmw_msg_ioctl [vmwgfx]] *ERROR* Failed to open channel.
[  119.095628] [drm:vmw_msg_ioctl [vmwgfx]] *ERROR* Failed to open channel.
[  119.095668] [drm:vmw_msg_ioctl [vmwgfx]] *ERROR* Failed to open channel.
[  119.443272] [drm:vmw_msg_ioctl [vmwgfx]] *ERROR* Failed to open channel.
[  119.443301] [drm:vmw_msg_ioctl [vmwgfx]] *ERROR* Failed to open channel.

dubhater, 2024-04-28:
rtl8xxxu only handles RTL8188EU since version 6.3. Maybe your kernel is older than 6.3. I think r8188eu was removed in 6.3, so your kernel must be older.

If you have two wifi devices connected, there is no guarantee about which one will be wlan0 and which will be wlan1. So they (udev?) rename the interfaces to something predictable. In this case, the new name is based on the MAC address. Other distros use the USB device number or something like that.

The wifi chip has a microcontroller, which requires some code in order to provide the full functionality. That's the firmware. The driver loads the firmware from the file system and uploads it to the chip.

ItzKala, 2024-04-28:
@dubhater OK so bear with me for a second. 

What should I do now to fix the main problem? (Activate monitor mode on my tp-link tl-w722n)

Do I remove all three mods and find another one? Upgrade kernel (if possible)? Or keep tampering with the three modules using the little knowledge I have hoping the whole thing gets fixed on its own?

dubhater, 2024-04-28:
You can try the latest version of rtl8xxxu with your current kernel: https://github.com/lwfinger/rtl8xxxu/

Other than that, I'm not sure.

dubhater, 2024-04-28:
I don't think this driver here is responsible for the changing MAC address. It's more likely that NetworkManager does that. For monitor mode, you want to turn off NetworkManager or at least make it ignore the TP-Link.

ItzKala, 2024-04-28:
@dubhater I see your point. 
Just to give you a clearer picture of what's going on. The driver starts on Auto mode rather than managed. Even then it changes the Mac address. And when I turn it to manged it still changes the Mac address automatically. 
I also make sure to use `airmon-ng check kill` before switching to monitor mode. Doesn't this command turn off network manager? 
Until you see this I'll be installing the version of rtl8xxxu you suggested and I'll comment the results right after

gglluukk, 2024-04-28:
@ItzKala 

- upgrade your distro -- that's for sure, and `rtl8xxxu` might do the job
- note that most of actions are to be done via console, not GUI tools, so turn them off, those can interfere
- research if your dongle can connect to AP in normal mode
- if it can do so, don't try to reswitch to monitor mode, better reboot without entering client mode and switch to monitor mode
- aircrack is bit outdated while still works, consider use hcxtools

ItzKala, 2024-04-28:
@dubhater  OK
so I followed the link https://github.com/lwfinger/rtl8xxxu/
got the https code
typed `git clone https://github.com/lwfinger/rtl8xxxu.git` 
`cd rtl8xxxu`
`make && make install`
and when I typed `modprobe rtl8xxxu_git` I got this:

```
~# modprobe rtl8xxxu_git
modprobe: ERROR: could not insert 'rtl8xxxu_git': Device or resource busy
```
result for dmesg: 

`[ 6014.916614] Error: Driver 'rtl8xxxu' is already registered, aborting...`

am I doing something wrong?

ItzKala, 2024-04-28:
@dubhater OK 

I installed the latest version of rtl8xxxu and loaded the driver

as a matter of fact, the Mac address changing issue indeed is not from the 8188eu driver. rtl8xxxu had the same problem but this one loaded on managed mode with this issue.

and unlike 8188eu, the adapter cant sniff packets in monitor mode with the rtl8xxxu driver.

should I just give up on rtl8xxxu and return back to 8188eu? (if turning off networkmanager  stops the Mac address from changing)

gglluukk, 2024-04-28:
> note that most of actions are to be done via console, not GUI tools, so turn them off, those can interfere

dubhater, 2024-04-28:
Yes, if rtl8xxxu is not working, the driver from this repository is your best option.

ItzKala, 2024-04-29:
> @ItzKala
>
> * upgrade your distro -- that's for sure, and `rtl8xxxu` might do the job
> * note that most of actions are to be done via console, not GUI tools, so turn them off, those can interfere
> * [...]

Yes, as a matter of fact I started working on upgrading to debian 12.5 hoping that would fix my issue.

I also am aware that the GUI isn't meant to preform all tasks. I'll turn anything unnecessary off in order to avoid unwanted interference 

I had made my research already and it seems  I can't connect to AP in normal mode. Does that make a huge difference?

I'll start learning how to utilize hcxtools right away.

ItzKala, 2024-04-29:
> Yes, if rtl8xxxu is not working, the driver from this repository is your best option.
> 
> I have an idea for fixing rtl8xxxu, but it's a big change and I don't want to start now. I'm in the middle of something.

Whenever you're ready. The issue stays open until I find some other way to fix it.

ItzKala, 2024-05-03:
@dubhater @gglluukk Thanks to the both of you.
rtl8xxxu is working and the network manager was the reason why my adapter's mac address kept changing automatically.
I made the NM ignore my TP-L and everything is fine now
Just one last thing before I close this issue.
Why does the NM change my TP-L's address automatically only after I load a module that can enable monitor mode?
and is there a way to stop the NM from doing that without making it ignore my TP-L?

closed 2024-05-08 by ItzKala