aircrack-ng/rtl8188eus #281: array-index-out-of-bounds in /var/lib/dkms/8188eu/5.3.9/build/core/rtw_wlan_util.c:1831:34 … when enabling hotspot … causes programs to hang and prevent shutdown.

navid-zamani, 2024-03-17:
With recent versions of the kernel (6.5.0-25 on Mint), enabling the hotspot with this driver causes the following kernel errors:

```
[10082.036833] usb 3-2.1: new high-speed USB device number 7 using xhci_hcd
[10082.139282] usb 3-2.1: New USB device found, idVendor=2357, idProduct=010c, bcdDevice= 0.00
[10082.139292] usb 3-2.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[10082.139296] usb 3-2.1: Product: 802.11n NIC
[10082.139299] usb 3-2.1: Manufacturer: Realtek
[10082.139302] usb 3-2.1: SerialNumber: 00E04C0001
[10082.318154] bFWReady == _FALSE call reset 8051...
[10082.377323] usbcore: registered new interface driver 8188eu
[10082.388310] 8188eu 3-2.1:1.0 wlan-stick: renamed from wlan0
[10082.926914] ==> rtl8188e_iol_efuse_patch
[10125.354855] ================================================================================
[10125.354862] UBSAN: array-index-out-of-bounds in /var/lib/dkms/8188eu/5.3.9/build/core/rtw_wlan_util.c:1817:48
[10125.354866] index 1 is out of range for type 'u8 [1]'
[10125.354869] CPU: 2 PID: 842 Comm: wpa_supplicant Tainted: G           OE      6.5.0-25-generic #25~22.04.1-Ubuntu
[10125.354873] Hardware name: Micro-Star International Co., Ltd. MS-7A38/B350M PRO-VDH (MS-7A38), BIOS A.L4 05/17/2023
[10125.354875] Call Trace:
[10125.354877]  <TASK>
[10125.354880]  dump_stack_lvl+0x48/0x70
[10125.354890]  dump_stack+0x10/0x20
[10125.354894]  __ubsan_handle_out_of_bounds+0xc6/0x110
[10125.354900]  HT_caps_handler+0xc8/0x310 [8188eu]
[10125.354992]  rtw_check_beacon_data+0xabc/0xb60 [8188eu]
[10125.355090]  rtw_add_beacon+0x149/0x280 [8188eu]
[10125.355194]  cfg80211_rtw_start_ap+0x47/0xe0 [8188eu]
[10125.355298]  nl80211_start_ap+0x857/0xaf0 [cfg80211]
[10125.355372]  ? rtnl_unlock+0xe/0x20
[10125.355377]  ? nl80211_pre_doit+0x225/0x2d0 [cfg80211]
[10125.355446]  genl_family_rcv_msg_doit.isra.0+0xe8/0x150
[10125.355452]  genl_family_rcv_msg+0x180/0x250
[10125.355455]  ? __pfx_nl80211_pre_doit+0x10/0x10 [cfg80211]
[10125.355523]  ? __pfx_nl80211_start_ap+0x10/0x10 [cfg80211]
[10125.355592]  ? __pfx_nl80211_post_doit+0x10/0x10 [cfg80211]
[10125.355660]  genl_rcv_msg+0x4c/0xb0
[10125.355663]  ? __pfx_genl_rcv_msg+0x10/0x10
[10125.355666]  netlink_rcv_skb+0x5d/0x110
[10125.355671]  genl_rcv+0x28/0x50
[10125.355673]  netlink_unicast+0x1b3/0x2a0
[10125.355676]  netlink_sendmsg+0x25e/0x4e0
[10125.355680]  ____sys_sendmsg+0x3ef/0x420
[10125.355684]  ___sys_sendmsg+0x9a/0xf0
[10125.355692]  __sys_sendmsg+0x89/0xf0
[10125.355697]  __x64_sys_sendmsg+0x1d/0x30
[10125.355700]  do_syscall_64+0x5b/0x90
[10125.355704]  ? exit_to_user_mode_prepare+0x30/0xb0
[10125.355707]  ? syscall_exit_to_user_mode+0x37/0x60
[10125.355712]  ? do_syscall_64+0x67/0x90
[10125.355714]  ? do_syscall_64+0x67/0x90
[10125.355717]  ? do_syscall_64+0x67/0x90
[10125.355720]  entry_SYSCALL_64_after_hwframe+0x6e/0xd8
[10125.355725] RIP: 0033:0x79cf16f27967
[10125.355750] Code: 0f 00 f7 d8 64 89 02 48 c7 c0 ff ff ff ff eb b9 0f 1f 00 f3 0f 1e fa 64 8b 04 25 18 00 00 00 85 c0 75 10 b8 2e 00 00 00 0f 05 <48> 3d 00 f0 ff ff 77 51 c3 48 83 ec 28 89 54 24 1c 48 89 74 24 10
[10125.355752] RSP: 002b:00007ffd19ccb188 EFLAGS: 00000246 ORIG_RAX: 000000000000002e
[10125.355756] RAX: ffffffffffffffda RBX: 00006342ca624aa0 RCX: 000079cf16f27967
[10125.355758] RDX: 0000000000000000 RSI: 00007ffd19ccb1c0 RDI: 0000000000000006
[10125.355760] RBP: 00006342ca624d80 R08: 0000000000000004 R09: 00006342ca72cfd0
[10125.355761] R10: 00007ffd19ccb2a0 R11: 0000000000000246 R12: 00006342ca74c9d0
[10125.355763] R13: 00007ffd19ccb1c0 R14: 0000000000000000 R15: 0000000000000000
[10125.355767]  </TASK>
[10125.355768] ================================================================================
[10125.355770] ================================================================================
[10125.355772] UBSAN: array-index-out-of-bounds in /var/lib/dkms/8188eu/5.3.9/build/core/rtw_wlan_util.c:1822:75
[10125.355775] index 2 is out of range for type 'u8 [1]'
[10125.355777] CPU: 2 PID: 842 Comm: wpa_supplicant Tainted: G           OE      6.5.0-25-generic #25~22.04.1-Ubuntu
[10125.355780] Hardware name: Micro-Star International Co., Ltd. MS-7A38/B350M PRO-VDH (MS-7A38), BIOS A.L4 05/17/2023
[10125.355782] Call Trace:
[10125.355783]  <TASK>
[10125.355784]  dump_stack_lvl+0x48/0x70
[... 123 lines cut ...]
[10125.357971]  ? exit_to_user_mode_prepare+0x30/0xb0
[10125.357973]  ? syscall_exit_to_user_mode+0x37/0x60
[10125.357977]  ? do_syscall_64+0x67/0x90
[10125.357979]  ? do_syscall_64+0x67/0x90
[10125.357982]  ? do_syscall_64+0x67/0x90
[10125.357984]  entry_SYSCALL_64_after_hwframe+0x6e/0xd8
[10125.357988] RIP: 0033:0x79cf16f27967
[10125.358005] Code: 0f 00 f7 d8 64 89 02 48 c7 c0 ff ff ff ff eb b9 0f 1f 00 f3 0f 1e fa 64 8b 04 25 18 00 00 00 85 c0 75 10 b8 2e 00 00 00 0f 05 <48> 3d 00 f0 ff ff 77 51 c3 48 83 ec 28 89 54 24 1c 48 89 74 24 10
[10125.358007] RSP: 002b:00007ffd19ccb188 EFLAGS: 00000246 ORIG_RAX: 000000000000002e
[10125.358009] RAX: ffffffffffffffda RBX: 00006342ca624aa0 RCX: 000079cf16f27967
[10125.358010] RDX: 0000000000000000 RSI: 00007ffd19ccb1c0 RDI: 0000000000000006
[10125.358011] RBP: 00006342ca624d80 R08: 0000000000000004 R09: 00006342ca72cfd0
[10125.358013] R10: 00007ffd19ccb2a0 R11: 0000000000000246 R12: 00006342ca74c9d0
[10125.358014] R13: 00007ffd19ccb1c0 R14: 0000000000000000 R15: 0000000000000000
[10125.358017]  </TASK>
[10125.358018] ================================================================================
```

(It looks like repeated, but they all happen right away, so I thought it’s better to include them all.)

This then sometimes (the more likely the longer you use it) leads to NetworkManager using 100% CPU (on a single core), as well as all programs that use networking to completely hang, to a point where even SIGKILLing them won’t work. This prevents logging in or opening a shell to fix anything, as well as shutting down. (Alt-SysRq-REISUB works, but on Mint isn’t enabled by default.) (Hibernation also seems to be affected somehow, as it won’t wake up but boot instead. I could not find out why yet, as I had to disable the driver, as the PC is needed for work.)

It also happens with the fork by [gglluukk](https://github.com/gglluukk/rtl8188eus) which is a few commits ahead.

If you need any further info to reproduce it, or need me to do some diagnostics with access to the actual hardware, feel free to ask. I’m a programmer too.

gglluukk, 2024-03-20:
@navid-zamani i can't reproduce this error, but i extended array to hopefully prevent this error from happening. try to renew https://github.com/gglluukk/rtl8188eus

navid-zamani, 2024-03-21:
Thank you!

I just `git pull`ed the changes (`git log` confirms it),  
and confirmed `dkms-install.sh` installed the new version of the module  
and it’s the one that’s loaded.  
Yet I still get this error here in `dmesg`:

When plugging the device in (seems OK):
```
[Mär21 09:51] usb 3-2.1: new high-speed USB device number 6 using xhci_hcd
[  +0,102233] usb 3-2.1: New USB device found, idVendor=2357, idProduct=010c, bcdDevice= 0.00
[  +0,000011] usb 3-2.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  +0,000004] usb 3-2.1: Product: 802.11n NIC
[  +0,000003] usb 3-2.1: Manufacturer: Realtek
[  +0,000003] usb 3-2.1: SerialNumber: 00E04C0001
[  +0,011965] bFWReady == _FALSE call reset 8051...
[  +0,085528] 8188eu 3-2.1:1.0 wlan-stick: renamed from wlan0
[  +0,530134] ==> rtl8188e_iol_efuse_patch
```
When connecting (`nmcli connection up` for a hotspot):
```
[ +28,310138] ================================================================================
[  +0,000007] UBSAN: array-index-out-of-bounds in /var/lib/dkms/8188eu/5.3.9/build/core/rtw_wlan_util.c:1817:48
[  +0,000003] index 8 is out of range for type 'u8 [8]'
[  +0,000002] CPU: 2 PID: 672 Comm: wpa_supplicant Tainted: G           OE      6.5.0-25-generic #25~22.04.1-Ubuntu
[  +0,000003] Hardware name: Micro-Star International Co., Ltd. MS-7A38/B350M PRO-VDH (MS-7A38), BIOS A.L4 05/17/2023
[  +0,000001] Call Trace:
[  +0,000002]  <TASK>
[  +0,000003]  dump_stack_lvl+0x48/0x70
[  +0,000009]  dump_stack+0x10/0x20
[  +0,000002]  __ubsan_handle_out_of_bounds+0xc6/0x110
[  +0,000005]  HT_caps_handler+0xbc/0x2b0 [8188eu]
[  +0,000062]  rtw_check_beacon_data+0xabc/0xb60 [8188eu]
[  +0,000065]  rtw_add_beacon+0x149/0x280 [8188eu]
[  +0,000070]  cfg80211_rtw_start_ap+0x47/0xe0 [8188eu]
[  +0,000069]  nl80211_start_ap+0x857/0xaf0 [cfg80211]
[  +0,000052]  ? rtnl_unlock+0xe/0x20
[  +0,000004]  ? nl80211_pre_doit+0x225/0x2d0 [cfg80211]
[  +0,000053]  genl_family_rcv_msg_doit.isra.0+0xe8/0x150
[  +0,000007]  genl_family_rcv_msg+0x180/0x250
[  +0,000004]  ? __pfx_nl80211_pre_doit+0x10/0x10 [cfg80211]
[  +0,000067]  ? __pfx_nl80211_start_ap+0x10/0x10 [cfg80211]
[  +0,000058]  ? __pfx_nl80211_post_doit+0x10/0x10 [cfg80211]
[  +0,000057]  genl_rcv_msg+0x4c/0xb0
[  +0,000003]  ? __pfx_genl_rcv_msg+0x10/0x10
[  +0,000002]  netlink_rcv_skb+0x5d/0x110
[  +0,000004]  genl_rcv+0x28/0x50
[  +0,000002]  netlink_unicast+0x1b3/0x2a0
[  +0,000003]  netlink_sendmsg+0x25e/0x4e0
[  +0,000003]  ____sys_sendmsg+0x3ef/0x420
[  +0,000003]  ___sys_sendmsg+0x9a/0xf0
[  +0,000006]  __sys_sendmsg+0x89/0xf0
[  +0,000005]  __x64_sys_sendmsg+0x1d/0x30
[  +0,000002]  do_syscall_64+0x5b/0x90
[  +0,000004]  ? syscall_exit_to_user_mode+0x37/0x60
[  +0,000003]  ? do_syscall_64+0x67/0x90
[  +0,000003]  ? do_syscall_64+0x67/0x90
[  +0,000002]  ? do_syscall_64+0x67/0x90
[  +0,000003]  entry_SYSCALL_64_after_hwframe+0x6e/0xd8
[  +0,000003] RIP: 0033:0x79deb8d27967
[  +0,000026] Code: 0f 00 f7 d8 64 89 02 48 c7 c0 ff ff ff ff eb b9 0f 1f 00 f3 0f 1e fa 64 8b 04 25 18 00 00 00 85 c0 75 10 b8 2e 00 00 00 0f 05 <48> 3d 00 f0 ff ff 77 51 c3 48 83 ec 28 89 54 24 1c 48 89 74 24 10
[  +0,000002] RSP: 002b:00007ffe7af48058 EFLAGS: 00000246 ORIG_RAX: 000000000000002e
[  +0,000003] RAX: ffffffffffffffda RBX: 00005b00f8230aa0 RCX: 000079deb8d27967
[  +0,000002] RDX: 0000000000000000 RSI: 00007ffe7af48090 RDI: 0000000000000006
[  +0,000001] RBP: 00005b00f8230d80 R08: 0000000000000004 R09: 00005b00f833fb70
[  +0,000001] R10: 00007ffe7af48170 R11: 0000000000000246 R12: 00005b00f8348880
[  +0,000002] R13: 00007ffe7af48090 R14: 0000000000000000 R15: 0000000000000000
[  +0,000003]  </TASK>
[  +0,000001] ================================================================================
```

More errors like this happen, when I keep using it.

Seems it didn’t help. :-/

navid-zamani, 2024-03-21:
Thank you, but the error still happened.

I narrowed down the value, and the smallest one that works is … `26`.

So this is the patch that makes work:
```patch
diff --git a/include/wlan_bssdef.h b/include/wlan_bssdef.h
index d547b65..101fcfc 100644
--- a/include/wlan_bssdef.h
+++ b/include/wlan_bssdef.h
@@ -95,7 +95,7 @@ typedef struct _NDIS_802_11_FIXED_IEs {
 typedef struct _NDIS_802_11_VARIABLE_IEs {
        UCHAR  ElementID;
        UCHAR  Length;
-       UCHAR  data[8];
+       UCHAR  data[26];
 } NDIS_802_11_VARIABLE_IEs, *PNDIS_802_11_VARIABLE_IEs;
 
 
@@ -343,7 +343,7 @@ typedef struct _NDIS_802_11_FIXED_IEs {
 typedef struct _NDIS_802_11_VARIABLE_IEs {
        UCHAR  ElementID;
        UCHAR  Length;
-       UCHAR  data[8];
+       UCHAR  data[26];
 } NDIS_802_11_VARIABLE_IEs, *PNDIS_802_11_VARIABLE_IEs;
```

I am really curious what this is for, …  
(and if it’s a bug that it needs to be that big here.)

gglluukk, 2024-03-21:
in this case i set `data` array length to:
```
UCHAR  data[255];
```
since 255 -- maximum value of `(pIE->Length)`: 
https://github.com/gglluukk/rtl8188eus/blob/v5.3.9/core/rtw_wlan_util.c#L1813

dubhater, 2024-03-21:
``UCHAR data[];`` also works.

gglluukk, 2024-03-21:
i was incorrect since data[] is "flexible array member" and not stand-alone variable, correct example:

```
#include <stdio.h>

#define UCHAR           unsigned char

typedef struct _check1 {
        UCHAR  ElementID;
        UCHAR  Length;
        UCHAR  data[255];
} check1;

typedef struct _check2 {
        UCHAR  ElementID;
        UCHAR  Length;
        UCHAR  data[];
} check2;

int main() {
    check1 c1;
    check2 c2;

    printf("%lu %lu\n", sizeof(c1), sizeof(c2));
}
```
so using `data[]` might be better here hopefully to further correct memory allocations