aircrack-ng/rtl8188eus #283: Unable to compile program

tclyuuuu, 2024-05-17:
hello
I got this error while compiling the driver，
make                                          
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/6.6.15-amd64/build M=/home/kali/Desktop/wangka/rtl8188gu  modules
make[1]: Entering directory '/usr/src/linux-headers-6.6.15-amd64'
  CC [M]  /home/kali/Desktop/wangka/rtl8188gu/os_dep/linux/ioctl_cfg80211.o
/home/kali/Desktop/wangka/rtl8188gu/os_dep/linux/ioctl_cfg80211.c: In function ‘rtw_cfg80211_ch_switch_notify’:
/home/kali/Desktop/wangka/rtl8188gu/os_dep/linux/ioctl_cfg80211.c:398:9: error: too few arguments to function ‘cfg80211_ch_switch_notify’
  398 |         cfg80211_ch_switch_notify(adapter->pnetdev, &chdef, 0);
      |         ^~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/kali/Desktop/wangka/rtl8188gu/include/osdep_service_linux.h:93,
                 from /home/kali/Desktop/wangka/rtl8188gu/include/osdep_service.h:47,
                 from /home/kali/Desktop/wangka/rtl8188gu/include/drv_types.h:27,
                 from /home/kali/Desktop/wangka/rtl8188gu/os_dep/linux/ioctl_cfg80211.c:17:
/usr/src/linux-headers-6.6.15-common/include/net/cfg80211.h:8597:6: note: declared here
 8597 | void cfg80211_ch_switch_notify(struct net_device *dev,
      |      ^~~~~~~~~~~~~~~~~~~~~~~~~
/home/kali/Desktop/wangka/rtl8188gu/os_dep/linux/ioctl_cfg80211.c: In function ‘cfg80211_rtw_scan’:
/home/kali/Desktop/wangka/rtl8188gu/os_dep/linux/ioctl_cfg80211.c:2619:33: warning: the comparison will always evaluate as ‘true’ for the address of ‘ssid’ will never be NULL [-Waddress]
 2619 |                 if (ssids->ssid != NULL
      |                                 ^~
/usr/src/linux-headers-6.6.15-common/include/net/cfg80211.h:2441:12: note: ‘ssid’ declared here
 2441 |         u8 ssid[IEEE80211_MAX_SSID_LEN];
      |            ^~~~
make[3]: *** [/usr/src/linux-headers-6.6.15-common/scripts/Makefile.build:248: /home/kali/Desktop/wangka/rtl8188gu/os_dep/linux/ioctl_cfg80211.o] Error 1
make[2]: *** [/usr/src/linux-headers-6.6.15-common/Makefile:1938: /home/kali/Desktop/wangka/rtl8188gu] Error 2
make[1]: *** [/usr/src/linux-headers-6.6.15-common/Makefile:246: __sub-make] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-6.6.15-amd64'
make: *** [Makefile:1895: modules] Error 2

uname -r
6.6.15-amd64
How should this problem be solved?

gglluukk, 2024-05-17:
6.6.30-gentoo-dist
https://github.com/gglluukk/rtl8188eus
compiled ok

tclyuuuu, 2024-05-18:
Hello, I have been able to successfully compile through your method, but I still cannot recognize the network card。
This is my 
Bus 001 Device 002: ID 0bda:b711 Realtek Semiconductor Corp. RTL8188GU 802.11n WLAN Adapter (After Modeswitch)
[image]

tclyuuuu, 2024-05-18:
> 6.6.30-gentoo-dist https://github.com/gglluukk/rtl8188eus compiled ok 

Hello, I have been able to successfully compile through your method, but I still cannot recognize the network card。
This is my 
Bus 001 Device 002: ID 0bda:b711 Realtek Semiconductor Corp. RTL8188GU 802.11n WLAN Adapter (After Modeswitch)

[image]

gglluukk, 2024-05-18:
i've googled for you and found the following:
https://github.com/lwfinger/rtl8188gu

also you might try to use already in-kernel module: `rtl8xxxu`