aircrack-ng/rtl8188eus #285: How I can point the `rtl8188eus` build towards the cloned android kernel source instead of the system one?

pc-magas, 2024-06-10:
I try to build the driver for android so I can load it into the emulator. So far I have installed both android sdk, afterwards I have installed ndk using sdk manager:

```
sdkmanager "ndk;27.0.11902837"
```

And I have exported the following environmental variables:

```
export ANDROID_SDK_ROOT=/home/pcmagas/Android
export ANDROID_HOME=/home/pcmagas/Android

export PATH=$PATH:$ANDROID_SDK_ROOT/cmdline-tools/latest/bin
export PATH=$PATH:$ANDROID_SDK_ROOT/emulator

export NDK_HOME=$ANDROID_HOME/ndk/27.0.11902837
export PATH=$PATH:$NDK_HOME/toolchains/llvm/prebuilt/linux-x86_64/bin
```

Then I cloned the common kernel:

```
git clone https://android.googlesource.com/kernel/common.git
cd common
git checkout android14-6.1-2024-04_r11
```

Then I cloned the driver source:

```
cd ~
git clone https://github.com/aircrack-ng/rtl8188eus.git
cd rtl8188eus
 make ARCH=x86_64 CROSS_COMPILE=x86_64-linux-android- CC=$TOOLCHAIN/x86_64-linux-android25-clang 
```

But I get the error:

```
make[1]: Entering directory '/usr/src/linux-headers-5.15.0-112-generic'
arch/x86/Makefile:142: CONFIG_X86_X32 enabled but no binutils support
make[1]: /x86_64-linux-android25-clang: Δεν υπάρχει τέτοιο αρχείο ή κατάλογος
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: gcc (Ubuntu 11.4.0-1ubuntu1~22.04) 11.4.0
  You are using:           
/bin/sh: 1: /x86_64-linux-android25-clang: not found
(standard_in) 1: syntax error
  CC [M]  /home/pcmagas/dlink-decompile/rtl8188eus/core/rtw_cmd.o
/bin/sh: 1: /x86_64-linux-android25-clang: not found
make[2]: *** [scripts/Makefile.build:297: /home/pcmagas/dlink-decompile/rtl8188eus/core/rtw_cmd.o] Error 127
make[1]: *** [Makefile:1911: /home/pcmagas/dlink-decompile/rtl8188eus] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.15.0-112-generic'
make: *** [Makefile:2065: modules] Error 2

```

As you can see it is pointed towards the system's linux kernel and not towards the android kernel that I have cloned upon. How I can fix this?

pc-magas, 2024-06-14:
Update I tried with:

```
make ARCH=x86_64 CROSS_COMPILE=x86_64-linux-android- CC=x86_64-linux-android25-clang KSRC=~/dlink-decompile/common KDIR=~/dlink-decompile/common 
```

But I got:

```
make[1]: Entering directory '/home/pcmagas/dlink-decompile/common'
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: gcc (Ubuntu 11.4.0-1ubuntu1~22.04) 11.4.0
  You are using:           Android (11889484, +pgo, +bolt, +lto, +mlgo, based on r522817) clang version 18.0.1 (https://android.googlesource.com/toolchain/llvm-project d8003a456d14a3deb8054cdaa529ffbf02d9b262)
make[2]: *** No rule to make target 'scripts/mod/modpost', needed by '/home/pcmagas/dlink-decompile/rtl8188eus/Module.symvers'.  Stop.
make[1]: *** [Makefile:2009: modpost] Error 2
make[1]: Leaving directory '/home/pcmagas/dlink-decompile/common'
make: *** [Makefile:2065: modules] Error 2

```

And I am stuck

gglluukk, 2024-06-16:
hi,
never tried that but:
1. doesn't it say that you try to use other compiler (`clang`) than kernel needs (`gcc`)?
2. do you use same kernel version sources as your device runs?

pc-magas, 2024-06-16:
> hi, never tried that but:
> 
>     1. doesn't it say that you try to use other compiler (`clang`) than kernel needs (`gcc`)?
> 
>     2. do you use same kernel version sources as your device runs?

Then I switched into GCC via trying:

```
 make -C /home/pcmagas/dlink-decompile/common M=/home/pcmagas/dlink-decompile/rtl8188eus  KSRC=~/dlink-decompile/common KDIR=~/dlink-decompile/common ARCH=x86_64
make: Entering directory '/home/pcmagas/dlink-decompile/common'
  MODPOST /home/pcmagas/dlink-decompile/rtl8188eus/Module.symvers
WARNING: Module.symvers is missing.
         Modules may not have dependencies or modversions.
         You may get many unresolved symbol warnings.
make: Leaving directory '/home/pcmagas/dlink-decompile/common'

```

My folder structure is:

```
dlink-decompile
-- common << Android Kernel
-- rtl8188eus
```

I build the android kernel using:

```
mkdir dlink-decompile
cd  dlink-decompile
git clone https://android.googlesource.com/kernel/common.git
git checkout android14-6.1-2024-04\_r11

$ make mrproper
$ make menuconfig
$ make scripts
$ make -j$(nproc)
```

Then I tried building te driver like this:

```
git clone https://github.com/aircrack-ng/rtl8188eus.git
 make -C /home/pcmagas/dlink-decompile/common M=/home/pcmagas/dlink-decompile/rtl8188eus  KSRC=~/dlink-decompile/common KDIR=~/dlink-decompile/common ARCH=x86_64
```

And so far I saw the file `Module.symvers` to be missing:

```
make -C /home/pcmagas/dlink-decompile/common M=/home/pcmagas/dlink-decompile/rtl8188eus  KSRC=~/dlink-decompile/common KDIR=~/dlink-decompile/common ARCH=x86_64
make: Entering directory '/home/pcmagas/dlink-decompile/common'
  MODPOST /home/pcmagas/dlink-decompile/rtl8188eus/Module.symvers
WARNING: symversis missing.
         Modules may not have dependencies or modversions.
         You may get many unresolved symbol warnings.
make: Leaving directory '/home/pcmagas/dlink-decompile/common'

```

How I can make the miussing file???

gglluukk, 2024-06-16:
i don't think that you have any issues with the `rtl8188eus` driver itself but with how to compile and install *any* driver under your android, so it could be better to search and ask in android-related groups/communities how to do so.