aircrack-ng/rtl8188eus #287: Kernel 5.10 ERROR: modpost

LazalazaRus, 2024-07-23:
How can I fix this behavior?

ERROR: modpost: "proc_get_parent_data" [/root/Desktop/rtl8188eus/8188eu.ko] undefined!
ERROR: modpost: "kernel_read" [/root/Desktop/rtl8188eus/8188eu.ko] undefined!
ERROR: modpost: "yield" [/root/Desktop/rtl8188eus/8188eu.ko] undefined!

```
make ARCH=arm64 CROSS_COMPILE=aarch64-linux-gnu- -C /lib/modules/5.10.198-android12-9/build M=/root/Desktop/rtl8188eus  modules
make[1]: Entering directory '/root/Desktop/a24/out/target/product/a24/obj/KERNEL_OBJ/kernel-5.10'
  CC [M]  /root/Desktop/rtl8188eus/core/rtw_cmd.o
  CC [M]  /root/Desktop/rtl8188eus/core/rtw_security.o
  CC [M]  /root/Desktop/rtl8188eus/core/rtw_debug.o
  CC [M]  /root/Desktop/rtl8188eus/core/rtw_io.o
  CC [M]  /root/Desktop/rtl8188eus/core/rtw_ioctl_query.o
  CC [M]  /root/Desktop/rtl8188eus/core/rtw_ioctl_set.o
  CC [M]  /root/Desktop/rtl8188eus/core/rtw_ieee80211.o
  CC [M]  /root/Desktop/rtl8188eus/core/rtw_mlme.o
  CC [M]  /root/Desktop/rtl8188eus/core/rtw_mlme_ext.o
  CC [M]  /root/Desktop/rtl8188eus/core/rtw_mi.o
  CC [M]  /root/Desktop/rtl8188eus/core/rtw_wlan_util.o
  CC [M]  /root/Desktop/rtl8188eus/core/rtw_vht.o
  CC [M]  /root/Desktop/rtl8188eus/core/rtw_pwrctrl.o
  CC [M]  /root/Desktop/rtl8188eus/core/rtw_rf.o
  CC [M]  /root/Desktop/rtl8188eus/core/rtw_recv.o
  CC [M]  /root/Desktop/rtl8188eus/core/rtw_sta_mgt.o
  CC [M]  /root/Desktop/rtl8188eus/core/rtw_ap.o
  CC [M]  /root/Desktop/rtl8188eus/core/mesh/rtw_mesh.o
  CC [M]  /root/Desktop/rtl8188eus/core/mesh/rtw_mesh_pathtbl.o
  CC [M]  /root/Desktop/rtl8188eus/core/mesh/rtw_mesh_hwmp.o
  CC [M]  /root/Desktop/rtl8188eus/core/rtw_xmit.o
  CC [M]  /root/Desktop/rtl8188eus/core/rtw_p2p.o
  CC [M]  /root/Desktop/rtl8188eus/core/rtw_rson.o
  CC [M]  /root/Desktop/rtl8188eus/core/rtw_tdls.o
  CC [M]  /root/Desktop/rtl8188eus/core/rtw_br_ext.o
  CC [M]  /root/Desktop/rtl8188eus/core/rtw_iol.o
  CC [M]  /root/Desktop/rtl8188eus/core/rtw_sreset.o
  CC [M]  /root/Desktop/rtl8188eus/core/rtw_btcoex_wifionly.o
  CC [M]  /root/Desktop/rtl8188eus/core/rtw_btcoex.o
  CC [M]  /root/Desktop/rtl8188eus/core/rtw_beamforming.o
  CC [M]  /root/Desktop/rtl8188eus/core/rtw_odm.o
  CC [M]  /root/Desktop/rtl8188eus/core/rtw_rm.o
  CC [M]  /root/Desktop/rtl8188eus/core/rtw_rm_fsm.o
  CC [M]  /root/Desktop/rtl8188eus/core/efuse/rtw_efuse.o
  CC [M]  /root/Desktop/rtl8188eus/os_dep/osdep_service.o
  CC [M]  /root/Desktop/rtl8188eus/os_dep/linux/os_intfs.o
  CC [M]  /root/Desktop/rtl8188eus/os_dep/linux/usb_intf.o
  CC [M]  /root/Desktop/rtl8188eus/os_dep/linux/usb_ops_linux.o
  CC [M]  /root/Desktop/rtl8188eus/os_dep/linux/ioctl_linux.o
  CC [M]  /root/Desktop/rtl8188eus/os_dep/linux/xmit_linux.o
  CC [M]  /root/Desktop/rtl8188eus/os_dep/linux/mlme_linux.o
  CC [M]  /root/Desktop/rtl8188eus/os_dep/linux/recv_linux.o
  CC [M]  /root/Desktop/rtl8188eus/os_dep/linux/ioctl_cfg80211.o
  CC [M]  /root/Desktop/rtl8188eus/os_dep/linux/rtw_cfgvendor.o
  CC [M]  /root/Desktop/rtl8188eus/os_dep/linux/wifi_regd.o
  CC [M]  /root/Desktop/rtl8188eus/os_dep/linux/rtw_android.o
  CC [M]  /root/Desktop/rtl8188eus/os_dep/linux/rtw_proc.o
  CC [M]  /root/Desktop/rtl8188eus/os_dep/linux/rtw_rhashtable.o
  CC [M]  /root/Desktop/rtl8188eus/os_dep/linux/ioctl_mp.o
  CC [M]  /root/Desktop/rtl8188eus/hal/hal_intf.o
  CC [M]  /root/Desktop/rtl8188eus/hal/hal_com.o
  CC [M]  /root/Desktop/rtl8188eus/hal/hal_com_phycfg.o
  CC [M]  /root/Desktop/rtl8188eus/hal/hal_phy.o
  CC [M]  /root/Desktop/rtl8188eus/hal/hal_dm.o
  CC [M]  /root/Desktop/rtl8188eus/hal/hal_dm_acs.o
  CC [M]  /root/Desktop/rtl8188eus/hal/hal_btcoex_wifionly.o
  CC [M]  /root/Desktop/rtl8188eus/hal/hal_btcoex.o
  CC [M]  /root/Desktop/rtl8188eus/hal/hal_mp.o
  CC [M]  /root/Desktop/rtl8188eus/hal/hal_mcc.o
  CC [M]  /root/Desktop/rtl8188eus/hal/hal_hci/hal_usb.o
  CC [M]  /root/Desktop/rtl8188eus/hal/led/hal_led.o
  CC [M]  /root/Desktop/rtl8188eus/hal/led/hal_usb_led.o
  CC [M]  /root/Desktop/rtl8188eus/hal/HalPwrSeqCmd.o
  CC [M]  /root/Desktop/rtl8188eus/hal/rtl8188e/Hal8188EPwrSeq.o
  CC [M]  /root/Desktop/rtl8188eus/hal/rtl8188e/rtl8188e_xmit.o
  CC [M]  /root/Desktop/rtl8188eus/hal/rtl8188e/rtl8188e_sreset.o
  CC [M]  /root/Desktop/rtl8188eus/hal/rtl8188e/rtl8188e_hal_init.o
  CC [M]  /root/Desktop/rtl8188eus/hal/rtl8188e/rtl8188e_phycfg.o
  CC [M]  /root/Desktop/rtl8188eus/hal/rtl8188e/rtl8188e_rf6052.o
  CC [M]  /root/Desktop/rtl8188eus/hal/rtl8188e/rtl8188e_dm.o
  CC [M]  /root/Desktop/rtl8188eus/hal/rtl8188e/rtl8188e_rxdesc.o
  CC [M]  /root/Desktop/rtl8188eus/hal/rtl8188e/rtl8188e_cmd.o
  CC [M]  /root/Desktop/rtl8188eus/hal/rtl8188e/hal8188e_s_fw.o
  CC [M]  /root/Desktop/rtl8188eus/hal/rtl8188e/hal8188e_t_fw.o
  CC [M]  /root/Desktop/rtl8188eus/hal/rtl8188e/usb/usb_halinit.o
  CC [M]  /root/Desktop/rtl8188eus/hal/rtl8188e/usb/rtl8188eu_led.o
  CC [M]  /root/Desktop/rtl8188eus/hal/rtl8188e/usb/rtl8188eu_xmit.o
  CC [M]  /root/Desktop/rtl8188eus/hal/rtl8188e/usb/rtl8188eu_recv.o
  CC [M]  /root/Desktop/rtl8188eus/hal/rtl8188e/usb/usb_ops_linux.o
  CC [M]  /root/Desktop/rtl8188eus/hal/efuse/rtl8188e/HalEfuseMask8188E_USB.o
  CC [M]  /root/Desktop/rtl8188eus/hal/phydm/phydm_debug.o
  CC [M]  /root/Desktop/rtl8188eus/hal/phydm/phydm_antdiv.o
  CC [M]  /root/Desktop/rtl8188eus/hal/phydm/phydm_soml.o
  CC [M]  /root/Desktop/rtl8188eus/hal/phydm/phydm_smt_ant.o
  CC [M]  /root/Desktop/rtl8188eus/hal/phydm/phydm_antdect.o
  CC [M]  /root/Desktop/rtl8188eus/hal/phydm/phydm_interface.o
  CC [M]  /root/Desktop/rtl8188eus/hal/phydm/phydm_phystatus.o
  CC [M]  /root/Desktop/rtl8188eus/hal/phydm/phydm_hwconfig.o
  CC [M]  /root/Desktop/rtl8188eus/hal/phydm/phydm.o
  CC [M]  /root/Desktop/rtl8188eus/hal/phydm/phydm_dig.o
  CC [M]  /root/Desktop/rtl8188eus/hal/phydm/phydm_pathdiv.o
  CC [M]  /root/Desktop/rtl8188eus/hal/phydm/phydm_rainfo.o
  CC [M]  /root/Desktop/rtl8188eus/hal/phydm/phydm_dynamictxpower.o
  CC [M]  /root/Desktop/rtl8188eus/hal/phydm/phydm_adaptivity.o
  CC [M]  /root/Desktop/rtl8188eus/hal/phydm/phydm_cfotracking.o
  CC [M]  /root/Desktop/rtl8188eus/hal/phydm/phydm_noisemonitor.o
  CC [M]  /root/Desktop/rtl8188eus/hal/phydm/phydm_beamforming.o
  CC [M]  /root/Desktop/rtl8188eus/hal/phydm/phydm_dfs.o
  CC [M]  /root/Desktop/rtl8188eus/hal/phydm/txbf/halcomtxbf.o
  CC [M]  /root/Desktop/rtl8188eus/hal/phydm/txbf/haltxbfinterface.o
  CC [M]  /root/Desktop/rtl8188eus/hal/phydm/txbf/phydm_hal_txbf_api.o
  CC [M]  /root/Desktop/rtl8188eus/hal/phydm/phydm_adc_sampling.o
  CC [M]  /root/Desktop/rtl8188eus/hal/phydm/phydm_ccx.o
  CC [M]  /root/Desktop/rtl8188eus/hal/phydm/phydm_psd.o
  CC [M]  /root/Desktop/rtl8188eus/hal/phydm/phydm_primary_cca.o
  CC [M]  /root/Desktop/rtl8188eus/hal/phydm/phydm_cck_pd.o
  CC [M]  /root/Desktop/rtl8188eus/hal/phydm/phydm_rssi_monitor.o
  CC [M]  /root/Desktop/rtl8188eus/hal/phydm/phydm_auto_dbg.o
  CC [M]  /root/Desktop/rtl8188eus/hal/phydm/phydm_math_lib.o
  CC [M]  /root/Desktop/rtl8188eus/hal/phydm/phydm_api.o
  CC [M]  /root/Desktop/rtl8188eus/hal/phydm/phydm_pow_train.o
  CC [M]  /root/Desktop/rtl8188eus/hal/phydm/halrf/halrf.o
  CC [M]  /root/Desktop/rtl8188eus/hal/phydm/halrf/halphyrf_ce.o
  CC [M]  /root/Desktop/rtl8188eus/hal/phydm/halrf/halrf_powertracking_ce.o
  CC [M]  /root/Desktop/rtl8188eus/hal/phydm/halrf/halrf_powertracking.o
  CC [M]  /root/Desktop/rtl8188eus/hal/phydm/halrf/halrf_kfree.o
  CC [M]  /root/Desktop/rtl8188eus/hal/phydm/rtl8188e/halhwimg8188e_mac.o
  CC [M]  /root/Desktop/rtl8188eus/hal/phydm/rtl8188e/halhwimg8188e_bb.o
  CC [M]  /root/Desktop/rtl8188eus/hal/phydm/rtl8188e/halhwimg8188e_rf.o
  CC [M]  /root/Desktop/rtl8188eus/hal/phydm/halrf/rtl8188e/halrf_8188e_ce.o
  CC [M]  /root/Desktop/rtl8188eus/hal/phydm/rtl8188e/phydm_regconfig8188e.o
  CC [M]  /root/Desktop/rtl8188eus/hal/phydm/rtl8188e/hal8188erateadaptive.o
  CC [M]  /root/Desktop/rtl8188eus/hal/phydm/rtl8188e/phydm_rtl8188e.o
  CC [M]  /root/Desktop/rtl8188eus/platform/platform_ops.o
  CC [M]  /root/Desktop/rtl8188eus/core/rtw_mp.o
  AR [M]  /root/Desktop/rtl8188eus/8188eu.o
+ [ -n  ]
+ echo -n -g3fae7237ba12-dirty
+ set +x
  LTO [M] /root/Desktop/rtl8188eus/8188eu.lto.o
  MODPOST /root/Desktop/rtl8188eus/Module.symvers
ERROR: modpost: "proc_get_parent_data" [/root/Desktop/rtl8188eus/8188eu.ko] undefined!
ERROR: modpost: "kernel_read" [/root/Desktop/rtl8188eus/8188eu.ko] undefined!
ERROR: modpost: "yield" [/root/Desktop/rtl8188eus/8188eu.ko] undefined!
make[3]: *** [/root/Desktop/a24/kernel-5.10/scripts/Makefile.modpost:168: /root/Desktop/rtl8188eus/Module.symvers] Error 1
make[3]: *** Deleting file '/root/Desktop/rtl8188eus/Module.symvers'
make[2]: *** [/root/Desktop/a24/kernel-5.10/Makefile:1851: modules] Error 2
make[1]: *** [/root/Desktop/a24/kernel-5.10/Makefile:192: __sub-make] Error 2
make[1]: Leaving directory '/root/Desktop/a24/out/target/product/a24/obj/KERNEL_OBJ/kernel-5.10'
make: *** [Makefile:2065: modules] Error 2
```

gglluukk, 2024-07-23:
it seems to be similar to this issue:
https://github.com/cilynx/rtl88x2bu/issues/185

if it's the only way to add missing calls by copying function's bodies from kernel sources to the driver's sources which could be complicated since each call can be connected to other function etc:
https://elixir.bootlin.com/linux/v6.10/source/fs/proc/generic.c#L782
https://elixir.bootlin.com/linux/v5.10.222/source/fs/read_write.c#L465
https://elixir.bootlin.com/linux/v5.10.222/source/kernel/sched/core.c#L6189

LazalazaRus, 2024-07-23:
> похоже, это похоже на эту проблему: [цилинкс/rtl88x2bu#185 ](https://github.com/cilynx/rtl88x2bu/issues/185)
> 
> если это единственный способ добавить недостающие вызовы путем копирования тел функций из исходных кодов ядра в исходные коды драйвера, что может оказаться сложным, поскольку каждый вызов может быть связан с другой функцией и т. д.: [https://elixir.bootlin.com/linux/v6.10/source/fs/proc/generic.c#L782 ](https://elixir.bootlin.com/linux/v6.10/source/fs/proc/generic.c#L782) [https://elixir.bootlin.com/linux/v5.10.222/source/fs/read_write.c#L465 ](https://elixir.bootlin.com/linux/v5.10.222/source/fs/read_write.c#L465) [https://elixir.bootlin.com/linux/v5.10.222/source/kernel/sched/core.c#L6189 ](https://elixir.bootlin.com/linux/v5.10.222/source/kernel/sched/core.c#L6189)

I do not deny that this can be tried, but where do I make changes to which "<file>.c", mod post only lets me know /root/Desktop/rtl8188eus/8188eu.ko ?

LazalazaRus, 2024-07-26:
If I keep it more or less brief, at first we tried using the original kernel headers from kernel.org to compile the module. In the end, we got an "invalid module format" error, which indicated that the module was not built for the correct kernel. After that, I tried changing vermagic, but it didn't help. Later, I got to the GKI kernel of the same version from Google — the same result. The first problem was with the headers; we needed to use the ones patched by Samsung, but at that time, they simply didn't want to publish my new kernel. When they did publish it, I started working on it and realized that the kernel was built by one build script with several intermediaries involved. After studying the build script, I extracted the correct headers and compiled a test module as the build script does, using LLVM CLANG + GCC. It successfully loaded into the kernel. I started building a driver for rtl8188xx and encountered minor code issues and a problem with modpost, which was detecting undefined kernel functions like kernel_read, proc_get, etc. At this stage, I was stuck for two or three days because I couldn't understand why it couldn't find them or where it was looking. It turned out that modpost looks in Module.symvers, which stores all the exported symbols (functions, variables) of the modules and kernel, CRC, the module or binary from which it is exported, and the macro that exports the symbol (EXPORT_SYMBOL_GPL, EXPORT_SYMBOL). So, I looked in Module.symvers and realized that the functions modpost was trying to find were not there, although they were in vermagic and System.map. I thought, "What the hell is going on?" because normally, when the kernel is compiled, it should record every symbol that is present in the build and has a macro. But for some reason, they were not there, even though the macro is there and they are in the build. I tried various options and used makeshift solutions until I reached a function that could not be replaced or removed from the code. I realized that something was filtering the symbols based on certain parameters, and the first parameter that came to mind was: symbols that are used somewhere in the code are successfully exported. It turned out to be true; they were filtered out by Google's kernel build script. I exported a couple of variables into the working environment, and finally, everything worked. I spent a lot of time and effort on this, studied a lot of material, and didn't sleep like a normal person, working from 4 to 6-7 in the morning.