aircrack-ng/rtl8188eus #294: array-index-out-of-bounds in …/ioctl_cfg80211.c:134[67]:100 —  index (16|24) is out of range for type 'u8 [*]'

navid-zamani, 2024-08-31:
Since updating to Mint 22 aka Linux **6.8.0-41**-generic, this driver crashes with the following, when trying to bring **up** any connection that uses **hotspot** mode:
```
[Aug31 02:33] ------------[ cut here ]------------
[  +0,000006] UBSAN: array-index-out-of-bounds in /var/lib/dkms/8188eu/5.3.9/build/os_dep/linux/ioctl_cfg80211.c:1346:100
[  +0,000008] index 16 is out of range for type 'u8 [*]'
[  +0,000005] CPU: 2 PID: 846 Comm: wpa_supplicant Tainted: G        W  OE      6.8.0-41-generic #41-Ubuntu
[  +0,000006] Hardware name: Micro-Star International Co., Ltd. MS-7A38/B350M PRO-VDH (MS-7A38), BIOS A.L4 05/17/2023
[  +0,000003] Call Trace:
[  +0,000002]  <TASK>
[  +0,000005]  dump_stack_lvl+0x76/0xa0
[  +0,000012]  dump_stack+0x10/0x20
[  +0,000005]  __ubsan_handle_out_of_bounds+0xc6/0x110
[  +0,000009]  cfg80211_rtw_add_key+0x5ad/0x900 [8188eu]
[  +0,000165]  nl80211_new_key+0x1da/0x380 [cfg80211]
[  +0,000111]  ? rtw_set_key+0xc0/0x290 [8188eu]
[  +0,000117]  genl_family_rcv_msg_doit+0xfa/0x160
[  +0,000010]  genl_family_rcv_msg+0x184/0x250
[  +0,000005]  ? __pfx_nl80211_pre_doit+0x10/0x10 [cfg80211]
[  +0,000104]  ? __pfx_nl80211_new_key+0x10/0x10 [cfg80211]
[  +0,000072]  ? __pfx_nl80211_post_doit+0x10/0x10 [cfg80211]
[  +0,000000]  genl_rcv_msg+0x4c/0xb0
[  +0,000000]  ? __pfx_genl_rcv_msg+0x10/0x10
[  +0,000000]  netlink_rcv_skb+0x5d/0x110
[  +0,000000]  genl_rcv+0x28/0x50
[  +0,000000]  netlink_unicast+0x24d/0x390
[  +0,000000]  netlink_sendmsg+0x214/0x470
[  +0,000000]  ____sys_sendmsg+0x38e/0x3f0
[  +0,000000]  ___sys_sendmsg+0x9a/0xf0
[  +0,000000]  __sys_sendmsg+0x89/0xf0
[  +0,000000]  __x64_sys_sendmsg+0x1d/0x30
[  +0,000000]  x64_sys_call+0x25e/0x25c0
[  +0,000000]  do_syscall_64+0x7f/0x180
[  +0,000000]  ? irqentry_exit+0x43/0x50
[  +0,000000]  ? common_interrupt+0x54/0xb0
[  +0,000000]  entry_SYSCALL_64_after_hwframe+0x78/0x80
[  +0,000000] RIP: 0033:0x7835efd2c004
[  +0,000000] Code: 15 19 6e 0d 00 f7 d8 64 89 02 b8 ff ff ff ff eb bf 0f 1f 44 00 00 f3 0f 1e fa 80 3d 45 f0 0d 00 00 74 13 b8 2e 00 00 00 0f 05 <48> 3d 00 f0 ff ff 77 4c c3 0f 1f 00 55 48 89 e5 48 83 ec 20 89 55
[  +0,000000] RSP: 002b:00007ffd7928cc08 EFLAGS: 00000202 ORIG_RAX: 000000000000002e
[  +0,000000] RAX: ffffffffffffffda RBX: 00005ffc20175870 RCX: 00007835efd2c004
[  +0,000000] RDX: 0000000000000000 RSI: 00007ffd7928cc40 RDI: 0000000000000006
[  +0,000000] RBP: 00007ffd7928cc30 R08: 0000000000000004 R09: 00000000000000f0
[  +0,000000] R10: 00007ffd7928cd40 R11: 0000000000000202 R12: 00005ffc20257f30
[  +0,000000] R13: 00005ffc20175b50 R14: 00007ffd7928cc40 R15: 0000000000000000
[  +0,000000]  </TASK>
[  +0,000306] ---[ end trace ]---
[  +0,000003] ------------[ cut here ]------------
[  +0,000001] UBSAN: array-index-out-of-bounds in /var/lib/dkms/8188eu/5.3.9/build/os_dep/linux/ioctl_cfg80211.c:1347:100
[  +0,000005] index 24 is out of range for type 'u8 [*]'
[  +0,000003] CPU: 2 PID: 846 Comm: wpa_supplicant Tainted: G        W  OE      6.8.0-41-generic #41-Ubuntu
[  +0,000004] Hardware name: Micro-Star International Co., Ltd. MS-7A38/B350M PRO-VDH (MS-7A38), BIOS A.L4 05/17/2023
[  +0,000003] Call Trace:
[  +0,000001]  <TASK>
[  +0,000002]  dump_stack_lvl+0x76/0xa0
[  +0,000006]  dump_stack+0x10/0x20
[  +0,000004]  __ubsan_handle_out_of_bounds+0xc6/0x110
[  +0,000007]  cfg80211_rtw_add_key+0x5f7/0x900 [8188eu]
[  +0,000163]  nl80211_new_key+0x1da/0x380 [cfg80211]
[  +0,000108]  ? rtw_set_key+0xc0/0x290 [8188eu]
[  +0,000117]  genl_family_rcv_msg_doit+0xfa/0x160
[  +0,000008]  genl_family_rcv_msg+0x184/0x250
[  +0,000005]  ? __pfx_nl80211_pre_doit+0x10/0x10 [cfg80211]
[  +0,000104]  ? __pfx_nl80211_new_key+0x10/0x10 [cfg80211]
[  +0,000106]  ? __pfx_nl80211_post_doit+0x10/0x10 [cfg80211]
[  +0,000104]  genl_rcv_msg+0x4c/0xb0
[  +0,000005]  ? __pfx_genl_rcv_msg+0x10/0x10
[  +0,000005]  netlink_rcv_skb+0x5d/0x110
[  +0,000007]  genl_rcv+0x28/0x50
[  +0,000004]  netlink_unicast+0x24d/0x390
[  +0,000005]  netlink_sendmsg+0x214/0x470
[  +0,000006]  ____sys_sendmsg+0x38e/0x3f0
[  +0,000007]  ___sys_sendmsg+0x9a/0xf0
[  +0,000011]  __sys_sendmsg+0x89/0xf0
[  +0,000007]  __x64_sys_sendmsg+0x1d/0x30
[  +0,000004]  x64_sys_call+0x25e/0x25c0
[  +0,000005]  do_syscall_64+0x7f/0x180
[  +0,000005]  ? irqentry_exit+0x43/0x50
[  +0,000005]  ? common_interrupt+0x54/0xb0
[  +0,000007]  entry_SYSCALL_64_after_hwframe+0x78/0x80
[  +0,000005] RIP: 0033:0x7835efd2c004
[  +0,000012] Code: 15 19 6e 0d 00 f7 d8 64 89 02 b8 ff ff ff ff eb bf 0f 1f 44 00 00 f3 0f 1e fa 80 3d 45 f0 0d 00 00 74 13 b8 2e 00 00 00 0f 05 <48> 3d 00 f0 ff ff 77 4c c3 0f 1f 00 55 48 89 e5 48 83 ec 20 89 55
[  +0,000004] RSP: 002b:00007ffd7928cc08 EFLAGS: 00000202 ORIG_RAX: 000000000000002e
[  +0,000004] RAX: ffffffffffffffda RBX: 00005ffc20175870 RCX: 00007835efd2c004
[  +0,000003] RDX: 0000000000000000 RSI: 00007ffd7928cc40 RDI: 0000000000000006
[  +0,000002] RBP: 00007ffd7928cc30 R08: 0000000000000004 R09: 00000000000000f0
[  +0,000002] R10: 00007ffd7928cd40 R11: 0000000000000202 R12: 00005ffc20257f30
[  +0,000003] R13: 00005ffc20175b50 R14: 00007ffd7928cc40 R15: 0000000000000000
[  +0,000005]  </TASK>
[  +0,000018] ---[ end trace ]---
```

If you need any additional infos, ask me, please. :)

gglluukk, 2024-08-31:
@navid-zamani 

plz try it now

navid-zamani, 2024-09-05:
Thanks! It doesn’t crash anymore!

(Although when starting the AP via `nmcli`, I still get `wpa_supplicant[122835]: Note: nl80211 driver interface is not designed to be used with ap_scan=2; this can result in connection failures` and `nl80211: kernel reports: key setting validation failed`, and apparently NetworkManager ignores `wpa_supplicant.conf` and has no option to set `ap_scan`, so I’m still stuck. But that’s not because of you. [-: … If you know what do to though, I’d appreciate it, since I tried to solve it for literally the entire day. :-/ ] )

closed 2024-09-05 by navid-zamani

gglluukk, 2024-09-05:
follow: 
https://chatgpt.com/share/8f3ff807-eb3d-49a6-b7bb-7cdd86d059b2

check that all other drivers are offloaded:
https://github.com/gglluukk/rtl8188eus?tab=readme-ov-file#howto-buildinstall

BUT:
- at 1) change to:
```
sudo apt install hostapd dnsmasq wireless-tools
```
- before 5):
```
ifconfig wlan0 up
```
- at 7) instead of all iptables:
```
iptables -t nat -I POSTROUTING -s 192.168.1.0/24 -j MASQUERADE
```

hint: in linux/it world you need to google

navid-zamani, 2024-09-09:
LOL. I will most certainly NOT follow the instructions of the stochastic parrot of criminal con men. 😆
Nor use an openly hostile search engine by the biggest brain rapist (=advertiser) in the world.  

I used Gentoo (stage 1 ultra-ricer, baby! Hurt me plenty!) since 2004 and am only “stuck” in the sense of having been too busy to finish the USE flag manager of my self-written installer to be able to keep all installations of Gentoo in sync. I got too used to using Microsoft SystemD and Microsoft NetworkManager by the Microsoft Poettering (aka Elop 2.0) humanoid mole AI. 😋 
No more! You are right: I should, and will, use the real Linux tools. Mint gets the boot within the next days! 🦶