aircrack-ng/rtl8188eus #296: make error

Evixityz, 2024-09-28:
arctichonour@penguin:~$ cd rtl8188eus
make && sudo make install
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/6.6.41-03520-gd3d77f15f842/build M=/home/arctichonour/rtl8188eus  modules
make[1]: *** /lib/modules/6.6.41-03520-gd3d77f15f842/build: No such file or directory.  Stop.
make: *** [Makefile:2070: modules] Error 2

pallinko, 2024-10-06:
I have same problem

└─# make && sudo make install
/bin/sh: 1: bc: not found
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/6.10.9-amd64/build M=/home/pallinko/rtl8188eus  modules
make[1]: *** /lib/modules/6.10.9-amd64/build: Adresár alebo súbor neexistuje.  Stop.
make: *** [Makefile:2065: modules] Error 2

gglluukk, 2024-10-06:
you're out of package:
`linux-headers-YOUR_KERNEL_VERSION`

to install it under debian/ubuntu:
```
apt install linux-headers-`uname -r`
```

pallinko, 2024-10-06:
Thanks  fix for me

Franci000, 2024-11-05:
Linux Headers doesn't exist
How can i resolve?

gglluukk, 2024-11-05:
what distro do you use?
try for debian/ubuntu:
`apt update && apt -y upgrade`
and reboot

after that:
```
apt --yes install build-essential linux-headers-`uname -r`
```

younotknowme, 2026-09-29:
root@localhost:~# git clone https://github.com/aircrack-ng/rtl8188eus                    Cloning into 'rtl8188eus'...
remote: Enumerating objects: 2450, done.
remote: Counting objects: 100% (322/322), done.                                          remote: Compressing objects: 100% (88/88), done.
remote: Total 2450 (delta 268), reused 234 (delta 234), pack-reused 2128 (from 2)        Receiving objects: 100% (2450/2450), 5.43 MiB | 2.50 MiB/s, done.
Resolving deltas: 100% (1224/1224), done.                                                Updating files: 100% (499/499), done.
root@localhost:~# cd rtl8188eus                                                          make && sudo make install
make ARCH=arm64 CROSS_COMPILE= -C /lib/modules/6.17.0-PRoot-Distro/build M=/root/rtl8188eus  modules
make[1]: Entering directory '/root/rtl8188eus'
make[1]: *** /lib/modules/6.17.0-PRoot-Distro/build: No such file or directory.  Stop.   make[1]: Leaving directory '/root/rtl8188eus'
make: *** [Makefile:2065: modules] Error 2
root@localhost:~/rtl8188eus#
I am using Debian root

gglluukk, 2026-09-29:
```
apt --yes install build-essential linux-headers-`uname -r`
```
better use `rtl8xxxu`