aircrack-ng/rtl8188eus #297: Manjaro Linux: Wifi adapter (rtl8188eu) not detected after driver installation

FixedFish, 2024-10-05:
I'm facing an issue with my rtl8188eu wifi adapter on Manjaro Linux with kernel 6.1.0. After compiling and installing the driver from here,  the adapter isn't being detected by the system, even though the driver seems to be successfully installed and loaded.

**What I've tried:**

* Compiled and installed the `rtl8188eu` driver from here.
* `mhwd -l` shows a installed rtl8188eu driver
* `dmesg` says new interface driver registered
* `lsmod` shows 8188eu using by 0, but cfg80211 uses 8188
* Found NetworkManager attempting to read `rtl8188eu.conf/rtl8188eus.conf` in `journalctl`.
* Renamed/removed `rtl8188eus.conf` and restarted NetworkManager without success.

**My system:**

* OS: Manjaro Linux
* Kernel: 6.1.0
* Wifi Adapter: rtl8188eu ("ZTopInc 802.11n INC" by `lsusb`)

FixedFish, 2024-10-07:
Used this fix for 6.1: https://github.com/gglluukk/rtl8188eus
@gglluukk help me

gglluukk, 2024-10-07:
I don't use Manjaro as well as any GUI tools: `iwconfig` could show if there are any interfaces for wifi.

Best approach is to upgrade to Manjaro 24.1 where it runs kernel 6.10 with builtin driver `rtl8xxxu` supporting all dongles with less bugs than `rtl8188eus` has

FixedFish, 2024-10-08:
> I don't use Manjaro as well as any GUI tools: `iwconfig` could show if there are any interfaces for wifi.
> 
> Best approach is to upgrade to Manjaro 24.1 where it runs kernel 6.10 with builtin driver `rtl8xxxu` supporting all dongles with less bugs than `rtl8188eus` has

Im already on manjaro 24.1, switched to kernel 6.10 and same, installed rtl8xxxu from github, iwconfig doesnt show my usb adapter

`iwconfig`
iwconfig
lo        no wireless extensions.

enp0s29u1u4  no wireless extensions.

`ip link`
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN mode DEFAULT group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
4: enp0s29u1u4: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc fq_codel state UNKNOWN mode DEFAULT group default qlen 1000
    link/ether aa:5d:83:65:81:aa brd ff:ff:ff:ff:ff:ff

`lsusb`
Bus 002 Device 006: ID 350b:9101 ZTopInc 802.11n NIC

gglluukk, 2024-10-08:
- you don't need any external `rtl8xxxu` -- use **builtin in kernel** (github one is a backport from kernel so can be buggy)
- i don't have any `ZTopInc` dongles so can't check if it is supported... bad thing is that it may not to be supported currently
- i've googled and found https://www.hetercap.com/en_index.html being connected to `ZTopInc` -> `Shandong ZTop Microelectronics Co` -> `Shandong Heter Microelectronic Technology Co.` so you could request them if they can assist you

dubhater, 2024-10-08:
You can always add the ID 350b:9101 to usb_intf.c in this driver, or to core.c in rtl8xxxu and test. Maybe it's that simple.

FixedFish, 2024-10-09:
> You can always add the ID 350b:9101 to usb_intf.c in this driver, or to core.c in rtl8xxxu and test. Maybe it's that simple.

Adding id to `usb_intf.c` looks like worked, `ip_link` shows wlan0, thanks.
Should do a pr with the addition of my usb adapter?

FixedFish, 2024-10-09:
> If you can connect to a network, yes.

cant up wlan0

`ip link show wlan0`
4: wlan0: <BROADCAST,MULTICAST> mtu 2312 qdisc noop state DOWN mode DEFAULT group default qlen 1000
    link/ether 00:e0:4c:5e:14:1f brd ff:ff:ff:ff:ff:ff

`sudo ip link set wlan0 up`
RTNETLINK answers: Operation not permitted

`ifconfig wlan0`
wlan0: flags=4098<BROADCAST,MULTICAST>  mtu 2312
        ether 00:e0:4c:5e:14:1f  txqueuelen 1000  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

dubhater, 2024-10-09:
Change these lines in the makefile:
```
CONFIG_RTW_DEBUG = n           <- change to y
CONFIG_RTW_LOG_LEVEL = 0       <- change to 4
```
Recompile and reload the driver, then post the unfiltered output of dmesg. Then we can see what is going wrong.

FixedFish, 2024-10-09:
> Change these lines in the makefile:
> 
> ```
> CONFIG_RTW_DEBUG = n           <- change to y
> CONFIG_RTW_LOG_LEVEL = 0       <- change to 4
> ```
> 
> Recompile and reload the driver, then post the unfiltered output of dmesg. Then we can see what is going wrong.

After i compiled with adding `350b:9101` into `usb_intf.c`, when compiled i get that warning

`rtl8188eus/8188eu.o: warning: objtool: mp_xmit_packet_thread+0xb7: thread_exit() is missing a __noreturn annotation`

Here dmesg output, driver cant initialize:
[dmesg_output.txt](https://github.com/user-attachments/files/17311446/dmesg_output.txt)

FixedFish, 2024-10-09:
After purchasing the adapter, it came with a driver disk (for Windows only). I requested Linux drivers from seller, downloaded them, but couldn't compile them. It seems they are intended for an older kernel version (probably 4.1.4). I can provide a link to the driver source code if needed.

dubhater, 2024-10-09:
Yes, the Linux driver from the seller would be helpful. I think your chip is not RTL8188EUS.

FixedFish, 2024-10-09:
> Yes, the Linux driver from the seller would be helpful. I think your chip is not RTL8188EUS.

[YandexDisk](https://disk.yandex.ru/d/FQ54LUw0qPZbRw)
Only on YandexDisk, because seller from Russia

dubhater, 2024-10-09:
That's strange, because I can't find 350b:9101 in any of those Windows drivers, or in the Linux driver. But if you have it working in Windows, just look at Device Manager. The name of the chip should be there.

FixedFish, 2024-10-09:
> That's strange, because I can't find 350b:9101 in any of those Windows drivers, or in the Linux driver. But if you have it working in Windows, just look at Device Manager. The name of the chip should be there.

I dont have access Windows now, I'm on Manjaro, what i can do?

On this site shows it have 2310:9086 and 350b:910 https://treexy.com/products/driver-fusion/database/network-adapters/ztop-microelectronics/9101-usb-80211b-g-n-wireless-device/

Maybe it helpful

dubhater, 2024-10-09:
Can you copy the Windows driver from the CD?

treexy.com wants you to install their driver downloader.

FixedFish, 2024-10-10:
> Can you copy the Windows driver from the CD?
> 
> treexy.com wants you to install their driver downloader.

Driverz from CD here https://www.mediafire.com/file/l1yq742v67j6zft/Drivers.zip/file

In drivers cd i found that in version.log

`####driver info: 
version = 6271 
####firmware info: 
ZT9101_fw_r2688_win 
`

FixedFish, 2024-10-10:
`lspci` output

`03:00.0 Ethernet controller: Realtek Semiconductor Co., Ltd. RTL8111/8168/8211/8411 PCI Express Gigabit Ethernet Controller (rev 06)`

dubhater, 2024-10-10:
Okay, so the chip is called ZT9101. It's not a Realtek chip. I couldn't find any Linux driver.

FixedFish, 2024-10-10:
> But apparently this chip is used in some cameras, which run Linux, so a driver must exist. You should ask the OpenIPC folks if they know anything about it.

ndiswrapper can help or not?

dubhater, 2024-10-10:
I don't know.

FixedFish, 2024-10-10:
So what i can do except for help from OpenIPC?

dubhater, 2024-10-10:
Or you can use Wireshark in Windows to record all the USB communications between the driver and the chip and try to write a Linux driver. That could be fun!

FixedFish, 2024-10-13:
> Okay, so the chip is called ZT9101. It's not a Realtek chip. I couldn't find any Linux driver.

Hi, chipset if `UW01`, seller finally answered me

FixedFish, 2024-10-13:
Cant find drivers for that, can you help me?

dubhater, 2024-10-13:
The seller doesn't know what they're talking about. :) What I found searching for "UW01" is a device, not a chip.

Based on lsusb, your chip is ZT9101. Ask the OpenIPC community, they probably encountered this chip before: https://openipc.org/our-channels

yjun123, 2024-12-09:
Hi, you can try this driver. Although it is a model applicable to ZT9101 UV20/UV30, the VID /PID can match.
https://github.com/yjun123/ZT9101_vendor_linux_driver/

FixedFish, 2024-12-11:
> Hi, you can try this driver. Although it is a model applicable to ZT9101 UV20/UV30, the VID /PID can match. https://github.com/yjun123/ZT9101_vendor_linux_driver/ [...]

I'll check it out today. And where did you find drivers (already saw that it was ported for kernel 6.12) for it?

FixedFish, 2024-12-11:
> Hi, you can try this driver. Although it is a model applicable to ZT9101 UV20/UV30, the VID /PID can match. https://github.com/yjun123/ZT9101_vendor_linux_driver/ [...]

Thank you very much, it worked on kernel 6.6 (x86-x64), but Selinux was giving a warning  me and wouldn't let the driver work, but I fixed it and everything works.

yjun123, 2024-12-13:
> I'll check it out today. And where did you find drivers (already saw that it was ported for kernel 6.12) for it?

The products developed by our company use this Wi-Fi chip. I have obtained the Linux driver from the chip agent. The code was released under GPL2. I think it can be made public if others need this driver. 

I simply adapted it in my environment (Linux v6.12.1). However, my computer doesn't have this chip, so it hasn't been tested.

yjun123, 2024-12-13:
> Thank you very much, it worked on kernel 6.6 (x86-x64), but Selinux was giving a warning me and wouldn't let the driver work, but I fixed it and everything works.

Awesome! I'm so glad that I could help you.

Cluuny, 2024-12-17:
> Thank you very much, it worked on kernel 6.6 (x86-x64), but Selinux was giving a warning me and wouldn't let the driver work, but I fixed it and everything works.

Hi, how did you install the driver, I thought I installed it well but I still have no wlan access from the antenna.

FixedFish, 2024-12-22:
> Hi, how did you install the driver, I thought I installed it well but I still have no wlan access from the antenna.

Can you send the steps that you did?

Cluuny, 2025-01-15:
> Can you send the steps that you did?

It actually started working out of the blue, I automated everything in a script so that it could be configured at machine startup and everything worked fine. thanks a lot!

FixedFish, 2025-01-15:
> It actually started working out of the blue, I automated everything in a script so that it could be configured at machine startup and everything worked fine. thanks a lot!

Can you send your script to me? Do you also need write these command?

`sudo modprobe zt9101_driver
sudo rmmod zt9101_driver
sudo modprobe zt9101_driver
`
For someone reason it loads driver on start but doesn't work correctly, needs reload, do you have same problem?

gglluukk, 2025-09-24:
`rtl8188eu` is NOT your choice for the your NIC. 
you could try those ones:
- https://github.com/yjun123/ZT9101_vendor_linux_driver
- https://codeberg.org/sallecta/driver_wifi_ztopinc

JABIR-EXE, 2025-09-24:
> @JABIR-EXE <https://github.com/JABIR-EXE>
> rtl8188eu is NOT your choice for your NIC.
> you could try those ones:
>
>    - https://github.com/yjun123/ZT9101_vendor_linux_driver
>    - https://codeberg.org/sallecta/driver_wifi_ztopinc

could you tell me which version of kali i need to use so it can work for me

gglluukk, 2025-09-24:
i don't use neither kali, nor having your nic

juanvaloye-op, 2026-03-05:
This worked for me on kernel 6.18
https://gitlab.com/frebonesteban/zt9101_driver